hbldh/bleak #1107: Why some times rasberry pi will raise this log ->KeyError: 'Icon'

SOYOJOE, 2022-11-02:
* bleak version:0.19.1
* Python version:3.9
* Operating System:rasberry pi 4

### Description

Why some times rasberry pi will raise this log **->KeyError: 'Icon'**

### What I Did

when I run the **get_services.py** code,frist connection this exception will not raise,when the script run finished,and run again，the exception come，but when i wait longer，the exception is not raising, but run again ,this comes

```
/bin/python /home/soyo/bleak/examples/get_services.py
```

### Logs
```
soyo@raspberrypi:~ $ /bin/python /home/soyo/bleak/examples/get_services.py
Services:
00010203-0405-0607-0809-0a0b0c0d1912 (Handle: 46): Unknown
0000180f-0000-1000-8000-00805f9b34fb (Handle: 42): Battery Service
0000180a-0000-1000-8000-00805f9b34fb (Handle: 12): Device Information
00001801-0000-1000-8000-00805f9b34fb (Handle: 8): Generic Attribute Profile
00010203-0405-0607-0809-0a0b0c0d1911 (Handle: 51): Unknown
ERROR:root:A message handler raised an exception: 'Icon'.
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/dbus_fast/message_bus.py", line 778, in _process_message
    result = user_handler(msg)
  File "/usr/local/lib/python3.9/dist-packages/bleak/backends/bluezdbus/manager.py", line 798, in _parse_msg
    del self_interface[name]
KeyError: 'Icon'
```

dlech, 2022-11-02:
Can you run this again with `BLEAK_LOGGING=1` so we can see the D-Bus message that triggers this?

SOYOJOE, 2022-11-03:
Hi dlech，
 the logs,what can we find.
```
'org.bluez.HealthManager1': {}}, '/org/bluez/hci0': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Adapter1': {'Address': 'E4:5F:01:9D:19:D7', 'AddressType': 'public', 'Name': 'raspberrypi', 'Alias': 'raspberrypi', 'Class': 0, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001801-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0537', 'Roles': ['central', 'peripheral']}, 'org.freedesktop.DBus.Properties': {}, 'org.bluez.GattManager1': {}, 'org.bluez.LEAdvertisingManager1': {'ActiveInstances': 0, 'SupportedInstances': 5, 'SupportedIncludes': ['tx-power', 'appearance', 'local-name']}, 'org.bluez.Media1': {}, 'org.bluez.NetworkServer1': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': 'C4:19:D1:C4:5B:50', 'AddressType': 'public', 'Name': 'tSample', 'Alias': 'tSample', 'Paired': False, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '00001812-0000-1000-8000-00805f9b34fb', '00010203-0405-0607-0809-0a0b0c0d1911', '00010203-0405-0607-0809-0a0b0c0d1912'], 'Modalias': 'usb:v248Ap8266d0001', 'Adapter': '/org/bluez/hci0', 'ManufacturerData': {147: bytearray(b'\x87\x81')}, 'ServicesResolved': False, 'WakeAllowed': True}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattService1': {'UUID': '00010203-0405-0607-0809-0a0b0c0d1911', 'Device': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50', 'Primary': True, 'Includes': []}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033/char0038': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattCharacteristic1': {'UUID': '00010203-0405-0607-0809-0a0b0c0d2b19', 'Service': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033', 'Value': bytearray(b''), 'Flags': ['write-without-response'], 'WriteAcquired': False}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033/char0038/desc003a': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattDescriptor1': {'UUID': '00002901-0000-1000-8000-00805f9b34fb', 'Characteristic': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033/char0038', 'Value': bytearray(b'')}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033/char0034': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattCharacteristic1': {'UUID': '00010203-0405-0607-0809-0a0b0c0d2b18', 'Service': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033', 'Value': bytearray(b''), 'Notifying': False, 'Flags': ['read', 'notify'], 'NotifyAcquired': False}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033/char0034/desc0037': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattDescriptor1': {'UUID': '00002901-0000-1000-8000-00805f9b34fb', 'Characteristic': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033/char0034', 'Value': bytearray(b'')}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033/char0034/desc0036': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattDescriptor1': {'UUID': '00002902-0000-1000-8000-00805f9b34fb', 'Characteristic': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service0033/char0034', 'Value': bytearray(b'')}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service002e': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattService1': {'UUID': '00010203-0405-0607-0809-0a0b0c0d1912', 'Device': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50', 'Primary': True, 'Includes': []}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service002e/char002f': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattCharacteristic1': {'UUID': '00010203-0405-0607-0809-0a0b0c0d2b12', 'Service': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service002e', 'Value': bytearray(b''), 'Notifying': False, 'Flags': ['read', 'write-without-response', 'notify'], 'WriteAcquired': False, 'NotifyAcquired': False}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service002e/char002f/desc0032': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattDescriptor1': {'UUID': '00002901-0000-1000-8000-00805f9b34fb', 'Characteristic': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service002e/char002f', 'Value': bytearray(b'')}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service002e/char002f/desc0031': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattDescriptor1': {'UUID': '00002902-0000-1000-8000-00805f9b34fb', 'Characteristic': '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service002e/char002f', 'Value': bytearray(b'')}, 'org.freedesktop.DBus.Properties': {}}, '/org/bluez/hci0/dev_C4_19_D1_C4_5B_50/service002a': {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.GattService1': {'UUID': '0000180f-0000-1000-8000-00805f9b34fb', 'Device': '/org/bluez/hci0/ [... cut ...]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C4_19_D1_D0_31_71): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', False)>}, []]
2022-11-03 03:25:30,032 bleak.backends.bluezdbus.client DEBUG: Device disconnected (/org/bluez/hci0/dev_C4_19_D1_D0_31_71)
DEBUG:bleak.backends.bluezdbus.client:Device disconnected (/org/bluez/hci0/dev_C4_19_D1_D0_31_71)
2022-11-03 03:25:30,033 bleak.backends.bluezdbus.client DEBUG: _cleanup_all(/org/bluez/hci0/dev_C4_19_D1_D0_31_71)
DEBUG:bleak.backends.bluezdbus.client:_cleanup_all(/org/bluez/hci0/dev_C4_19_D1_D0_31_71)
```

dlech, 2022-11-03:
Thanks. Apparently it is possible to remove properties that haven't been added. We can add a fix for that.

closed 2022-11-03 by dlech

SOYOJOE, 2022-11-04:
Nice!

PR #1110 (2022-11-03): bluezdbus/manager: ignore KeyError when properties removed
Apparently, BlueZ will sometimes request to remove properties that were not set in the first place.

Fixes: #1107
commit 0d3fa865 (2022-11-03): bluezdbus/manager: ignore KeyError when properties removed

Apparently, BlueZ will sometimes request to remove properties that
were not set in the first place.

Fixes: #1107
PR #1116 (2022-11-06): Release/0.19.2

Fixed
------
* Fixed crash when getting services in WinRT backend in Python 3.11. Fixes #1112.
* Fixed cache mode when retrying get services in WinRT backend. Merged #1102.
* Fixed ``KeyError`` crash in BlueZ backend when removing non-existent property. Fixes #1107.
commit 9c9dac6a (2022-11-06): bluezdbus/manager: ignore KeyError when properties removed

Apparently, BlueZ will sometimes request to remove properties that
were not set in the first place.

Fixes: #1107