hbldh/bleak #1108: Time for write_gatt_char() to complete fluctuates a lot

ChrisM0815, 2022-11-02:
* bleak version: 0.19.1
* Python version: 3.10.8
* Operating System: Windows 10, Raspberry Pi OS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description
Hello,
I´m using the bleak Library to connect to and send different Messages with 7 Bytes to my BLE-Device very fast. The BLE-Devicecan display the count of recieved Messages afterwards. The issue that I´m experiencing is that the time it takes for the Transfer to be completed is fluctuating a lot. I tested this on both a Windows 10 Laptop and a Raspberry Pi using a Bluetooth Dongle and the behaviour on both systems is very similar. 

### What I Did

The Code I´m using is:
```python
import asyncio
import sys

from bleak import BleakScanner, BleakClient

DEFAULT_MAC = "05:55:F5:73:C5:E7"
UUID_WRITE = "0000ff02-0000-1000-8000-00805f9b34fb"

messages = [ bytes([19,6,0,0,0,25,1]),
bytes([19,6,0,0,1,26,1]),
bytes([19,6,0,0,2,27,1]),
bytes([19,6,0,0,3,28,1]),
bytes([19,6,0,0,4,29,1]),
bytes([19,6,0,0,5,30,1]),
bytes([19,6,0,0,6,31,1])]
# real program contains more messages

async def main(ble_address: str):
    device = await BleakScanner.find_device_by_address(ble_address, timeout=20.0)
    print("Connected")
    async with BleakClient(device) as client:
        for i in range(50):
            data = messages[i%len(messages)]
            await client.write_gatt_char(UUID_WRITE, data)
            await asyncio.sleep(0.5)
        print("Finished")

if __name__ == "__main__":
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else DEFAULT_MAC))

```
When I use a lower delay than ~400ms between sending Messages some of the Packages are lost. 
I used wireshark(on windows) an btmon(on the Raspberry) to monitor the bluetooth packages. 
The output of btmon looks like this when no package Loss occours:
```bash
< ACL Data TX: Handle 70 flags 0x00 dlen 14               #267 [hci0] 61.754167
      ATT: Write Command (0x52) len 9
        Handle: 0x000c
          Data: 1306002bcc1001
> HCI Event: Number of Completed Packets (0x13) plen 5    #268 [hci0] 62.052300
        Num handles: 1
        Handle: 70
        Count: 1
```
An like this when packages are Lost(here one package was lost and one recieved):
```bash
< ACL Data TX: Handle 69 flags 0x00 dlen 14               #464 [hci0] 15.152587
      ATT: Write Command (0x52) len 9
        Handle: 0x000c
          Data: 13060012689301
< ACL Data TX: Handle 69 flags 0x00 dlen 14               #465 [hci0] 15.162435
      ATT: Write Command (0x52) len 9
        Handle: 0x000c
          Data: 130600416bc501
> HCI Event: Number of Completed Packets (0x13) plen 5    #466 [hci0] 15.167715
        Num handles: 1
        Handle: 69
        Count: 1
> HCI Event: Number of Completed Packets (0x13) plen 5    #467 [hci0] 15.169807
        Num handles: 1
        Handle: 69
        Count: 1
```
To prevent package loss I added a Delay after every transmission. In order to find the lowest possible delay where no packages are lost I calculated the time it took for each send signal to get the corresponding HCI Event. What i found is that this time fluctuates between under 10ms and over 300ms. Every Time a package is send rather fast (< 20ms) the next package takes a lot longer. This repeats basically every three packages. The time between package sends was relatively consistent at 500ms due to the manual delay.
Printed on a Graph it looks like this:
[image]
The orange line is the time between sending Signals and the Blue Line the time between Sending the Signal and recieving the HCI event.   

Here is the table with the calculated transmission times.

|package|time between ACL Data Tx and HCI Event|
|---|---|
|1|20,71|
|2|197,751|
|3|107,74|
|4|17,701|
|5|257,584|
|6|166,277|
|7|75,214|
|8|312,786|
|9|222,902|
|10|132,269|
|11|42,098|
|12|280,326|
|13|190,488|
|14|99,993|
|15|8,639|
|16|249,375|
|17|161,025|
|18|72,567|
|19|312,516|
|20|223,168|
|21|140,859|
|22|45,122|
|23|286,031|
|24|197,509|
|25|109,095|
|26|20,642|
|27|262,061|
|28|173,265|
|29|84,948|
|30|326,468|
|31|238,218|
|32|149,911|
|33|61,711|
|34|303,295|
|35|215,088|
|36|126,781|
|37|38,574|
|38|280,229|
|39|199,064|
|40|103,612|
|41|14,999|
|42|263,687|
|43|168,377|
|44|80,035|
|45|321,281|
|46|233,076|
|47|144,979|
|48|56,707|
|49|298,133|
|50|209,654|

I made the same measurements with the bluepy python library and and gatttool(automated with pexpect). Using those packages are send much more consistent and a lower delay between sends can be used.
[image]
 
 I attached all Log files(Wireshark and btmon) for bleak, pluepy and gatttool. The relevant packages are at the end of the files. I don´t think this is related to the Hardware I´m using as the measurements for bleak are very similar on both Windows and the Raspberry Pi.

Thanks

Log File for bluepy:
[bluepy_50ms.txt](https://github.com/hbldh/bleak/files/9920293/bluepy_50ms.txt)

Log File for gatttool:
[gatttool_btmon.txt](https://github.com/hbldh/bleak/files/9920294/gatttool_btmon.txt)

Log File for bleak with 500ms Delay
[bleak_500ms_btmon.txt](https://github.com/hbldh/bleak/files/9920297/bleak_500ms_btmon.txt)

Log File for Bleak with no additional Delay
[bleak_0ms_btmon.txt](https://github.com/hbldh/bleak/files/9920309/bleak_0ms_btmon.txt)

ChrisM0815, 2022-11-03:
> When doing "write without response", packet delivery is not guaranteed. If you do "write with response" then you shouldn't loose packets like that, but there is the additional overhead of waiting for the response from the remote device.

My BLE device does not support "write with response" so that´s sadly not an option. 

> Have you done any profiling to see if the extra time is in the Python code or in the OS APIs?

I haven't done any digging yet where the delay happens. My guess at this point would be that it's probably somewhere in the code that runs independent of the used platform as the behavior is nearly identical on both test systems.

bdraco, 2022-11-04:
Make sure you are using the latest `dbus-fast` as there are significant latency improvements in newer versions.

Also replace your dbus implementation with https://github.com/bus1/dbus-broker/wiki as it will also help with the latency as the reference implementation that comes with Raspberry Pi OS is quite slow.

ChrisM0815, 2022-11-07:
I currently do not have access to the hardware I used for testing due to illness, but i used a Laptop running Kubuntu with dbus-fast(1.0) and a smartphone as a BLE-Server. I did not have any package loss or latency issues whatsoever in this configuration. I will do some more testing with my usual test hardware when possible. Pretty safe to say that bleak is not the issue here so I'm closing this issue. 
Thanks for your help.

closed 2022-11-07 by ChrisM0815

ChrisM0815, 2023-02-03:
I have an update on this issue. The increased latency was caused by the BLE-Device sending a connection Parameter Update Request with a pretty high latency value as soon as the notifications are enabled. This didn't happen with bluepy because I did not enable the notifications there while testing. I also did not enable the notifications in the bleak test script, but It seems to be enough to have them enabled at some point before using the test script. Restarting the PC fixed this, and I was able to write without any noticeable Latency.