hbldh/bleak #1115: Suggestion: Add non-async Functionality

ChrisDev8, 2022-11-06:
The async loop is really annoying because in the future I want my bluetooth lights to change color whenever I call a function instead of using a async loop. This would be a-lot easier for many people who are not familiar with async loops like me. I would like if you would add no async loop option so I could do this below:
```python
from bleak import BleakClient
import asyncio

class Lights:
    def __init__(self):
        self.device = 'xx:xx:xx:xx:xx:xx'
        self.handler = 'xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx'
        self.client = BleakClient(self.device)

	def connect(self):
            self.client.connect()

	def disconnect(self):
	    self.client.disconnect()

	def change_color(self, code: str):
	    code = code.replace('#', '')
	    code = f'78{code}00f0ee'
	    if self.client.is_connected is True:
	        self.client.write_gatt_char(
		    self.handler,
		    code.encode('utf-8')
	        )
            else:
                return

lights = Lights()
lights.connect()
lights.change_color('FF0000')
lights.disconnect()
```

dlech, 2022-11-06:
Using asyncio is one of the core design choices of Bleak. If you don't want to use asyncio, then https://github.com/OpenBluetoothToolbox/SimpleBLE looks like a promising alternative.

ChrisDev8, 2022-11-08:
# Problem
When I tried to install SimpleBLE, it showed a weird error but still managed to install but you cannot import simplepyble it. For some reason I cannot recreate the install error when reinstalling it. The error I get when trying to import it is:
```
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/simplepyble.cpython-311-darwin.so, 0x0002): tried: '/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/simplepyble.cpython-311-darwin.so' (mach-o file, but is an incompatible architecture (have (x86_64), need (arm64e)))
```
Maybe suggests that it's not compatible with my MAC M1 using arm64 architecture.
# Solution
Just use BleakClient / BleakScanner the normal way but by using something like this.
```python
from bleak import (
    BleakClient,
    BleakScanner,
    BLEDevice,
    AdvertisementData
)
from asyncio import run as run_loop
from rich import print

class Bluetooth:
    @staticmethod
	async def base_discover(timeout: float):
	    devices = await BleakScanner.discover(timeout, return_adv=True)
            return devices

    @staticmethod
	def base_get_info(
	    device: BLEDevice,
            advertise_data: AdvertisementData
	):
	    info_dict = {
	        'device_info': {
		    'address': device.address,
		    'name': device.name,
		    'details': device.details
		},
		'advertise_info': {
		    'local_name': advertise_data.local_name,
		    'manufacturer_data': advertise_data.manufacturer_data,
		    'rssi': advertise_data.rssi,
		    'service_data': advertise_data.service_data,
		    'service_uuids': advertise_data.service_uuids,
		    'tx_power': advertise_data.tx_power
		}
            }
	return info_dict

	@classmethod
	def discover(cls, timeout: float = 5):
	    devices = run_loop(cls.base_discover(timeout))
	    information = []
	    for device in devices.values():
	        info = cls.base_get_info(device[0], device[1])
		information.append(info)
	    return information

info = Bluetooth.discover()
print(info)
```
But still would like something like this implemented into Bleak.

ChrisDev8, 2022-11-08:
# One Suggestion Though
The code mentioned above produces this dumb error:
```
IndexError: NSRangeException - Cannot remove an observer <CentralManagerDelegate 0x106074cf0> for the key path "isScanning" from <CBCentralManager 0x600003bd4200> because it is not registered as an observer.
```
I spend two hours on it and find my bluetooth was turned off.

closed 2023-03-17 by dlech