hbldh/bleak #1120: Does not reconnect after day of not using

Cleric1199, 2022-11-08:
* bleak version: 0.19.4
* Python version: 3.10.4
* Operating System: Windows

### Description

I have a bluetooth chip which does not have a pairing option but I have to connect to it once in a while. I initialize a connection and remember the device (and the client). Every day I want to do a check of the system state so I try to connect to the device again, however after a day it will say it does not know the device anymore. The MAC address does not change and I can still discover the device when running another python script.

Maybe this is not a problem with bleak but rather that Windows does something in the background?

The error that pops up:
_Failed to connect to <random_id>. If the device requires pairing, then pair first. If the device uses a random address, it may have changed._

dlech, 2022-11-08:
Given that this error message doesn't come from Bleak, this sounds very much like a Windows problem.

Cleric1199, 2022-11-09:
I trigger the python code through a powershell terminal. Will get the debug log overnight

The minimal code:

```
class TheDevice:
    _client: BleakClient = None

    def __init__(self, device):
        self._device = device
        self._loop = asyncio.new_event_loop()
        self._connected = False
        self._connected_device = None

    def connect(self):
        self._loop.run_until_complete(self.establish_connection())

    async def establish_connection(self):
        if self._client:
            await self._client.connect()
            self._connected = self._client.is_connected
            if self._connected:
                print(F"Connected to {self._connected_device.name}")
        else:
            print(f"Connecting to {self._device.name}")
            self._connected_device = self._device
            self._client = BleakClient(self._device.address)

    def close(self):
        self._loop.run_until_complete(self.disconnect())

    async def disconnect(self):
        disconnect_attempts = 0
        await self._client.disconnect()
        self._connected = False
        print(f"Disconnected from {self._connected_device.name}!")
```

Application:

```
# find and connect to device
device = get_device() # returns TheDevice object
device.connect()

while True:
    device.close()
    print("Waiting 12 hours")
    time.sleep(3600 * 12)
    print("Connecting again!")
    device.connect()
```