hbldh/bleak #1123: Failing to get services on Windows in Bleak v0.19.4

dlech, 2022-11-09:
I have same issue here, when I upgrade to bleak-0.19.4, [WinError -2147024874] start to occur.
In bleak-0.17.0, I got occasionally success, about 3 times out of 10, the other fail will be timeout.
In bleak-0.19.4, fail case **Red mark** ATT command start to different from success case.

I need help

## Error 
bleak version:
bleak 0.19.4
bleak-winrt 1.2.0
Python version:
Python 3.9.6
Operating System:
Windows 10 Home, Version 10.0.19044 Build 19044
```
DEBUG:bleak.backends.winrt.client:Connecting to BLE device @ C2:5E:51:CF:C5:2E 
DEBUG:bleak.backends.winrt.client:Get Services...
DEBUG:bleak.backends.winrt.client:session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x00000146E5CB8270>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
DEBUG:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: services changed
WARNING:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: unhandled services changed event
DEBUG:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: services changed
WARNING:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: unhandled services changed event
DEBUG:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: services changed
WARNING:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: unhandled services changed event
ERROR:asyncio:Task exception was never retrieved
future: <Task finished name='Task-1' coro=<run() done, defined at d:\Git\python\debug_tool\crest_debugger\interface\ble.py:144> exception=PermissionError(13, 'The device does not recognize the command', None, -2147024874, None)>
Traceback (most recent call last):
  File "d:\Git\python\debug_tool\crest_debugger\interface\ble.py", line 155, in run
    async with BleakClient(address,winrt=dict(use_cached_services=False)) as client:
  File "C:\Users\.pyenv\pyenv-win\versions\3.9.6\lib\site-packages\bleak\__init__.py", line 433, in __aenter__
    await self.connect()
  File "C:\Users\.pyenv\pyenv-win\versions\3.9.6\lib\site-packages\bleak\__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Users\.pyenv\pyenv-win\versions\3.9.6\lib\site-packages\bleak\backends\winrt\client.py", line 380, in connect
    await asyncio.gather(wait_connect_task, wait_get_services_task)
  File "C:\Users\.pyenv\pyenv-win\versions\3.9.6\lib\site-packages\bleak\backends\winrt\client.py", line 627, in get_services
    await service.get_characteristics_async(*args),
PermissionError: [WinError -2147024874] The device does not recognize the command
```
[image]

## Success
bleak version:
bleak 0.17.0
bleak-winrt 1.2.0
Python version:
Python 3.9.6
Operating System:
Windows 10 Home, Version 10.0.19044 Build 19044
[image]

_Originally posted by @JohnsonChouCrestDiving in https://github.com/hbldh/bleak/issues/1079#issuecomment-1308877325_

dlech, 2022-11-09:
Thanks for the detailed report. This is a different error from the one reported in the other issue, so I started a new issue.

I wasn't able to reproduce the `PermissionError: [WinError -2147024874] The device does not recognize the command` error, but I was able to reproduce  `WARNING:bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: unhandled services changed event` which revealed some problems with the existing implementation.

Can you please try https://github.com/hbldh/bleak/pull/1124 to see if it fixes your issue?

JohnsonChouCrestDiving, 2022-11-10:
Nope, it find Nordic UART Service but not keep searching its characteristic(Wireshark Image below)
Is the next action should be **Sent Read By Type Request, GATT Characteristic Declaration, Handles: 0x0020..0x0029** ?

```sh
DEBUG:bleak_.bleak.backends.winrt.scanner:Received C2:5E:51:CF:C5:2E: CREST-CR5.
DEBUG:bleak_.bleak.backends.winrt.scanner:skipping callback, waiting for scan response
DEBUG:bleak_.bleak.backends.winrt.scanner:Received C2:5E:51:CF:C5:2E: .
DEBUG:bleak_.bleak.backends.winrt.scanner:21 devices found. Watcher status: 3.
DEBUG:bleak_.bleak.backends.winrt.client:Connecting to BLE device @ C2:5E:51:CF:C5:2E
DEBUG:bleak_.bleak.backends.winrt.client:getting services (service_cache_mode=<BluetoothCacheMode.UNCACHED: 1>, cache_mode=<BluetoothCacheMode.UNCACHED: 1>)...
DEBUG:bleak_.bleak.backends.winrt.client:session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x00000149145599B0>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
DEBUG:bleak_.bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: services changed
DEBUG:bleak_.bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: restarting get services due to services changed event
DEBUG:bleak_.bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: services changed
DEBUG:bleak_.bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: services changed
DEBUG:bleak_.bleak.backends.winrt.client:getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=<BluetoothCacheMode.UNCACHED: 1>)...
DEBUG:bleak_.bleak.backends.winrt.client:C2:5E:51:CF:C5:2E: restarting get services due to services changed event
DEBUG:bleak_.bleak.backends.winrt.client:getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=<BluetoothCacheMode.UNCACHED: 1>)...
DEBUG:bleak_.bleak.backends.winrt.client:closing requester
DEBUG:bleak_.bleak.backends.winrt.client:closing session
ERROR:asyncio:Task exception was never retrieved
future: <Task finished name='Task-1' coro=<run() done, defined at d:\Git\python\debug_tool\crest_debugger\interface\ble.py:144> exception=PermissionError(13, 'The device does not recognize 
the command', None, -2147024874, None)>
Traceback (most recent call last):
  File "d:\Git\python\debug_tool\crest_debugger\interface\ble.py", line 155, in run
    async with BleakClient(address,winrt=dict(use_cached_services=False)) as client:
  File "D:\Git\bleak_\bleak\__init__.py", line 433, in __aenter__
    await self.connect()
  File "D:\Git\bleak_\bleak\__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "D:\Git\bleak_\bleak\backends\winrt\client.py", line 413, in connect
    self.services = get_services_task.result()
  File "D:\Git\bleak_\bleak\backends\winrt\client.py", line 659, in get_services
    await FutureLike(service.get_characteristics_async(*args)),
  File "D:\Git\bleak_\bleak\backends\winrt\client.py", line 1041, in __await__
    yield self  # This tells Task to wait for completion.
PermissionError: [WinError -2147024874] The device does not recognize the command
```
[image]

dlech, 2022-11-10:
Thanks for testing. Can you add timestamps to the Python logging so we can see how much time passes before the device disconnects.

> **service_cache_mode has change after services change restarting**, does the changing has any influence on the issue?

Possibly. This is trying to follow the recommendation of the Windows documentation of how to use the API, but it is hard to say what it actually does behind the scenes.

dlech, 2022-11-10:
Also, can you share your code so we can rule out a problem with the code causing the device to disconnect early?

dlech, 2022-11-11:
And can you share the full wireshark log?

JohnsonChouCrestDiving, 2022-11-14:
@dlech  All files you need is in the attachment
I change some local path from 0.19.5 as bleak_, you can find it in the attachment

for better view in wireshark, you could use **bluetooth.src == C2:5E:51:CF:C5:2E || bluetooth.dst == C2:5E:51:CF:C5:2E** as filter

[bleak_issue_1123_221114_01.zip](https://github.com/hbldh/bleak/files/10000711/bleak_issue_1123_221114_01.zip)
[image]

```sh
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\ble_test.py - run                            - 36    - MainThread - 16280  : BLE start connect...
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 77:6F:E8:61:3C:02: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 06:EB:F7:8E:60:DC: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 18:EE:69:2B:C4:0C: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 139   - MainThread - 16280  : skipping callback, waiting for scan response
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 18:EE:69:2B:C4:0C: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 79:58:AE:92:36:F9: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 51:24:0C:99:4D:39: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 139   - MainThread - 16280  : skipping callback, waiting for scan response
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 51:24:0C:99:4D:39: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 1B:31:FB:43:24:B1: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 49:42:DB:FE:C9:51: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 139   - MainThread - 16280  : skipping callback, waiting for scan response
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 49:42:DB:FE:C9:51: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 44:73:D6:50:63:25: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 139   - MainThread - 16280  : skipping callback, waiting for scan response   
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 44:73:D6:50:63:25: MeetUp Soft Remote.
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 18:EE:69:2B:C4:0C: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 40:8F:D1:10:FE:C3: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 139   - MainThread - 16280  : skipping callback, waiting for scan response
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 16280  : Received 0C:8C:DC:2D:A8:50: .
2022-11-14 15:45:27 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 139   - MainThread - 16280  : skipping callback, waiting for scan response
[... cut ...] luetooth_GenericAttributeProfile.GattDeviceService object at 0x000001CCC95D83B0>
2022-11-14 15:45:35 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\backends\winrt\client.py - handle_disconnect              - 281   - MainThread - 16280  : closing requester
2022-11-14 15:45:35 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_01\bleak_\bleak\b

ackends\winrt\client.py - handle_disconnect              - 298   - MainThread - 16280  : closing session
```

dlech, 2022-11-16:
Looking at the logs, it seems like the device is misconfigured. In packet 503, a *Find By Type Value Request* for primary services starting at handle 0x0001 is sent. The response (packet 506) says there is a service that starts with handle 0x0011 and ends with 0x0013. The next request starts at handle 0x0014 and receives an error response that indicates there are no more primary services. However, later starting with packet 516, the *Read By Group Type Request*s and responses show a Nordic UART service that wasn't listed before. In packet 527, there are characteristics listed with handles less than 0x0011, but there was never a service reported for these characteristics. 

Can you modify the device so that it gives correct responses to see if that fixes the issue?

JohnsonChouCrestDiving, 2022-11-17:
Hi @dlech , 
according to [the Bluetooth Core Specification 4.0](https://www.bluetooth.com/specifications/specs/core-specification-4-0/), the ATT command **Find By Type Value Request** only obtain the handle which have 16-bit UUID(Nordic UART service have 128-bit UUID)
[image]

Could you change collect service method to **Read By Group Type Request** or other cmd could receive 128-bit UUID handle?

Our product works successfully on the other platform, ex: nRF Connect
[image]

[image]

JohnsonChouCrestDiving, 2022-11-17:
The problem seems occur at **get_characteristics_async**, I've check all services uuid is correct(Terminal below), but get characteristics method seems not work

```sh
2022-11-17 21:57:16 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 28548  : Received C2:5E:51:CF:C5:2E: CREST-CR5.
2022-11-17 21:57:16 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 139   - MainThread - 28548  : skipping callback, waiting for scan response
2022-11-17 21:57:16 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\scanner.py - _received_handler              - 104   - MainThread - 28548  : Received C2:5E:51:CF:C5:2E: .
2022-11-17 21:57:16 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\scanner.py - _stopped_handler               - 226   - MainThread - 28548  : 12 devices found. Watcher status: 3.
2022-11-17 21:57:16 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - connect                        - 251   - MainThread - 28548  : Connecting to BLE device @ C2:5E:51:CF:C5:2E
<_bleak_winrt_Windows_Devices_Bluetooth.BluetoothLEDevice object at 0x000001C03D21AEB0>
connect ------------ event <asyncio.locks.Event object at 0x000001C03D24E6A0 [unset]>
2022-11-17 21:57:16 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - get_services                   - 621   - MainThread - 28548  : getting services (service_cache_mode=None, cache_mode=None)...
2022-11-17 21:57:17 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - session_status_changed_event_handler - 325   - Dummy-1    - 31720  : session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000001C03C938B70>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - get_services                   - 650   - MainThread - 28548  : services: [<_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattDeviceService object at 0x000001C03D257670>, <_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattDeviceService object at 0x000001C03D257030>, <_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattDeviceService object at 0x000001C03D2575F0>, <_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattDeviceService object at 0x000001C03D257570>]
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - get_services                   - 654   - MainThread - 28548  : UUIDS: [UUID('00001800-0000-1000-8000-00805f9b34fb'), UUID('00001801-0000-1000-8000-00805f9b34fb'), UUID('6e400001-b5a3-f393-e0a9-e50e24dcca9e'), UUID('0000180d-0000-1000-8000-00805f9b34fb')]
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - get_services                   - 663   - MainThread - 28548  : service: <_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattDeviceService object at 0x000001C03D2572D0>
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - services_changed_handler       - 265   - Dummy-1    - 31720  : C2:5E:51:CF:C5:2E: services changed
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - connect                        - 419   - MainThread - 28548  : C2:5E:51:CF:C5:2E: restarting get services due to services changed event
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - services_changed_handler       - 265   - Dummy-1    - 31720  : C2:5E:51:CF:C5:2E: services changed
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - services_changed_handler       - 265   - Dummy-1    - 31720  : C2:5E:51:CF:C5:2E: services changed
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - services_changed_handler       - 265   - Dummy-1    - 31720  : C2:5E:51:CF:C5:2E: services changed
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - get_services                   - 621   - MainThread - 28548  : getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
2022-11-17 21:57:18 - DEBUG      - c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py - connect                        - 419   - MainThread - 28548  : C2:5E:51:CF:C5:2E: restarting get services due to services changed event
[... cut ...]
    self.services = get_services_task.result()
  File "c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py", line 670, in get_services
    await FutureLike(service.get_characteristics_async(*args)),
  File "c:\Users\eopi-Johnson\Documents\bleak_issue_1123_221114_01\bleak_\bleak\backends\winrt\client.py", line 1052, in __await__
    yield self  # This tells Task to wait for completion.
PermissionError: [WinError -2147024874] The device does not recognize the command
```

dlech, 2022-11-17:
> he ATT command **Find By Type Value Request** only obtain the handle which have 16-bit UUID(Nordic UART service have 128-bit UUID)

I think the mention of 16-bit is referring to the attribute type that is requested, which in this case is 0x2800 GATT Primary Service. So all Primary Services should be enumerated.

> Our product works successfully on the other platform, ex: nRF Connect

Windows uses *Find By Type Value Request* that other OSes don't (AFAIK). You can do a packet capture on other OSes to see the difference.

> ```shell
> PermissionError: [WinError -2147024874] The device does not recognize the command
> ```

I think this occurs because of packet 531 (repeated in in 534 and 540). The device is responding that it can't find the characteristic and Windows passes on this error.

commit 27e7f10b (2022-11-09): winrt/client: fix issues with getting services

There were a number of issues with getting services not being restarted
properly when a services changed event occurred. Cancellation wasn't
handled properly in FutureLike and it wasn't used in enough places.

Now, the entire `get_services()` method is restarted when an event is
received instead of only `self._requester.get_gatt_services_async()`.

Fixes #1123
commit 9daba5f7 (2022-11-09): winrt/client: fix issues with getting services

There were a number of issues with getting services not being restarted
properly when a services changed event occurred. Cancellation wasn't
handled properly in FutureLike and it wasn't used in enough places.

Now, the entire `get_services()` method is restarted when an event is
received instead of only `self._requester.get_gatt_services_async()`.

Fixes #1123
commit 60e820d0 (2022-11-09): winrt/client: fix issues with getting services

There were a number of issues with getting services not being restarted
properly when a services changed event occurred. Cancellation wasn't
handled properly in FutureLike and it wasn't used in enough places.

Now, the entire `get_services()` method is restarted when an event is
received instead of only `self._requester.get_gatt_services_async()`.

Fixes #1123
commit b4b29c09 (2022-11-09): winrt/client: fix issues with getting services

There were a number of issues with getting services not being restarted
properly when a services changed event occurred. Cancellation wasn't
handled properly in FutureLike and it wasn't used in enough places.

Now, the entire `get_services()` method is restarted when an event is
received instead of only `self._requester.get_gatt_services_async()`.

Fixes #1123
closed 2024-04-29 by dlech