hbldh/bleak #1126: scanner.discovered_devices seem ephemeral

jeffsf, 2022-11-11:
* bleak version: 0.19.4
* Python version: 3.11.0
* Operating System: Debian Bullseye on `Raspberry Pi 3 Model B Rev 1.2`
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

(additional version information at end of this post)

### Description

Exploring use of the `scanner.discovered_devices` property instead of a home-rolled implementation of similar functionality, I expected that `discovered_devices` would be accessible after the scanner "run" completed.

I found that the return value from `scanner.deisover(timeout=5)` returned a populated list. However, retrieving `scanner.discovered_devices` was empty. I also checked `scanner.discovered_devices_and_advertisement_data` and found it to be empty as well.

It was not clear to me from the comments in the code and resulting documentation of those properties that they would not be available after the scanner stopped. https://bleak.readthedocs.io/en/latest/api/scanner.html#bleak.BleakScanner.discovered_devices

### What I Did

```
@pytest.mark.asyncio
async def test_basic_scanner():
    scanner = bleak.BleakScanner()
    retval = await scanner.discover(timeout=5)
    ddval = scanner.discovered_devices
    ddaad = scanner.discovered_devices_and_advertisement_data
    assert len(retval) > 0, "This test needs discoverable devices"
    print()
    # This shows an empty list
    pprint.pprint(ddval)
    print()
    # This shows an empty dict
    pprint.pprint(ddaad)
    print()
    # This shows a populated list
    pprint.pprint(retval)
    assert len(retval) == len(ddval)
    assert len(retval) == len(ddaad)
```

### Environment

```
(pyde1-devel) jeff@pi-walnut:~ $ pip list
Package            Version
------------------ ---------
aiosqlite          0.17.0
async-timeout      4.0.2
attrs              22.1.0
bleak              0.19.4
certifi            2022.9.24
charset-normalizer 2.1.1
dbus-fast          1.73.0
idna               3.4
iniconfig          1.1.1
packaging          21.3
paho-mqtt          1.6.1
pip                22.3.1
pluggy             1.0.0
pyparsing          3.0.9
pytest             7.2.0
pytest-asyncio     0.20.2
pytest-timeout     2.1.0
PyYAML             6.0
requests           2.28.1
setuptools         65.5.1
typing_extensions  4.4.0
urllib3            1.26.12

(pyde1-devel) jeff@pi-walnut:~ $ python --version
Python 3.11.0

(pyde1-devel) jeff@pi-walnut:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

(pyde1-devel) jeff@pi-walnut:~ $ bluetoothctl --version
bluetoothctl: 5.55
```

dlech, 2022-11-11:
`discover()` is a class method, so calling it on an instance doesn't affect the instance.

jeffsf, 2022-11-11:
Thanks -- This behaves as expected
```
@pytest.mark.asyncio
async def test_basic_scanner():
    # NB: Scanner.discover() is a class method, not an instance method
    # NB: Scanning starts when the context is entered
    #     Scanning will be stopped if running when the context is exited
    async with bleak.BleakScanner() as scanner:
        await asyncio.sleep(5)
        await scanner.stop()
        ddval = scanner.discovered_devices
        ddaad = scanner.discovered_devices_and_advertisement_data
        assert len(ddval) > 0, "This test needs discoverable devices"
        # print()
        # pprint.pprint(ddval)
        # print()
        # pprint.pprint(ddaad)
        # print()
        assert len(ddval) == len(ddaad)
```

closed 2022-11-11 by dlech