hbldh/bleak #1128: Average time elapsed from discovery of a device to connection to it seems a little high

lorenzofelletti, 2022-11-14:
* bleak version: 0.19.0
* Python version: 3.9.6
* Operating System: macOS 12.6

### Description

I am trying to measure the time between the discovery of a device and the connection to it. To do so, I used the `time` library. 

What I found out is that the connection time (since discovery) is on average 1.5 to 2.5 seconds, depending on how crowded the room is.
This does seem a little high for me, _I wanted to know if this is normal for BLE, or if I am doing something wrong_. I tried to research on the web for information about the average time of connection from discovery, but didn't find out anything useful.

### What I Did

This is a minimal workable example you can use to test what I'm doing:
```Python
import asyncio
import time
from typing import List, Optional
from bleak import AdvertisementData, BLEDevice, BleakClient, BleakScanner

# Connection to the device duration
CONNECTION_TIME = 5  # seconds

my_service_uuid = "MY_SERVICE_UUID"

# variables to meter the connection latency
devices_discovery_time: dict[BLEDevice, Optional[float]] = {}
latencies: List[float] = []

async def run_ble_client(device: BLEDevice):
    print(f"Attempting connection to {device}")

    async with BleakClient(device) as client:
        await client.connect()
        connection_time = time.time()
        latencies.append(connection_time - devices_discovery_time[device])
        devices_discovery_time.pop(device)

        print(f"Connected to {device}")
        await asyncio.sleep(CONNECTION_TIME)
        print(f"disconnecting from {device}")
        await client.disconnect()

async def app():
    device_to_connect_to: Optional[BLEDevice] = None
    stop_event = asyncio.Event()

    def scan_callback(device: BLEDevice, advertising_data: AdvertisementData):
        nonlocal device_to_connect_to
        if devices_discovery_time.get(device) is None:
            devices_discovery_time[device] = time.time()

        device_to_connect_to = device
        stop_event.set()  # awakens stop_event.wait()

    async with BleakScanner(scan_callback, service_uuids=[my_service_uuid]) as _:
        await stop_event.wait()

    if isinstance(device_to_connect_to, BLEDevice):
        await run_ble_client(device_to_connect_to)

if __name__ == "__main__":
    for i in range(1, 5):
        asyncio.run(app())

    mean_latency = sum(latencies) / len(latencies)
    print(f"mean connection latency: {mean_latency}")
```

You only need to change the `my_service_uuid` to the UUID of your service.

I run:
```Python
python mve.py
```

The output was:
```Bash
Attempting connection to ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
Connected to ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
disconnecting from ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
Attempting connection to ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
Connected to ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
disconnecting from ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
Attempting connection to ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
Connected to ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
disconnecting from ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
Attempting connection to ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
Connected to ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
disconnecting from ECC1BAB3-EDDF-92AA-7B1F-F9AA26CCCC4E: Pixel 3a
mean connection latency: 2.3234023451805115
```

dlech, 2022-11-16:
Two seconds does seem a bit long for macOS in my experience - but this is just my memory which is known to be very faulty, so don't take it to seriously.

You could log Bluetooth packets (see troubleshooting page in the docs) to see what is actually going on under the hood. Some BLE devices may have low advertising intervals, so may be slow to connect because of that.

Also, there is #654 that could possibly help speed things up on macOS by only enumerating the service(s) that will actually be used.

lorenzofelletti, 2022-11-21:
@dlech Thank you for your response. I did some further experiment and indeed enhancing the advertising rate of my Peripheral did improve the latency. Also, thanks to your advice of logging the packets, I found out that in the time I accounted for the connection was instead included the time to discover the peripheral's services, which accounted for the majority of it. Thus, the actual connection time is in fact much lower, and nearer to what I expected it to be.

As for #654, am I missing something or was `services` argument removed in later versions, because I cannot find the option right now, so I am not able to use it?

Really thank you for your answer, it solved my problem

dlech, 2022-11-21:
#654 has not been merged yet. You can check out that branch or `pip install https://github.com/hbldh/bleak/archive/refs/heads/requested-services.zip` to test it.

closed 2022-11-21 by dlech