hbldh/bleak #1134: Docs: what about creating an installation guide?

lorenzofelletti, 2022-11-22:
* bleak version:
* Python version:
* Operating System:
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

It would be nice to have an in-depth installation guide for the library for the different platforms. For example, I had a hard time figuring out that `pyobjc` was needed on macOS in order for the library to work properly, and I still have not figured it out the requirements to install it on Windows, since there is no official installation guide on the docs.  I think it would be of great value for beginners (like me).

### What I Did

### Logs

bojanpotocnik, 2022-11-22:
So running
```shell
pip install bleak
```
or
```shell
python3 -m pip install bleak
```
was successful, but then running e.g.
```shell
python3 bleak/examples/discover.py
```
failed with __, or how did you figure it out?

lorenzofelletti, 2022-11-22:
Honestly I don't remember exacly.
On macOS I think discovery was working but connection was problematic, then a friend of mine told me that for `pybluez` he needed `pyobjc` and that maybe it was the same case, so I did amd thing started working properly.

Anyway, the point is that I cannot find that `pyobjc` was required anywhere on the documentation, same for the Windows requirements, and I think this is a problem especially for newbies, so a dedicated docs page could be created.

dlech, 2022-11-22:
We already have an official install guide at https://bleak.readthedocs.io/en/latest/installation.html. And `pip` takes care of installing dependencies so we don't list the dependencies in the docs.

closed 2023-02-19 by dlech