hbldh/bleak #1137: Windows 10 TimeOut Error

celiafb, 2022-11-22:
* bleak version: 22.2.2
* Python version: 3.11
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I'm trying to run a custom script to connect to a peripheral. This script works perfectly on my Mac and on my other Windows 10, and everything works, but when trying to set it up on a new Windows 10 laptop, I keep getting this TimeOut error.

### What I Did

This is the error that I get:
```
Traceback (most recent call last):
  File "C:\Users\Celia\Scripts\Python\csblesimp.py", line 103, in <module>
    loop.run_until_complete(main())
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "C:\Users\Celia\Scripts\Python\csblesimp.py", line 85, in main
    await data_client(device)
  File "C:\Users\Celia\Scripts\Python\csblesimp.py", line 31, in data_client
    async with BleakClient(device) as client:
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\__init__.py", line 433, in __aenter__
    await self.connect()
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 380, in connect
    await asyncio.gather(wait_connect_task, wait_get_services_task)
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 612, in get_services
    get_services_task.result(),
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 933, in result
    return self._async_result.get_results()
OSError: [WinError -2147483629] The object has been closed

= RESTART: C:\Users\Celia\Scripts\Python\csblesimp.py

Warning (from warnings module):
  File "C:\Users\Celia\Scripts\Python\csblesimp.py", line 101
    loop = asyncio.get_event_loop()
DeprecationWarning: There is no current event loop
Device: DE:D3:4D:CA:60:FD
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 380, in connect
    await asyncio.gather(wait_connect_task, wait_get_services_task)
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Celia\Scripts\Python\csblesimp.py", line 103, in <module>
    loop.run_until_complete(main())
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "C:\Users\Celia\Scripts\Python\csblesimp.py", line 85, in main
    await data_client(device)
  File "C:\Users\Celia\Scripts\Python\csblesimp.py", line 31, in data_client
    async with BleakClient(device) as client:
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\__init__.py", line 433, in __aenter__
    await self.connect()
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 379, in connect
    async with async_timeout.timeout(timeout):
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\async_timeout\__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "C:\Users\Celia\AppData\Local\Programs\Python\Python310\lib\site-packages\async_timeout\__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
```

dlech, 2022-11-22:
Please provide Bleak debug logs and Wireshark Bluetooth packet capture as described in the troubleshooting page of the docs.

dlech, 2022-11-22:
> * bleak version: 22.2.2

There is no such Bleak version, what is the actual version used?

dlech, 2022-11-22:
> * Operating System: Windows 10

What is the exact Windows 10 version, e.g. 22H1. It is the same or different from the working computer?

celiafb, 2022-11-22:
I'm sorry about that, let me clarify:

- Bleak version: latest (v0.19.5)
- Windows version: Windows 10 Pro, version 1909 (in the working computer, I have Windows 11 Home, version 21H2)

Bleak debugs:
DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
Device: DE:D3:4D:CA:60:FD
2022-11-22 12:44:29,945 bleak.backends.winrt.scanner DEBUG: Received B2:5C:DA:64:FE:26: .
2022-11-22 12:44:29,946 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:29,953 bleak.backends.winrt.scanner DEBUG: Received 62:76:1C:3B:B8:B4: .
2022-11-22 12:44:29,954 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:29,955 bleak.backends.winrt.scanner DEBUG: Received 74:C9:AB:C8:0D:BB: .
2022-11-22 12:44:29,956 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:29,959 bleak.backends.winrt.scanner DEBUG: Received 74:C9:AB:C8:0D:BB: .
2022-11-22 12:44:29,960 bleak.backends.winrt.scanner DEBUG: Received 7B:25:33:9C:A0:7E: .
2022-11-22 12:44:29,964 bleak.backends.winrt.scanner DEBUG: Received 25:47:E6:8F:3A:2A: .
2022-11-22 12:44:29,971 bleak.backends.winrt.scanner DEBUG: Received 73:56:4B:BE:83:9A: .
2022-11-22 12:44:29,972 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:29,974 bleak.backends.winrt.scanner DEBUG: Received 73:56:4B:BE:83:9A: .
2022-11-22 12:44:30,077 bleak.backends.winrt.scanner DEBUG: Received 16:EC:3F:48:BC:E5: .
2022-11-22 12:44:30,187 bleak.backends.winrt.scanner DEBUG: Received 32:37:4A:5F:A0:23: .
2022-11-22 12:44:30,190 bleak.backends.winrt.scanner DEBUG: Received 04:D4:75:3F:92:7A: .
2022-11-22 12:44:30,195 bleak.backends.winrt.scanner DEBUG: Received 23:D8:7F:B3:C9:64: .
2022-11-22 12:44:30,295 bleak.backends.winrt.scanner DEBUG: Received 2C:E2:A3:A0:41:4E: .
2022-11-22 12:44:30,305 bleak.backends.winrt.scanner DEBUG: Received 7A:E8:D1:10:BA:EA: .
2022-11-22 12:44:30,305 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:30,310 bleak.backends.winrt.scanner DEBUG: Received 3A:A1:E0:68:FC:4C: .
2022-11-22 12:44:30,311 bleak.backends.winrt.scanner DEBUG: Received 3A:FE:C2:76:F6:6F: .
2022-11-22 12:44:30,318 bleak.backends.winrt.scanner DEBUG: Received 29:74:13:41:A6:6B: .
2022-11-22 12:44:30,416 bleak.backends.winrt.scanner DEBUG: Received 3A:FE:C2:76:F6:6F: .
2022-11-22 12:44:30,420 bleak.backends.winrt.scanner DEBUG: Received 0F:97:25:E9:25:C4: .
2022-11-22 12:44:30,424 bleak.backends.winrt.scanner DEBUG: Received 1A:56:3B:D0:CE:80: .
2022-11-22 12:44:30,429 bleak.backends.winrt.scanner DEBUG: Received 11:7E:7F:CC:90:A6: .
2022-11-22 12:44:30,437 bleak.backends.winrt.scanner DEBUG: Received 58:C2:EA:31:5C:50: .
2022-11-22 12:44:30,438 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:30,439 bleak.backends.winrt.scanner DEBUG: Received 58:C2:EA:31:5C:50: .
2022-11-22 12:44:30,533 bleak.backends.winrt.scanner DEBUG: Received 11:7E:7F:CC:90:A6: .
2022-11-22 12:44:30,547 bleak.backends.winrt.scanner DEBUG: Received 38:F0:13:43:07:8B: .
2022-11-22 12:44:30,657 bleak.backends.winrt.scanner DEBUG: Received 73:EA:CA:C7:66:DF: .
2022-11-22 12:44:30,658 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:30,660 bleak.backends.winrt.scanner DEBUG: Received 73:EA:CA:C7:66:DF: .
2022-11-22 12:44:30,665 bleak.backends.winrt.scanner DEBUG: Received 54:4E:F4:81:71:A3: .
2022-11-22 12:44:30,665 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:30,666 bleak.backends.winrt.scanner DEBUG: Received 54:4E:F4:81:71:A3: .
2022-11-22 12:44:30,672 bleak.backends.winrt.scanner DEBUG: Received 5D:FC:71:70:D9:2E: .
2022-11-22 12:44:30,673 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:30,675 bleak.backends.winrt.scanner DEBUG: Received 5D:FC:71:70:D9:2E: .
2022-11-22 12:44:30,677 bleak.backends.winrt.scanner DEBUG: Received 25:47:E6:8F:3A:2A: .
2022-11-22 12:44:30,775 bleak.backends.winrt.scanner DEBUG: Received 7B:F9:05:D2:19:13: .
2022-11-22 12:44:30,776 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:30,779 bleak.backends.winrt.scanner DEBUG: Received 25:47:E6:8F:3A:2A: .
2022-11-22 12:44:30,790 bleak.backends.winrt.scanner DEBUG: Received B2:5C:DA:64:FE:26: .
2022-11-22 12:44:30,790 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:30,792 bleak.backends.winrt.scanner DEBUG: Received 39:F8:69:48:59:F8: .
2022-11-22 12:44:30,794 bleak.backends.winrt.scanner DEBUG: Received 04:D4:75:3F:92:7A: .
2022-11-22 12:44:30,796 bleak.backends.winrt.scanner DEBUG: Received 16:9A:5A:59:30:BF: .
2022-11-22 12:44:30,891 bleak.backends.winrt.scanner DEBUG: Received 04:D4:75:3F:92:7A: .
2022-11-22 12:44:30,897 bleak.backends.winrt.scanner DEBUG: Received 5F:B1:59:15:F5:BE: .
2022-11-22 12:44:30,898 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-11-22 12:44:30,904 bleak.backends.winrt.scanner DEBUG: Received 16:9A:5A:59:30:BF: .
2022-11-22 12:44:31,005 bleak.backends.winrt.scanner DEBUG: Received B2:5C:DA:64:FE:26: .
2022-11-22 12:44:31,005 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
[... cut ...]
167	22.921214	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
168	24.120058	de:d3:4d:ca:60:fd (Celia's ADC Measure)	localhost ()	ATT	18	Rcvd Read Response, Handle: 0x0007 (Generic Access Profile: Peripheral Preferred Connection Parameters)
169	27.184063	host	controller	HCI_CMD	7	Sent Disconnect
170	27.187187	controller	host	HCI_EVT	7	Rcvd Command Status (Disconnect)
171	27.722205	controller	host	HCI_EVT	7	Rcvd Disconnect Complete
172	27.723210	controller	host	HCI_EVT	19	Rcvd Vendor-Specific

dlech, 2022-11-22:
Thanks for the logs. Can you zip and upload the full wireshark logs?

dlech, 2022-11-22:
Also, can you share the program to reproduce the problem?

celiafb, 2022-11-22:
Yes, I think these are the full Wireshark logs -- please let me know if there's something else you need.

Thank you!
[BTP.pcapng.zip](https://github.com/hbldh/bleak/files/10070230/BTP.pcapng.zip)
[csblesimp.py.zip](https://github.com/hbldh/bleak/files/10070235/csblesimp.py.zip)

dlech, 2022-11-22:
> File "C:\Users\Celia\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 382, in connect

Line 382 is blank in Bleak v0.19.5. Are you sure you are using that version?

https://github.com/hbldh/bleak/blob/bbb79b714eacf5d6c900e01da7a287978c391a13/bleak/backends/winrt/client.py#LL382

dlech, 2022-11-22:
Also try rebooting the computer to rule out any Windows Bluetooth stack issues.

celiafb, 2022-11-22:
Just tried rebooting -- same issue, unfortunately. I'm not sure about the Bleak version. I installed it last week so I figured it would be the latest one. I can't find where to check which version I have, but I'll comment back when I do.

dlech, 2022-11-22:
Please try with v0.19.5.

celiafb, 2022-11-22:
Just updated to v0.19.5 using "pip install -U bleak", then tried again, and getting the same TimeOutError.

dlech, 2022-11-22:
Can you redo the logs with the new version?

celiafb, 2022-11-22:
Yes, will start on those now.

celiafb, 2022-11-22:
[logs.txt](https://github.com/hbldh/bleak/files/10071275/logs.txt)
[BTP0195.pcapng.zip](https://github.com/hbldh/bleak/files/10071276/BTP0195.pcapng.zip)

dlech, 2022-11-22:
logs.txt still looks like the old version.

>   File "C:\Users\Celia\AppData\Local\Programs\Python\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 382, in connect

celiafb, 2022-11-22:
Is there another way I can ensure I'm updating bleak? Running "pip install -U bleak" seemed to update it, and then running "pip show bleak" shows v019.5 as the current version...

dlech, 2022-11-22:
Do you have multiple versions of Python installed? Maybe pip is running on a different Python install? You can try running `py -3.11 -m pip show bleak` to ensure a specific Python version and then run the program with `py -3.11 program,py`.

celiafb, 2022-11-22:
Okay, yeah, I think the Bleak v0.19.5 was only updated for my past Python 3.10 version. Now, I updated Bleak to v0.19.5 on Python 3.11 which I am using to run the script. Same issue, and I'm attaching the logs.
[logs3.txt](https://github.com/hbldh/bleak/files/10071400/logs3.txt)
[BTP3.pcapng.zip](https://github.com/hbldh/bleak/files/10071406/BTP3.pcapng.zip)

dlech, 2022-11-22:
Looks like there was a typo setting BLEAK_LOGGING in logs3.txt so it didn't take effect.

dlech, 2022-11-22:
Also, you should change 
```python

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    try:
        loop.run_until_complete(main())
    finally:
        loop.close()
``` 

to 

```python

if __name__ == "__main__":
    asyncio.run(main())
```

 for proper cleanup of tasks on exit/crash

celiafb, 2022-11-22:
Sorry about that. I fixed the typo and regenerated the logs, and I changed the code according to your suggestion (thank you by the way!).
[logs4.txt](https://github.com/hbldh/bleak/files/10071456/logs4.txt)
[BTP4.pcapng.zip](https://github.com/hbldh/bleak/files/10071460/BTP4.pcapng.zip)

dlech, 2022-11-22:
Does the other Windows 10 computer that was working still work with Bleak v0.19.5?

 Does it make a difference if you make the timeout longer?

dlech, 2022-11-23:
Do older versions of Bleak work (i.e. 0.18.x)?

Does #1143 make a difference? Even it it doesn't can you capture the debug logs since more logging was added there?

celiafb, 2022-11-23:
The computer where it does work has v0.18.1 installed. How can I install this version on my current desktop to check if it would work?

Also, I'm not sure how to make the Timeout longer. I'll check out #1143 and get back to you.

Thanks!

dlech, 2022-11-23:
you can `pip install bleak==0.18.1` to install a specific version.  Likewise, you can `pip install https://github.com/hbldh/bleak/archive/refs/heads/winrt-fix-leaking-services.zip` to get the branch from #1143.

To change the timeout, add a `timeout=30` argument to `BleakClient(...)`.

celiafb, 2022-11-23:
Okay, I haven't changed the timeout yet, but I'll do that next. 

Running with the #1143 branch seems to connect to my peripheral, but then nothing gets logged and everything becomes sort of stuck, until I keyboard interrupt with Ctrl+C.
[BTPlogs5.pcapng.zip](https://github.com/hbldh/bleak/files/10078542/BTPlogs5.pcapng.zip)
[logs5.txt](https://github.com/hbldh/bleak/files/10078543/logs5.txt)

celiafb, 2022-11-23:
With this setup, changing the timeout to 30 does not make a difference.

dlech, 2022-11-23:
FYI, I just pushed some more changes to #1143 to disable the retry altogether since the hang seems to always happen after that.

celiafb, 2022-11-29:
Thanks for following up. Updating to the latest version in #1143 unfortunately didn't solve the issue either.

celiafb, 2022-11-29:
[BTPlogs6.pcapng.zip](https://github.com/hbldh/bleak/files/10116585/BTPlogs6.pcapng.zip)
[logs6.txt](https://github.com/hbldh/bleak/files/10116586/logs6.txt)

dlech, 2022-11-29:
I still see "restarting get services due to services changed event" in the logs, so either I didn't do the change right or you didn't get the latest changes. Do you see `_retry_on_services_changed` in your local copy of `bleak/backends/winrt/client.py`?

celiafb, 2022-11-29:
No, I can't find _retry_on_services_changed in my local copy of bleak/backends/winrt/client.py.

celiafb, 2022-11-29:
I installed the updated branch using `pip install https://github.com/hbldh/bleak/archive/refs/heads/winrt-fix-leaking-services.zip`

dlech, 2022-11-29:
Maybe you got a cached version of the zip file? The latest changes should have `_retry_on_services_changed` in the code.

celiafb, 2022-11-29:
Hm, okay. Is there a different link that I should be using for the update?

dlech, 2022-11-29:
Add ` --no-cache-dir` to the `pip install` command?

celiafb, 2022-11-29:
Done. Still no change. 

[logs7.txt](https://github.com/hbldh/bleak/files/10116790/logs7.txt)
[BTPlogs7.pcapng.zip](https://github.com/hbldh/bleak/files/10116796/BTPlogs7.pcapng.zip)

dlech, 2022-11-29:
It looks like it is still running the wrong version. Are you sure you are installing the package in the right place? (like in https://github.com/hbldh/bleak/issues/1137#issuecomment-1324296824)

celiafb, 2022-11-29:
I think so? My logs7.txt file shows the command I used to install it, and I made sure to run `py -3.11` to specify this. Does that seem incorrect?

dlech, 2022-11-29:
Uninstall it then install it again? If I manually download that zip file, it looks right.

celiafb, 2022-11-29:
Thanks! Okay, so now `_retry_on_services_changed` is definitely in `client.py`. Here are my logs.

[logs8.txt](https://github.com/hbldh/bleak/files/10117065/logs8.txt)
[BTPlogs8.pcapng.zip](https://github.com/hbldh/bleak/files/10117070/BTPlogs8.pcapng.zip)

dlech, 2022-11-29:
OK, making progress now. I had a mistake in my latest change that caused the new exception. I just pushed the fix, so try uninstalling and installing the package again.

celiafb, 2022-11-29:
Oh, it works now! It was able to connect to the peripheral and receive the bytearray values as expected. I'm attaching the logs again in case that's helpful now or in the future. 

Thank you so much, David, for helping me deal with this issue!

[logs9.txt](https://github.com/hbldh/bleak/files/10117172/logs9.txt)
[BTPlogs9.pcapng.zip](https://github.com/hbldh/bleak/files/10117174/BTPlogs9.pcapng.zip)

dlech, 2022-11-29:
Thanks for all of the testing!