hbldh/bleak #1138: Improve the short description of the project

kamil-cy, 2022-11-22:
Hi, please consider changing the short description from `Bluetooth Low Energy platform Agnostic Klient for Python` to something like `Bluetooth Low Energy platform Agnostic Klient for Python (Bluetooth LE client)`. It's important to add `client` and `LE` (Low Energy).

Why? For better GitHub's SEO. 

Google finds Bleak without concern.
[image]

But GitHub has poor search engine. Bleak can be found if `client` is with a typo as a `klient`:
 [image]

The `client` results doesn't show Bleak:
[image]

The same for `Bluetooth LE client`
[image]

kamil-cy, 2022-11-22:
I humbly suggest to use both of them, to better find Bleak. Maybe `Bluetooth Low Energy platform Agnostic Klient for Python (Bluetooth LE client to work with BLE devices).` In that case we have all keywords (bluetooth, LE, BLE, client) that helps to find the project just using GitHub's search bar.

hbldh, 2022-12-02:
Ok, I can do that. Aren't the tags included in the search as well? So we can have a description that is descriptive and not SEO optimized and less readable?

kamil-cy, 2022-12-08:
Sorry for the late reply. This is weird either for me. As in screenshots it's impossible to find Bleak at a glance by typing e.g. `Bluetooth low energy client python` or even `Bluetooth LE client python`. However `Bluetooth low energy klient` it's a perfect shot, but to be honest `klient` isn't an often used word.

hbldh, 2022-12-12:
The short description is now changed to `A cross platform Bluetooth Low Energy Client for Python using asyncio`. 
It felt silly to add BLE or similar when it already says that. SEO should not have to lead to stupid repetition of similar terms in a description field.

Is it good enough?

kamil-cy, 2022-12-12:
Sounds good. When searching for `Bluetooth low energy client python` then Bleak is in the result, but for `Bluetooth LE client python` is still missing, however it's not a big deal. At the beginning I suggested to add `BLE` and `LE`, but my second thoughts are slightly different. So as You said, there is no sense to add `BLE` or `LE` just for better SEO.

Bleak now is easily to find, so thank you for that.

closed 2022-12-12 by dlech