hbldh/bleak #1150: BLE device disconnects on Windows when sending a lot of writes

ChrisM0815, 2022-11-28:
* bleak version: 0.20.0a1
* Python version: 3.10.8
* Operating System: Windows 10.0.19044 

### Description
I´m using bleak to send 7 Byte packages to my BLE device. Here´s the Script i´m using:
```python
import asyncio
import sys
import logging

from bleak import BleakScanner, BleakClient
from bleak.backends.winrt.client import logger as bleak_log

DEFAULT_MAC = "05:55:F5:73:C5:E7"
UUID_WRITE = "0000ff02-0000-1000-8000-00805f9b34fb"

messages = [ bytes([19,6,0,0,0,25,1]),
bytes([19,6,0,0,1,26,1]),
bytes([19,6,0,0,2,27,1]),
bytes([19,6,0,0,3,28,1]),
bytes([19,6,0,0,4,29,1]),
bytes([19,6,0,0,5,30,1]),
bytes([19,6,0,0,6,31,1]),
bytes([19,6,0,0,7,32,1]),
bytes([19,6,0,0,8,33,1]),
bytes([19,6,0,0,9,34,1]),
bytes([19,6,0,0,10,35,1]),
bytes([19,6,0,0,11,36,1]),
bytes([19,6,0,0,12,37,1])]
# More messages in real script

async def main(ble_address: str):

    # Logging
    bleak_log.setLevel(logging.DEBUG)
    ch = logging.StreamHandler()
    ch.setLevel(logging.DEBUG)
    ch.setFormatter(logging.Formatter('---BLEAK:%(levelname)s %(message)s'))     
    bleak_log.addHandler(ch)
    bleak_log.disabled = False

    print("Connecting")
    device = await BleakScanner.find_device_by_address(ble_address, timeout=20.0)
 
    async with BleakClient(device) as client:
        print("\nSending")
        i = 0
        while True:
            data = messages[i%len(messages)]
            await client.write_gatt_char(UUID_WRITE, data)
            i += 1

if __name__ == "__main__":
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else DEFAULT_MAC))

```
On a Raspberry this script runs fine until manually stopped. On my Windows PC the BLE device is disconnected after sending some Messages. 

### What I Did
When running the Script on Windows i get the following output:
```
>python bleaktest_bulk.py
Connecting
---BLEAK:DEBUG Connecting to BLE device @ 05:55:F5:73:C5:E7
---BLEAK:DEBUG getting services (service_cache_mode=None, cache_mode=None)...
---BLEAK:DEBUG session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000002240B95F730>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
---BLEAK:DEBUG max_pdu_size_changed_handler: 512
---BLEAK:DEBUG 05:55:F5:73:C5:E7: services changed
---BLEAK:DEBUG 05:55:F5:73:C5:E7: restarting get services due to services changed event
---BLEAK:DEBUG 05:55:F5:73:C5:E7: services changed
---BLEAK:DEBUG getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
---BLEAK:DEBUG 05:55:F5:73:C5:E7: restarting get services due to services changed event
---BLEAK:DEBUG 05:55:F5:73:C5:E7: services changed
---BLEAK:DEBUG getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
---BLEAK:DEBUG 05:55:F5:73:C5:E7: restarting get services due to services changed event
---BLEAK:DEBUG 05:55:F5:73:C5:E7: services changed
---BLEAK:DEBUG getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
---BLEAK:DEBUG 05:55:F5:73:C5:E7: restarting get services due to services changed event
---BLEAK:DEBUG getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...

Sending
---BLEAK:DEBUG Disconnecting from BLE device...
```
In a previous run I got the Following Traceback with an Windows Error after the disconnect:
```
Traceback (most recent call last):
  File "bleaktest_bulk.py", line 306, in <module>
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else DEFAULT_MAC))
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\asyncio\base_events.py", line 649, in run_until_complete
    return future.result()
  File "bleaktest_bulk.py", line 294, in main
    await client.write_gatt_char(UUID_WRITE, data)
  File "C:\Users\xxxxx\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\__init__.py", line 630, in write_gatt_char
    await self._backend.write_gatt_char(char_specifier, data, response)
  File "C:\Users\xxxxx\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\bleak\backends\winrt\client.py", line 817, in write_gatt_char
    await characteristic.obj.write_value_with_result_async(buf, response),
OSError: [WinError -2147467260] Vorgang abgebrochen
```

### Logs
Here´s the Wireshark Capture File + a Capture File where I got the Windows error(bleaktest_packets2):
[bleaktest_packets.zip](https://github.com/hbldh/bleak/files/10104599/bleaktest_packets.zip)

dlech, 2022-11-28:
By default, `BleakClient.,write_gatt_char()` does "write without response" which doesn't have any way to provide backpressure to ensure that the writes are not being sent faster than the device can handle. You could add `response=True` to wait for a response from the device before continuing or you can add a sufficiently long `asyncio.sleep()` between writes to slow things down.

ChrisM0815, 2022-11-29:
Hello,
thanks for your help. My BLE device doesn´t support writes with response so that´s sadly not an option. Adding an `asyncio.sleep()` with approx 45ms after each send does indeed prevent those disconnects. When running the script(without  `asyncio.sleep()`) on a raspberry pi there are no disconnects either so i don´t think the BLE device is the issue. I will test the script on a different PC running Windows and see if it behaves any different.