hbldh/bleak #1172: [WinError -2140864509] The attribute cannot be written, when trying to establish notifications.

olt96, 2022-12-12:
* bleak version: 0.19.5
* Python version: 3.7
* Operating System:  
Edition	Windows 10 Pro
Version	20H2
Installed on	‎27-‎Dec-‎20
OS build	19042.1706
Experience	Windows Feature Experience Pack 120.2212.4170.0

### Description
 
Trying to get notifications to work for a custom BLE device (esp32).
Notifications work when testing from my phone, and the 2902 descriptor can be written to
However when trying to do the same with bleak, I get the following error
proof of notifications working on phone 
[image]

### What I Did
This is the testing BLE code that I have 
```
import asyncio
import bleak
#
BOARD_NAME = 'table_ble'
BOARD_STATE_CHARACTERISTIC_UUID = 'beb5483e-36e1-4688-b7f5-ea07361b26a8'

async def findTable():
    global TableDevice
    TableDevice = None
    print ("Searching for table...")
    while TableDevice == None:
        devices = await bleak.BleakScanner.discover()
        TableDevice = next((ble_device for ble_device in devices if ble_device.name == BOARD_NAME), None)
    print("table found!", TableDevice)

async def connectAndReceiveData():
    async with bleak.BleakClient(TableDevice) as client:
        print("Connected: {0}".format(client.is_connected))
        await client.start_notify(BOARD_STATE_CHARACTERISTIC_UUID, callback=newBoardStateReceived)
        await asyncio.sleep(10000000)

def newBoardStateReceived(sender: int, data: bytearray):
    print(f"{sender}: {data}")
    # Device will disconnect when block exits.
asyncio.run(findTable())
asyncio.run(connectAndReceiveData())

```

### Logs
```

Searching for table...
2022-12-12 07:04:01,708 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:01,708 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-12-12 07:04:01,708 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:01,770 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:01,770 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:02,145 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:02,161 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:02,395 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:02,395 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:02,598 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:02,614 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:02,962 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:02,978 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:03,087 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:03,103 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:03,196 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
2022-12-12 07:04:03,212 bleak.backends.winrt.scanner DEBUG: Received 38:18:4C:3D:18:A3: LE_WH-CH700N.
[... 84 lines cut ...]
2022-12-12 07:04:12,446 bleak.backends.winrt.client DEBUG: max_pdu_size_changed_handler: 517
2022-12-12 07:04:15,356 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x043D9F50>, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED
2022-12-12 07:04:15,356 bleak.backends.winrt.client DEBUG: max_pdu_size_changed_handler: 23
2022-12-12 07:04:15,356 bleak.backends.winrt.client DEBUG: closing requester
2022-12-12 07:04:15,356 bleak.backends.winrt.client DEBUG: closing session
Traceback (most recent call last):
  File "C:/Users/MyName/PycharmProjects/testBle/main.py", line 1, in <module>
    import ble_connection
  File "C:\Users\MyName\PycharmProjects\testBle\ble_connection.py", line 30, in <module>
    asyncio.run(connectAndReceiveData())
  File "C:\Users\MyName\AppData\Local\Programs\Python\Python37-32\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Users\MyName\AppData\Local\Programs\Python\Python37-32\lib\asyncio\base_events.py", line 584, in run_until_complete
    return future.result()
  File "C:\Users\MyName\PycharmProjects\testBle\ble_connection.py", line 19, in connectAndReceiveData
    await client.start_notify(BOARD_STATE_CHARACTERISTIC_UUID, callback=newBoardStateReceived)
  File "D:\Python Projects 2022\testBle\lib\site-packages\bleak\__init__.py", line 649, in start_notify
    await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
  File "D:\Python Projects 2022\testBle\lib\site-packages\bleak\backends\winrt\client.py", line 891, in start_notify
    cccd
OSError: [WinError -2140864509] The attribute cannot be written
```

here's the log from wireshark aswell
[captured packets.zip](https://github.com/hbldh/bleak/files/10205127/captured.packets.zip)

dlech, 2022-12-12:
It looks like Windows might be using cached services for that device. So try setting the `winrt`, `used_cached_services` to `False`.

olt96, 2022-12-12:
I tried setting this with: 
`
bleak.backends.winrt.client.WinRTClientArgs.use_cached_services = False`
in the beginning of the code but I still get the same error. 
Am I doing this right ?

dlech, 2022-12-12:
Nope. Its a constructor arg for `BleakClient`.

```python
BleakClient(..., winrt=dict(use_cached_services=False))...
```

olt96, 2022-12-12:
this seems to work fine. Thank you!

closed 2022-12-12 by olt96

alexanderturner, 2023-09-24:
Appear to be having the same issues on MacOS

julienroulle, 2023-10-09:
Hi,

I am facing the same issue on Windows 11, bleak 0.21.1, python 3.11.6

The error message is as follow:
```
File "\.venv\Lib\site-packages\bleak\backends\winrt\client.py", line 974, in start_notify
await winrt_char.write_client_characteristic_configuration_descriptor_async(
OSError: [WinError -2140864509] The attribute cannot be written
```

I've tested with the above code, and I get the same error. I've tried the solution provided, on @olt96's code, and got the same output.

I'll continue searching, but if you have any idea, let me know :)
Thanks