hbldh/bleak #1173: Not able to send acknowledgment to the BLE device

aakefin, 2022-12-14:
* bleak version: 0.19.5
* Python version: 3.8.8
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I have a BLE device from which I am trying to get the data in my computer. I used the uart_service.py from the bleak examples and it helped me in receiving the data that I require. 
But to erase the data from the BLE device, I must send an acknowledgement to the device after I have received the data from the device in my computer. I tried using the write_gatt_char() method to send the data to the device but there is no change in the device. 

### What I Did

I tried sending data to both RX and TX CHAR UUID but no response from the device. 
I am trying to implement this - 
[image]

Code - await client.write_gatt_char(rx_char, b'#E' ,True)

Do provide assistance.

aakefin, 2022-12-15:
> Without any logs showing what is going on under the hood, we cannot provide any insight.

2022-12-15 09:52:10,325 bleak.backends.winrt.scanner DEBUG: Received 34:81:F4:35:F7:8D: NT Y1001101190009.
2022-12-15 09:52:10,325 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2022-12-15 09:52:10,333 bleak.backends.winrt.scanner DEBUG: Received 34:81:F4:35:F7:8D: .
2022-12-15 09:52:10,337 bleak.backends.winrt.scanner DEBUG: 5 devices found. Watcher status: 3.
2022-12-15 09:52:10,341 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ 34:81:F4:35:F7:8D
2022-12-15 09:52:10,537 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2022-12-15 09:52:10,605 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x00000182B007D350>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
2022-12-15 09:52:10,713 bleak.backends.winrt.client DEBUG: max_pdu_size_changed_handler: 158
Pair- True
Connected, start typing and press ENTER...
*Received data gets printed*

2022-12-15 09:52:19,996 bleak.backends.winrt.client DEBUG: Disconnecting from BLE device...
2022-12-15 09:52:23,108 bleak.backends.winrt.client DEBUG: max_pdu_size_changed_handler: 23
2022-12-15 09:52:23,112 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x00000182B007D6F0>, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED
Device was disconnected, goodbye.
2022-12-15 09:52:23,116 bleak.backends.winrt.client DEBUG: closing requester
2022-12-15 09:52:23,116 bleak.backends.winrt.client DEBUG: closing session

aakefin, 2022-12-15:
So the device that I am connecting to is a BM71 Bluetooth Low Energy Module. It has auto pattern. 

Below here are the services that I am getting from the device - 

INFO:__main__:[Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
INFO:__main__:  [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2): Device Name (read), Value: b'NT Y1001101190009'
INFO:__main__:  [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4): Appearance (read), Value: b'\x00\x00'
INFO:__main__:[Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 16): Device Information
INFO:__main__:  [Characteristic] 00002a29-0000-1000-8000-00805f9b34fb (Handle: 17): Manufacturer Name String (read), Value: b'ISSC'
INFO:__main__:  [Characteristic] 00002a24-0000-1000-8000-00805f9b34fb (Handle: 19): Model Number String (read), Value: b'BM70'
INFO:__main__:  [Characteristic] 00002a25-0000-1000-8000-00805f9b34fb (Handle: 21): Serial Number String (read), Value: b'0000'
INFO:__main__:  [Characteristic] 00002a27-0000-1000-8000-00805f9b34fb (Handle: 23): Hardware Revision String (read), Value: b'5505 102_BLDK3'
INFO:__main__:  [Characteristic] 00002a26-0000-1000-8000-00805f9b34fb (Handle: 25): Firmware Revision String (read), Value: b'01040101'
INFO:__main__:  [Characteristic] 00002a28-0000-1000-8000-00805f9b34fb (Handle: 27): Software Revision String (read), Value: b'0000'
INFO:__main__:  [Characteristic] 00002a23-0000-1000-8000-00805f9b34fb (Handle: 29): System ID (read), Value: b'\x00\x00\x00\x00\x00\x00\x00\x00'
INFO:__main__:  [Characteristic] 00002a2a-0000-1000-8000-00805f9b34fb (Handle: 31): IEEE 11073-20601 Regulatory Cert. Data List (read), Value: b'\x00\x00\x00\x00\x01\x00\x00\x00'
INFO:__main__:[Service] 49535343-5d82-6099-9348-7aac4d5fbc51 (Handle: 48): Unknown
INFO:__main__:  [Characteristic] 49535343-026e-3a9b-954c-97daef17e26e (Handle: 49): Unknown (write,notify), Value: None
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 51): Client Characteristic Configuration) | Value: b'\x00\x00'
INFO:__main__:[Service] 49535343-c9d0-cc83-a44a-6fe238d06d33 (Handle: 64): Unknown
INFO:__main__:  [Characteristic] 49535343-aca3-481c-91ec-d85e28a60318 (Handle: 65): Unknown (write,notify), Value: None
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 67): Client Characteristic Configuration) | Value: b'\x00\x00'
INFO:__main__:[Service] 49535343-fe7d-4ae5-8fa9-9fafd205e455 (Handle: 80): Unknown
INFO:__main__:  [Characteristic] 49535343-1e4d-4bd9-ba61-23c647249616 (Handle: 81): Unknown (write-without-response,write,notify), Value: None
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 83): Client Characteristic Configuration) | Value: b'\x00\x00'
INFO:__main__:  [Characteristic] 49535343-8841-43f4-a8d4-ecbe34729bb3 (Handle: 84): Unknown (write-without-response,write), Value: None
INFO:__main__:  [Characteristic] 49535343-4c8a-39b3-2f49-511cff073b7e (Handle: 86): Unknown (write,notify), Value: None
INFO:__main__:          [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 88): Client Characteristic Configuration) | Value: b'\x00\x00'

And these are the UUIDs that I am using in uart_services.py file -
 
UART_SERVICE_UUID = "49535343-FE7D-4AE5-8FA9-9FAFD205E455"
UART_RX_CHAR_UUID = "49535343-8841-43F4-A8D4-ECBE34729BB3"
UART_TX_CHAR_UUID = "49535343-1E4D-4BD9-BA61-23C647249616"

dlech, 2022-12-15:
Are you sure you are using the correct characteristics?

aakefin, 2022-12-15:
> Are you sure you are using the correct characteristics?

It is correct.
It is written in BM71 document.