hbldh/bleak #1180: bleak should have a consistent cross-platform exception if no Bluetooth is available on a device

MarkusPiotrowski, 2022-12-22:
* bleak version: probably 0.19.5 (`bleak` seem to have no `.__version__`)
* Python version: 3.8.10
* Operating System: Win7/Win10, MacOS
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Trying to use `BleakScanner.discover` if no bluetooth hardware is present or if bluetooth is turned off raises different exception on different platforms.

On Windows it raises an `OSError` (sorry for the German error message):
```
OSError: [WinError -2147221231] ClassFactory kann angeforderte Klasse nicht liefern
```
On MacOS it raises a generic `BleakError('Bluetooth device is turned off')`

I don't know which error is raised on Linux (not tested yet).

This makes it difficult to handle this situation ('no bluetooth') properly in a cross-platform independent manner, e.g. by `try... except` clauses.

It would be nice to have a consistent platform-indenpendent exception, e.g.  `BleakBluetoothNotFoundError` for this situation.

### What I Did

Minimal workable example:
Run this on different devices (Windows/Mac/Linux) with or without Bluetooth adapters or with Bluetooth turned off:

```
import asyncio
from bleak import BleakScanner, BleakClient

async def test():
    devices = await BleakScanner.discover()

asyncio.run(test())
```

### Logs
see above in **Description**

dlech, 2022-12-22:
Related request: #320

MarkusPiotrowski, 2022-12-23:
To be more precise regarding the Windows `OSError`:

The abovementioned:
```
OSError: [WinError -2147221231] ClassFactory kann angeforderte Klasse nicht liefern
```
was on a Win7 computer (Python 3.8.10, 32 bit) with no Bluetooth hardware.
More details about the WinError:
winerror.h from Windows SDK 10.0.14393.0, CLASS_E_CLASSNOTAVAILABLE, message: ClassFactory cannot supply requested class

On a Win10 computer (Python 3.8.8, 64 bit) with Bluetooth hardware but Bluetooth turned off it raises:
```
OSError: [WinError -2147020577] Das Gerät kann nicht verwendet werden
```
More details about the WinError:
winerror.h from Windows SDK 10.0.14393.0: DEVICE_NOT_AVAILABLE, message: The device is not ready for use.

dlech, 2025-12-16:
We now have `BleakBluetoothNotAvailableError` for this (since Bleak v2.0.0).

closed 2025-12-16 by dlech