hbldh/bleak #1184: App crash at startup

brinata, 2022-12-29:
if I insert the following line in my Android App:

"import bleak"

the app crashes at start up in my mobile, but it works under windows 10.

On windows I'm using the following versions:

Pythpn 3.10.9
bleak            0.19.5
Kivy               2.1.0
buildozer      1.4.0

Inside the builddozer.spec file I haven't specify the version of the packages

Can Someone Help me ?

dlech, 2022-12-29:
Without details about the crash and steps to reproduce the problem, it will be difficult for anyone to provide any help.

dlech, 2022-12-30:
> ModuleNotFoundError: No module named 'async_timeout'

This is a 3rd-party module and likely needs to be added to some depedency list somewhere. Maybe here?

https://github.com/hbldh/bleak/blob/4a6d1c495634f38449180bef113bcdb06a3cf783/bleak/backends/p4android/recipes/bleak/__init__.py#L14

brinata, 2022-12-30:
Maybe, but the Android Log Stream write that there is the following import error:

 "/home/ronni/app/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/myapp/arm64-v8a/bleak/__init__.py", line 31, in <module>
**12-30 11:14:38.496 16697 16734 I python  :  ModuleNotFoundError: No module named 'async_timeout**'

at the line 31 of the file __init__.py , there is the followinfg code:

 sys.version_info < (3, 11):
    from **async_timeout** import timeout as async_timeout
else:
    from asyncio import timeout as async_timeout

it seems that the script recognize a python version older that 3.11 and so it import the timeout library drom the external async_timeout package instead of from the asyncio .

I think that Android doesn't find the "async_timeout" lib to import. On the contrary, on Windows, where the this library is installed as dependecy of the bleak package, the python finds the lib to import.

do I have to declare something special inside my buildozer.spec file ?

dlech, 2022-12-30:
I don't know much about how buildozer works. The line of code I linked to above was my best guess.

brinata, 2022-12-30:
I solver the problem adding "async-timeout" in the requirements( the lib name was different from label "async_timeout" used in the import declaration ...)

Now the app starts, but when I try to scan for ble devices, i give back an exception. It desn'pt find the class "com.github.hbldh.bleak.PythonScanCallback

12-30 19:20:36.594  6669  6728 I python  :  Task exception was never retrieved
12-30 19:20:36.595  6669  6728 I python  :  future: <Task finished name='Task-2' coro=<MainApp.avvia_scansione() done, defined at /home/ronni/progetti/app_antibrina/.buildozer/android/app/main.py:44> exception=JavaException('JVM exception occurred: Didn\'t find class "com.github.hbldh.bleak.PythonScanCallback" on path: DexPathList[[zip file "/data/app/-vXBgDmvIyucUqS1PalQ1A==/org.test.myapp-YSxHHXnrmEfi1WXd1BB06A==/base.apk"],nativeLibraryDirectories=[/data/app/~~-vXBgDmvIyucUqS1PalQ1A==/org.test.myapp-YSxHHXnrmEfi1WXd1BB06A==/lib/arm64, /data/app/~~-vXBgDmvIyucUqS1PalQ1A==/org.test.myapp-YSxHHXnrmEfi1WXd1BB06A==/base.apk!/lib/arm64-v8a, /system/lib64, /system_ext/lib64]] java.lang.ClassNotFoundException')>
12-30 19:20:36.595  6669  6728 I python  :  Traceback (most recent call last):
12-30 19:20:36.595  6669  6728 I python  :    File "/home/ronni/progetti/app_antibrina/.buildozer/android/app/main.py", line 46, in avvia_scansione
12-30 19:20:36.595  6669  6728 I python  :    File "/home/ronni/progetti/app_antibrina/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/myapp/arm64-v8a/bleak/__init__.py", line 216, in discover
12-30 19:20:36.595  6669  6728 I python  :    File "/home/ronni/progetti/app_antibrina/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/myapp/arm64-v8a/bleak/__init__.py", line 118, in __init__
12-30 19:20:36.595  6669  6728 I python  :    File "/home/ronni/progetti/app_antibrina/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/myapp/arm64-v8a/bleak/backends/scanner.py", line 240, in get_platform_scanner_backend_type
12-30 19:20:36.595  6669  6728 I python  :    File "/home/ronni/progetti/app_antibrina/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/myapp/arm64-v8a/bleak/backends/p4android/scanner.py", line 21, in <module>
12-30 19:20:36.595  6669  6728 I python  :    File "/home/ronni/progetti/app_antibrina/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/myapp/arm64-v8a/bleak/backends/p4android/defs.py", line 31, in <module>
12-30 19:20:36.595  6669  6728 I python  :    File "/home/ronni/progetti/app_antibrina/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/myapp/arm64-v8a/jnius/reflect.py", line 211, in autoclass
12-30 19:20:36.595  6669  6728 I python  :    File "jnius/jnius_export_func.pxi", line 26, in jnius.jnius.find_javaclass
12-30 19:20:36.595  6669  6728 I python  :    File "jnius/jnius_utils.pxi", line 91, in jnius.jnius.check_exception
12-30 19:20:36.596  6669  6728 I python  :  jnius.jnius.JavaException: JVM exception occurred: Didn't find class "com.github.hbldh.bleak.PythonScanCallback" on path: DexPathList[[zip file "/data/app/~~ vXBgDmvIyucUqS1PalQ1A==/org.test.myapp-YSxHHXnrmEfi1WXd1BB06A==/base.apk"],nativeLibraryDirectories=[/data/app/~~-vXBgDmvIyucUqS1PalQ1A==/org.test.myapp-YSxHHXnrmEfi1WXd1BB06A==/lib/arm64, /data/app/~~-vXBgDmvIyucUqS1PalQ1A==/org.test.myapp-YSxHHXnrmEfi1WXd1BB06A==/base.apk!/lib/arm64-v8a, /system/lib64, /system_ext/lib64]] java.lang.ClassNotFoundException

Can you help me ?

Of course it works On windows.

dlech, 2022-12-30:
> I solver the problem adding "async-timeout" in the requirements( the lib name was different from label "async_timeout" used in the import declaration ...)

Can you make a pull request to fix this?

> Now the app starts, but when I try to scan for ble devices, i give back an exception. It desn'pt find the class "com.github.hbldh.bleak.PythonScanCallback

Sounds like [#756](https://github.com/hbldh/bleak/issues/756)

brinata, 2023-01-03:
Sounds Good !

I've included the java files in the buildozer.spec, using the keywords "android.add_src" and it works for the moment, I've tested only the scan function of the library.

is it possible to make some change to load the java files automatically ??

Prakyathkantharaju, 2023-03-12:
If someone else is facing a similar issue with the example app. Along with these fixes mentioned (This issue and #756).

Specify your python3 version in requirements eg. python3==3.10.6

fenzeldan, 2023-06-30:
I had similar problems with the example app. With those fixes, mentioned above, i was able to run it. But i have a different problem now: 
I get this exeption thrown:
['android.permission.ACCESS_FINE_LOCATION', 'android.permission.ACCESS_COARSE_LOCATION', 'android.permission.ACCESS_BACKGROUND_LOCATION']

The app has no permissions to access the BLE Module of the phone i think. I already tried different fixes but im not able to fix it.

This is the error log from the phone: 
_06-29 22:59:00.070 19163 19209 I python  : [ERROR  ] [Task exception was never retrieved
06-29 22:59:00.070 19163 19209 I python  : future] <Task finished name='Task-2' coro=<MainScreen.search_ble() done, defined at /home/daniel/projects/bleapp1/.buildozer/android/app/main.py:32> exception=BleakError("User denied access to ['android.permission.ACCESS_FINE_LOCATION', 'android.permission.ACCESS_COARSE_LOCATION', 'android.permission.ACCESS_BACKGROUND_LOCATION']")>_

Im new to coding so im sorry if i missed important information you may need to help.
I would be thankful for help and can provide more information if needed

obzerving, 2023-08-22:
It's August 2023 and brinata's solution of adding async-timeout to the requirements line in buildozer still worked for me.

closed 2023-08-29 by dlech