hbldh/bleak #1205: TimeoutError while disconnecting devices using BLE Client for Windows 10.

smsln1, 2023-01-19:
* bleak version: 0.19.5
* Python version: 3.10.9
* Operating System: Windows 10 Enterprise (OS Build 19041.264)

### Description

Hi,

I'm trying to check if I can connect to the BLE device and then disconnect without any issue. Unfortunately, it seems that when devices are about to disconnect, only debug log shows up and the process cannot be finished as bleak just waits for the library specified 120 seconds before the timeout error will be raised. 

### What I Did

This is the script that shows the problem:

```
import argparse
import asyncio
import logging

from bleak import BleakScanner, BleakClient

logger = logging.getLogger(__name__)

async def main(args: argparse.Namespace):
    logger.info("scanning...")

    if args.address:
        device = await BleakScanner.find_device_by_address(
            args.address, cb=dict(use_bdaddr=args.macos_use_bdaddr)
        )
        if device is None:
            logger.error("could not find device with address '%s'", args.address)
            return
    else:
        device = await BleakScanner.find_device_by_filter(
            lambda d, ad: d.name == args.name,
            timeout=10.0,
            cb=dict(use_bdaddr=args.macos_use_bdaddr)
        )
        if device is None:
            logger.error("could not find device with name '%s'", args.name)
            return

    async with BleakClient(device) as client:
        await asyncio.sleep(5)
        logger.info('Disconnecting...')

if __name__ == "__main__":
    parser = argparse.ArgumentParser()

    device_group = parser.add_mutually_exclusive_group(required=True)

    device_group.add_argument(
        "--name",
        metavar="<name>",
        help="the name of the bluetooth device to connect to",
    )
    device_group.add_argument(
        "--address",
        metavar="<address>",
        help="the address of the bluetooth device to connect to",
    )

    parser.add_argument(
        "--macos-use-bdaddr",
        action="store_true",
        help="when true use Bluetooth address instead of UUID on macOS",
    )

    args = parser.parse_args()

    log_level = logging.DEBUG
    logging.basicConfig(
        level=log_level,
        format="%(asctime)-15s %(name)-8s %(levelname)s: %(message)s",
    )

    asyncio.run(main(args))
```

Also, i have tried to increase the delay to 30 seconds after the connection was established but the behavior is still the same. 

### Logs

Logs from the terminal:

```
2023-01-19 14:47:00,857 asyncio  DEBUG: Using proactor: IocpProactor
2023-01-19 14:47:00,857 __main__ INFO: scanning...
2023-01-19 14:47:01,005 bleak.backends.winrt.scanner DEBUG: Received 0A:65:55:F9:9D:D2: .
2023-01-19 14:47:01,005 bleak.backends.winrt.scanner DEBUG: Received 1A:91:60:12:2D:0C: .               
2023-01-19 14:47:01,005 bleak.backends.winrt.scanner DEBUG: Received 1B:92:14:96:01:2B: .               
2023-01-19 14:47:01,005 bleak.backends.winrt.scanner DEBUG: Received 65:F1:4D:B5:59:A9: .               
2023-01-19 14:47:01,005 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2023-01-19 14:47:01,005 bleak.backends.winrt.scanner DEBUG: Received 65:F1:4D:B5:59:A9: .               
2023-01-19 14:47:01,005 bleak.backends.winrt.scanner DEBUG: Received 2F:B1:92:76:7E:CF: .               
2023-01-19 14:47:01,005 bleak.backends.winrt.scanner DEBUG: Received 5A:25:8F:57:5A:2F: .               
2023-01-19 14:47:01,005 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2023-01-19 14:47:01,021 bleak.backends.winrt.scanner DEBUG: Received 01:67:39:93:3F:64: .               
2023-01-19 14:47:01,027 bleak.backends.winrt.scanner DEBUG: Received 7B:AF:23:11:1C:32: .               
2023-01-19 14:47:01,027 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2023-01-19 14:47:01,027 bleak.backends.winrt.scanner DEBUG: Received 24:12:90:7F:C4:4F: .               
2023-01-19 14:47:01,127 bleak.backends.winrt.scanner DEBUG: Received 2F:B1:92:76:7E:CF: .
2023-01-19 14:47:01,127 bleak.backends.winrt.scanner DEBUG: Received 01:67:39:93:3F:64: .
2023-01-19 14:47:01,127 bleak.backends.winrt.scanner DEBUG: Received 36:85:E0:F0:74:41: .
2023-01-19 14:47:01,127 bleak.backends.winrt.scanner DEBUG: Received 26:89:7F:1D:77:43: .
2023-01-19 14:47:01,142 bleak.backends.winrt.scanner DEBUG: Received C0:00:00:00:00:00: .               
2023-01-19 14:47:01,142 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2023-01-19 14:47:01,142 bleak.backends.winrt.scanner DEBUG: Received C0:00:00:00:00:00: App0 Server.    
2023-01-19 14:47:01,142 bleak.backends.winrt.scanner DEBUG: Received 7E:A0:B0:E8:91:29: .               
2023-01-19 14:47:01,142 bleak.backends.winrt.scanner DEBUG: 11 devices found. Watcher status: 3.        
2023-01-19 14:47:01,158 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ C0:00:00:00:00:00
2023-01-19 14:47:01,173 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2023-01-19 14:47:01,343 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x0000019330E0C610>, error: BluetoothError.SUCCESS, status: GattSe
ssionStatus.ACTIVE
2023-01-19 14:47:01,459 bleak.backends.winrt.client DEBUG: max_pdu_size_changed_handler: 517
2023-01-19 14:47:02,592 bleak.backends.winrt.client DEBUG: C0:00:00:00:00:00: services changed
2023-01-19 14:47:02,592 bleak.backends.winrt.client DEBUG: C0:00:00:00:00:00: restarting get services due to services changed event
2023-01-19 14:47:02,624 bleak.backends.winrt.client DEBUG: C0:00:00:00:00:00: services changed
2023-01-19 14:47:02,646 bleak.backends.winrt.client DEBUG: C0:00:00:00:00:00: services changed
2023-01-19 14:47:02,909 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
2023-01-19 14:47:02,909 bleak.backends.winrt.client DEBUG: C0:00:00:00:00:00: restarting get services due to services changed event
2023-01-19 14:47:02,931 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
2023-01-19 14:47:08,506 __main__ INFO: Disconnecting...
2023-01-19 14:47:08,506 bleak.backends.winrt.client DEBUG: Disconnecting from BLE device...
Traceback (most recent call last):
  File "...\bleak\backends\winrt\client.py", line 485, in disconnect
    await event.wait()
  File "...\Python310\lib\asyncio\locks.py", line 214, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File ...\bleak_disconnect_issue.py", line 69, in <module>
    asyncio.run(main(args))
  File "...\Python310\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "...\Python310\lib\asyncio\base_events.py", line 649, in run_until_complete
    return future.result()
  File ...\bleak_disconnect_issue.py", line 33, in main
    async with BleakClient(device) as client:
  File "...\bleak\__init__.py", line 437, in __aexit__
    await self.disconnect()
  File "...\bleak\__init__.py", line 480, in disconnect
    return await self._backend.disconnect()
  File "...\bleak\backends\winrt\client.py", line 484, in disconnect
    async with async_timeout.timeout(120):
  File "...\async_timeout\__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "...\async_timeout\__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

dlech, 2023-01-19:
I have one computer where it takes anywhere from 1 to 2 minutes for a device to disconnect. I made support request to Microsoft to try to figure it out, but they haven't been able to resolve it yet.

If you change the timeout to longer, does it eventually disconnect? You could also try something like [Bluetooth LE Explorer](https://apps.microsoft.com/store/detail/bluetooth-le-explorer/9N0ZTKF1QD98?hl=en-us&gl=us) or `chrome://bluetooth-internals/#devices` in a Chrome browser to confirm this is an OS issue rather than a Bleak issue (connect a device then disconnect it and use a stopwatch to time how long it takes to disconnect).

EtteTomi, 2023-01-23:
I had similar issues with bleak version 0.19.5, as I was unable to disconnect the device. I switched to bleak version 0.19.0, and it seemed to have fixed the issue.

bconrad98, 2023-02-09:
Running on Windows 10 and it takes me around 30 seconds to disconnect. No errors though. May try on Linux to see how it changes.

dlech, 2023-02-19:
> I had similar issues with bleak version 0.19.5, as I was unable to disconnect the device. I switched to bleak version 0.19.0, and it seemed to have fixed the issue.

Can you try with the latest `develop` branch to make sure it will be fixed in the next release as well?

dlech, 2023-03-17:
This should be fixed in the `develop` branch. If anyone has a similar problem and does *not* see `restarting get services due to services changed event` in the debug logs, please start a new issue. If you do see that message, upgrade to a newer version of Bleak.

closed 2023-03-17 by dlech