hbldh/bleak #1207: Random disconnects

KelvinSan, 2023-01-28:
I connected to bluetooth device and tried to read GATT characteristics on an interval basis, it works for about 30 seconds then I get an error about Operation Aborted, 

Code:
from bleak import BleakClient,BleakScanner
import asyncio
import uuid
import struct

async def read_characters(client:BleakClient):
    data = await client.read_gatt_char(uuid.UUID(hex="DB450003-8E9A-4818-ADD7-6ED94A328AB4"))
    data2 = await client.read_gatt_char(uuid.UUID(hex="DB450002-8E9A-4818-ADD7-6ED94A328AB4"))
    # data3 = await client.read_gatt_char(uuid.UUID(hex="DB450004-8E9A-4818-ADD7-6ED94A328AB4"))
    data4 = await client.read_gatt_char(uuid.UUID(hex="DB450005-8E9A-4818-ADD7-6ED94A328AB4"))
    # data5 = await client.read_gatt_char(uuid.UUID(hex="DB450001-8E9A-4818-ADD7-6ED94A328AB4"))
    # print(int.from_bytes(voc,byteorder='big'))
    bme280_data_hex = data.hex()
    pmdata = data4.hex()
    print(bme280_data_hex)
    print(pmdata)
    print(data2.hex())
    humidity = int(bme280_data_hex[:2],16)
    temp = int(bme280_data_hex[2:4],16)
    pressure = int.from_bytes(bytes.fromhex(bme280_data_hex[4:12]), byteorder="little") / 100
    pm1 = int.from_bytes(bytes.fromhex(pmdata[:6]), byteorder="little") / 100
    pm2_5 = int.from_bytes(bytes.fromhex(pmdata[6:12]), byteorder="little") / 100
    pm10 = int.from_bytes(bytes.fromhex(pmdata[12:18]), byteorder="little") / 100
    pm4 = int.from_bytes(bytes.fromhex(pmdata[18:24]), byteorder="little") / 100
    voc = int.from_bytes(bytes.fromhex(data2.hex()[:4]), byteorder="little") / 1000
    print(humidity)
    print(temp)
    print(pressure)
    print(pm1)
    print(pm2_5)
    print(pm10)
    print(pm4)
    print(voc)

async def on_disconnect_async(cli:BleakClient):
    await cli.connect(timeout=15)
    asyncio.sleep(1)

def reconnect(c: BleakClient):
    print("Reconnect")
    asyncio.get_event_loop().call_soon(asyncio.ensure_future,on_disconnect_async(c))

async def decode_data():
    
        device_mac_address = "D7:A7:F7:C6:0B:0E"
        async with BleakClient(device_mac_address,timeout=15, disconnected_callback=reconnect) as client:
            while True:
                try:
                    await read_characters(client=client)
                    await asyncio.sleep(1)
                except OSError as error:
                    print(error)
                    continue
if __name__ == "__main__":
    asyncio.run(decode_data())

Logs:
2023-01-28 07:06:27,613 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ D7:A7:F7:C6:0B:0E
2023-01-28 07:06:27,642 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2023-01-28 07:06:28,122 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000001E9FC27A130>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
2023-01-28 07:06:28,144 bleak.backends.winrt.client DEBUG: max_pdu_size_changed_handler: 247
2023-01-28 07:06:28,891 bleak.backends.winrt.client DEBUG: D7:A7:F7:C6:0B:0E: services changed
2023-01-28 07:06:28,892 bleak.backends.winrt.client DEBUG: D7:A7:F7:C6:0B:0E: restarting get services due to services changed event
2023-01-28 07:06:28,930 bleak.backends.winrt.client DEBUG: D7:A7:F7:C6:0B:0E: services changed
2023-01-28 07:06:28,971 bleak.backends.winrt.client DEBUG: D7:A7:F7:C6:0B:0E: services changed
2023-01-28 07:06:29,013 bleak.backends.winrt.client DEBUG: D7:A7:F7:C6:0B:0E: services changed
2023-01-28 07:06:29,051 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
[... 374 lines cut ...]
987.9
1.0
1.97
2.94
1.0
0.0
2023-01-28 07:07:07,144 bleak.backends.winrt.client DEBUG: Read Characteristic 000F : bytearray(b'\x1b\x17\xe6\x81\x01\x00\xde\x08')
2023-01-28 07:07:07,264 bleak.backends.winrt.client DEBUG: Read Characteristic 000C : bytearray(b'\x00\x00\x00\x00')
2023-01-28 07:07:07,385 bleak.backends.winrt.client DEBUG: Read Characteristic 0015 : bytearray(b'd\x00\x00\xc5\x00\x00&\x01\x00d\x00\x00')
1b17e6810100de08
640000c50000260100640000
00000000
27
23
987.9
1.0
1.97
2.94
1.0
0.0
2023-01-28 07:07:17,766 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000001E9FC27A290>, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED

Error:
[WinError -2147467260] Operation aborted
2023-01-28 07:07:17,774 bleak.backends.winrt.client DEBUG: Disconnecting from BLE device...

closed 2023-01-28 by KelvinSan

852432070, 2023-03-07:
Hi,  @KelvinSan . Does this issue solved? I encountered nearly the same question as what you did.

KelvinSan, 2023-03-08:
> Hi,  @KelvinSan . Does this issue solved? I encountered nearly the same question as what you did.

Nope just gave up unfortunately, and realized that Windows isn't good for Bluetooth

852432070, 2023-03-08:
> > Hi,  @KelvinSan . Does this issue solved? I encountered nearly the same question as what you did.
> 
> Nope just gave up unfortunately, and realized that Windows isn't good for Bluetooth

Fine, I think I should consider to change the platform now. Thanks for your reply.