hbldh/bleak #1214: start_notify does not deactivate previous notification callback with 0.19.5 and WinRT

eudoxos, 2023-02-03:
* bleak version: 0.19.5
* Python version: 3.9.x
* Operating System: Windows
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

We have some (in-house, no MWE yet) code which calls `start_notify` on a single characteristic twice, assuming the old callback will be overwritten (deactivate) by the second one. This worked for both Linux and Windows with 0.12.1 (that's the platforms we use and test on), and for Linux with 0.19.5 — the old (first) callback is deactivated when `start_notify` is called the second time, and the new callback is used since then.

With Windows and Bleak 0.19.5, what we (reproducibly) observe instead is that after second `start_notify`, *both* callbacks get called upon notification from the characteristic.

### What I Did

The workaround we use is to call `stop_notify` before every `start_notify`, catching `KeyError` from Bleak if there is no callback yet. This resolves the issue of double-callback.

This is either documentation issue (whether a new callback deactivates the old one, the documentation seems to assume there is at most one callback per characteristic) or portability issue (same API behaves differently on different platforms).

Please let me know if it is worth trying to craft a MWE — perhaps I am hitting at something known which I overlooked in the docs.

Thanks!

dlech, 2023-02-03:
Why do you need to call `start_notify` twice instead of once?

I think it would make sense that Bleak should always raise an exception if it is called a second time without stopping first to avoid the unknown/undefined behavior of each backend.

eudoxos, 2023-02-03:
> Why do you need to call start_notify twice instead of once?

The calls are not immediately after one another. It is a measurement device, the arriving data need different handling depending on other (interactive) settings.

> I think it would make sense that Bleak should always raise an exception if it is called a second time without stopping first to avoid the unknown/undefined behavior of each backend.

I would prefer to silently discard any existing callback (which clearly already happens with Linux, and did in Windows in 0.12). Raising an exception has its pros (explicit is better than implicit), though it should be a very specific exception type which can be caught, without catching other Bleak exceptions.

dlech, 2026-03-08:
If you don't want to do the actual over-the-air I/O of stopping and starting notifications again, I would just make one handler that works for both cases. I.e. the common handler pushes data to a queue and the handler1 and handler2 take turns subscribing to that queue.