hbldh/bleak #1217: DeviceNotFound & Could not get GATT services "Unreachable"

ztroop, 2023-02-09:
* bleak version: 0.19.5
* Operating System: Windows

### Description

I maintain a [library](https://github.com/ztroop/wave-reader-utils) that uses `bleak` as a dependency and I have a Windows user that is experiencing failures with connectivity. After spending time examining `bleak`  code, I don't see any smoking guns.
- User is physically near the BTLE device.
- User encounters `DeviceNotFound` **OR** it connects, hangs, and then errors out with:

```
Traceback (most recent call last):
  File "C:\Users\ronro\OneDrive\Desktop\Github\PIPHI-CLI-v2\venv\lib\site-packages\bleak\backends\winrt\client.py", line 643, in get_services
    services: Sequence[GattDeviceService] = _ensure_success(
  File "C:\Users\ronro\OneDrive\Desktop\Github\PIPHI-CLI-v2\venv\lib\site-packages\bleak\backends\winrt\client.py", line 120, in _ensure_success
    raise BleakError(f"{fail_msg}: Unreachable")
bleak.exc.BleakError: Could not get GATT services: Unreachable
```

Any suggestions/comments would be appreciated!

dlech, 2023-02-09:
"Unreachable" comes from the Windows Bluetooth stack, so Bluetooth packet capture would be the best way to investigate that.

There are many issues that mention this error that might provide more insight as well: https://github.com/search?q=repo%3Ahbldh%2Fbleak+Unreachable&type=issues

dlech, 2023-02-19:
I found an old issue #60 that suggests that a retry may help with this particular error. Can you add a retry to your code and see if it helps? If it does, we can look at adding an automatic retry in Bleak.

ztroop, 2023-02-19:
I'll recommend that action once the user responds. As of this moment, I'm still waiting for a packet capture.

anikseng, 2024-09-11:
I was able to connect to the BLE device once and after that it gives me this error "bleak.exc.BleakError: Could not get GATT services: Unreachable". I tried every possible solution but still the error is there.

joeygold, 2024-09-11:
Try one of the silicon vendor  BLE App's to confirm you can connect with
the BLE device.

The connection with one of these App should be able to identify the
services and characteristics....  you might have a digit wrong in your
service or characteristic.

dlech, 2024-09-11:
IIRC, the "Unreachable" error usually happens because the Windows Bluetooth stack is waiting for a response for something that never comes from the device. The only way to find the source of the problem is to log Bluetooth packets as described in the troubleshooting page of the Bleak docs.