hbldh/bleak #1220: Automatic Retry Logic During Connection

ztroop, 2023-02-14:
* bleak version: **0.19.5**
* Python version: **3.10**
* Operating System: **Ubuntu 22.04**
* BlueZ version (`bluetoothctl -v`) in case of Linux: **5.64**

### Description

It might be worthwhile to have a handler in the library that allows retrying. For example, on rare occasion I may get a `bleak.exc.BleakDBusError: [org.bluez.Error.Failed] le-connection-abort-by-local` exception. Re-connecting will work. This is a DBus specific problem, I understand... However, it would be ideal to have some sort of `retry` connection logic baked in.

dlech, 2023-02-14:
I think this is a good idea. There have been quite a few reports lately of getting an `Unreachable` error on Windows lately. This lead me to #60 which suggests a retry could be helpful on Windows as well. But since these are very backend-specific, we should probably consider them separate issues.

So rather than a general retry mechanism, I think we should handle these specific errors as needed.

ztroop, 2023-02-15:
I'd been keen to help with the effort. I think tracking and handling "automatic retries" separately in each backend is probably a good idea. I would suggest that this ticket is specific to the `bluezdbus`.

dlech, 2023-02-15:
> I'd been keen to help with the effort.

Does this mean you will make a pull request?

ztroop, 2023-02-15:
> > I'd been keen to help with the effort.
> 
> Does this mean you will make a pull request?

Yeah, I'll put something together when I have the time to do so.

PR #1257 (2023-03-17): Release/v0.20.0

Added
-----
* Added ``BLEAK_DBUS_AUTH_UID`` environment variable for hardcoding D-Bus UID. Merged #1182.
* Added return type ``None`` to some scanner methods.
* Added optional hack to use Bluetooth address instead of UUID on macOS. Merged #1073.
* Added ``BleakScanner.find_device_by_name()`` class method.
* Added optional command line argument to use debug log level to all applicable examples.
* Added ``bleak.uuids.normalize_uuid_str()`` function.
* Added optional ``services`` argument to ``BleakClient()`` to filter services. Merged #654.
* Added automatic retry on ``le-connec [... cut]
closed 2023-03-17 by dlech