hbldh/bleak #1223: client.disconnect() hangs on Win 10

zapta, 2023-02-16:
* bleak version:  bleak==0.19.5   bleak-winrt==1.2.0
* Python version: Python 3.10.7
* Operating System: Win 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:
* 
OS: Windows-10-10.0.19044-SP0
Platform:: uname_result(system='Windows', node='Ryzen', release='10', version='10.0.19044', machine='AMD64')
Python 3.10.7 (tags/v3.10.7:6cc6b13, Sep  5 2022, 14:08:36) [MSC v.1933 64 bit (AMD64)]

### Description

I encountered a problem in my app, closing a connection, so reproduced it with the small test program below. When the program tries to close a BLE connection, it hangs (I had to abort with Ctl-c in the run that was used for the log and packet capture below). Not sure if it's a bleak bug or a problem in my python program or BLE device. The devices is similar to the sample ESP32 device marked '2', I emailed last week.

### What I Did

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well.

### Test program:
```
import asyncio
import platform
import sys
import atexit
import time
import signal
from bleak import BleakClient, BleakScanner
import signal

# Adapt to your actual device.
device_address = "0C:8B:95:F2:B4:36"

signal.signal(signal.SIGINT, lambda number, frame: sys.exit())

print(f"OS: {platform.platform()}", flush=True)
print(f"Platform:: {platform.uname()}", flush=True)
print(f"Python {sys.version}", flush=True)

async def test():
    #global client
    print(f"Trying to connect to {device_address}", flush=True)
    device = await BleakScanner.find_device_by_address(device_address, timeout=10.0)
    assert device
    client = BleakClient(device)
    assert client
    await client.connect(timeout=10.0)
    assert client.is_connected
    print(f"Connected", flush=True)
    print(f"Waiting 5 secs...", flush=True)
    time.sleep(5.0)
    print(f"Closing...", flush=True)
    assert client.is_connected
    await client.disconnect()
    print(f"Test done", flush=True)

event_loop = asyncio.new_event_loop()
event_loop.run_until_complete(test())
```

### Logs

```
/c/projects/ble_stepper_motor_analyzer/repo/app/test$ python connection_close.py
OS: Windows-10-10.0.19044-SP0
Platform:: uname_result(system='Windows', node='Ryzen', release='10', version='10.0.19044', machine='AMD64')
Python 3.10.7 (tags/v3.10.7:6cc6b13, Sep  5 2022, 14:08:36) [MSC v.1933 64 bit (AMD64)]
Trying to connect to 0C:8B:95:F2:B4:36
2023-02-16 12:57:24,597 bleak.backends.winrt.scanner DEBUG: Received 0C:8B:95:F2:B4:36: STP-0C8B95F2B436.
2023-02-16 12:57:24,598 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2023-02-16 12:57:24,600 bleak.backends.winrt.scanner DEBUG: Received 0C:8B:95:F2:B4:36: .
2023-02-16 12:57:24,603 bleak.backends.winrt.scanner DEBUG: 1 devices found. Watcher status: 3.
2023-02-16 12:57:24,620 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ 0C:8B:95:F2:B4:36
2023-02-16 12:57:24,657 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2023-02-16 12:57:24,816 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x0000019AB567A510>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
2023-02-16 12:57:24,891 bleak.backends.winrt.client DEBUG: max_pdu_size_changed_handler: 247
2023-02-16 12:57:25,708 bleak.backends.winrt.client DEBUG: 0C:8B:95:F2:B4:36: services changed
2023-02-16 12:57:25,709 bleak.backends.winrt.client DEBUG: 0C:8B:95:F2:B4:36: restarting get services due to services changed event
2023-02-16 12:57:25,764 bleak.backends.winrt.client DEBUG: 0C:8B:95:F2:B4:36: services changed
2023-02-16 12:57:25,778 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
2023-02-16 12:57:25,778 bleak.backends.winrt.client DEBUG: 0C:8B:95:F2:B4:36: restarting get services due to services changed event
2023-02-16 12:57:25,808 bleak.backends.winrt.client DEBUG: 0C:8B:95:F2:B4:36: services changed
2023-02-16 12:57:25,861 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
2023-02-16 12:57:25,861 bleak.backends.winrt.client DEBUG: 0C:8B:95:F2:B4:36: restarting get services due to services changed event
2023-02-16 12:57:25,911 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=<BluetoothCacheMode.CACHED: 0>, cache_mode=None)...
Connected
Waiting 5 secs...
Closing...
2023-02-16 12:57:31,427 bleak.backends.winrt.client DEBUG: Disconnecting from BLE device...
/c/projects/ble_stepper_motor_analyzer/repo/app/test$

```

Wireshark file:

[bleak_disconnec_hangs.zip](https://github.com/hbldh/bleak/files/10760751/bleak_disconnec_hangs.zip)

zapta, 2023-02-16:
One additional data point, I tested the same on ubuntu and the connection is closed cleanly.

dlech, 2023-02-16:
I would try rebooting windows and see if that fixes it. If it persists, it is probably a Windows driver issue. You can run the following in powershell to get the Bluetooth adapter info.

    wget https://github.com/Microsoft/busiotools/raw/master/bluetooth/tracing/GetBluetoothRadioInfo.ps1 -UseBasicParsing | iex

If you can share that, it would help us make a list of known working and not working adapters.

zapta, 2023-02-17:
System reset didn't help. Here is the log

```
InstanceId        : USB\VID_8087&PID_0029\6&195F3125&0&2
MAC               : DC41A9D9332E
DriverDescription : Intel(R) Wireless Bluetooth(R)
DriverVersion     : 21.50.1.1
ErrorRecovery     : None
ScoType           : Unknown
InstanceId        : USB\VID_8087&PID_0A2A\6&195F3125&0&2
MAC               : 70CF499EC392
DriverDescription : Intel(R) Wireless Bluetooth(R)
DriverVersion     : 20.100.10.7
ErrorRecovery     : None
ScoType           : Unknown
```

I consider adding to my BLE device connection WDT capabilities. E..g the host need to keep sending app level heartbeats, otherwise the device disconnect.

mark-blackburn-orpyx, 2023-03-16:
I'm seeing this issue too with our device with v0.19.5. I don't see it with 0.19.4.

If you need me to run any testing, add logging, etc so I can help out, let me know.

dlech, 2023-03-16:
This should be fixed in the `develop` branch already.

mark-blackburn-orpyx, 2023-03-16:
Just tried the develop branch (138697f). This is fixed for me. Thanks.

Anything I can do to help with the next release?

dlech, 2023-03-17:
Closing since this is confirmed fixed in the `develop` branch.

closed 2023-03-17 by dlech

kins3i, 2024-09-30:
Hello. The same error persists with Windows 10. I have Python 3.12.4 with Bleak library 0.22.2.

Here is my driver:
[image]

Process hangs on disconnecting and catches timeout exception (in winrt\client function disconnect). It does disconnected device as my device has LED indication of connecting status. Here are logs from disconnect function:
```
Disconnecting device,  2024-09-30 13:37:44.798222
2024-09-30 13:37:44,798 bleak.backends.winrt.client Thread-7 (one_disconnect) DEBUG: Disconnecting from BLE device...
2024-09-30 13:37:47,957 bleak.backends.winrt.client Dummy-8 DEBUG: max_pdu_size_changed_handler: 23
2024-09-30 13:37:47,959 bleak.backends.winrt.client Dummy-3 DEBUG: session_status_changed_event_handler: id: BluetoothLE#BluetoothLE04:7f:0e:4b:7d:69-00:80:e1:26:bb:da, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.CLOSED: 0>
Error disconnecting to device,  

```
Last print should be ended with exception code, but it's empty.

Here is my function (which is called as asynchronous function adn asynchronous function is called in different thread):
```
async def single_disconnect(self, address):
        single_client = None
        device_class = None
        for client in self.client_list:
            if client.address == address:
                single_client = client
        if single_client is not None:
            try:
                if single_client.is_connected:
                    for device in self.client_device_list:
                        device_address = device.MAC_ST
                        if device_address == single_client.address:
                            device_class = device
                    if device_class is not None:
                        device_class.stop_all_notifies()
                        print("Disconnecting device, ", datetime.now())
                        for x in self.client_list:
                            if x == single_client:
                                self.client_list.remove(x)
                        for device in self.client_device_list:
                            if device.MAC_ST == single_client.address:
                                self.client_device_list.remove(device)
                        bool_var = await single_client.disconnect()
                        while single_client.is_connected:
                            continue
                        print("disconnected")

                        print('Disconnecting single client: ' + str(address))
                else:
                    for x in self.client_list:
                        if x == single_client:
                            self.client_list.remove(x)
                    for device in self.client_device_list:
                        if device.MAC_ST == single_client.address:
                            self.client_device_list.remove(device)
                    print("Already disconnected: ", address)
            except (ValueError, Exception) as e:
                print('Error disconnecting to device, ', e)
```
I tried changing code in library here:
```
        if self._requester:
            event = asyncio.Event()
            self._session_closed_events.append(event)
            try:
                self._requester.close()
                # sometimes it can take over one minute before Windows decides
                # to end the GATT session/disconnect the device
                async with async_timeout(120):
                    await event.wait()
            finally:
                self._session_closed_events.remove(event)
```
but unfortunately it didn't help. I tried adding async.wait and event.set(). With debugging I can see that every boolean after some time (about 30 seconds) shows connected status as disconnected, so I don't know why event .set() is not invoked.

Every help would be highly appreciated

PS I have too high version of Python for installing winrt library if that's important

dlech, 2024-09-30:
This issue is closed. Could you please start a new issue with a full minimal reproducible test case and include as much logging as possible? See troubleshooting page in the docs for how to get debug logs and also log Bluetooth packets.