hbldh/bleak #1224: Need Server example

tljstewart, 2023-02-16:
I've read you can create a custom service and characteristic using `bleak` and a laptop, but there are no examples for this. Is it possible to make a custom service to send data from a laptop?

dlech, 2023-02-16:
Bleak is GATT client only. [Bless](https://github.com/kevincar/bless) is a similar project for GATT server.

closed 2023-02-16 by dlech

tljstewart, 2023-02-16:
Ooo this looks interesting, [Bless](https://github.com/kevincar/bless) your ❤️ !

tljstewart, 2023-02-16:
`pyobjc-core<9.0.0,>=8.5.1`

tljstewart, 2023-02-16:
Interestingly, I am able to run the server and discover it with `nrf connect` app, however, `bleak` (running on same laptop) is unable to discover it... Any ideas why this might be?

dlech, 2023-02-16:
If the server and client on running on the same machine, I doubt they could see each other (unless you are using two separate Bluetooth adapters which is only possible on Linux).

tljstewart, 2023-02-16:
running  M1 `macOS` same adapter, about to try another laptop, intel `macOS`. That's interesting though, I wouldn't intuitively expect that to be a limiting factor

dlech, 2023-02-16:
Most radios can't transmit and receive at the same time unless they have separate antennas for tx and rx and they are physically separated enough. Bluetooth radios that I have seen don't have separate tx and rx. So it would be impossible for the same radio to receive data that is is transmitting.

tljstewart, 2023-02-17:
Now that totally makes sense, time slicing RX/TX! I Can add more evidence this must be the case, the service is found on all other devices, except same device providing the BLE service. I could imagine this would be a common and frustrating "Gotcha" moment starting dev with `bleak`. Thanks for your guidance!