hbldh/bleak #1226: Program works on windows but not on linux

zapta, 2023-02-17:
bleak version: 0.19.5
Python version: 3.10.6
Operating System: Ubuntu 22.04.1 LTS
BlueZ version (bluetoothctl -v) in case of Linux: bluetoothctl: 5.64

### Description

I have a test program that works on windows but fails on linux. The program writes two bytes to a characteristic, but the device returns and error to the write from linux and returns OK to the write from winodws. Comparing the write packets between the two shows a difference which likely causes to the failure with linux.

[image]

I don't know what that flag is but it seems to specify a different kind of request.

### What I Did

Test program
```
import asyncio
import platform
import sys
import time
import signal
from bleak import BleakClient, BleakScanner
import signal

# Adapt to your actual device.
device_address = "0C:8B:95:F2:B4:36"

signal.signal(signal.SIGINT, lambda number, frame: sys.exit())

print(f"OS: {platform.platform()}", flush=True)
print(f"Platform:: {platform.uname()}", flush=True)
print(f"Python {sys.version}", flush=True)

async def test():
    #global client
    print(f"Trying to connect to {device_address}", flush=True)
    device = await BleakScanner.find_device_by_address(device_address, timeout=10.0)
    assert device
    async with BleakClient(device) as client:
        assert client.is_connected
        print(f"Connected", flush=True)
        print(f"Waiting 5 secs...", flush=True)
        time.sleep(5.0)
        print(f"Closing...", flush=True)
        assert client.is_connected
        await client.disconnect()
        print(f"Test done", flush=True)

asyncio.run(test())
```

### Logs and packets

[linux_log.txt](https://github.com/hbldh/bleak/files/10769020/linux_log.txt)
[windows_log.txt](https://github.com/hbldh/bleak/files/10769021/windows_log.txt)
[linux_packets.zip](https://github.com/hbldh/bleak/files/10769022/linux_packets.zip)
[windows_packets.zip](https://github.com/hbldh/bleak/files/10769023/windows_packets.zip)

dlech, 2023-02-17:
[image]

According to the packet capture the "ff06" characteristic only support Write and not Write Without Response, so you need to add `response=True` to the call to `write_gatt_char()`.

zapta, 2023-02-18:
Thanks @dlech, I changed the device firmware to declare the characteristic as ESP_GATT_CHAR_PROP_BIT_WRITE_NR instead of ESP_GATT_CHAR_PROP_BIT_WRITE and it works now, so I am good.

You may want to look why the windows and linux version have different behaviors but this may be in the underlying stacks. Linux does consider the declaration of the charateristic and windows doesn't.

dlech, 2023-02-19:
I'm going to go ahead and close this one since it is working now and we have #909 to improve the default situation.

closed 2023-02-19 by dlech