hbldh/bleak #1238: Bleak connects to device only if windows BLE search is active.

purpl3F0x, 2023-02-24:
* bleak version:
* Python version: 3.11
* Operating System: Windows 11 21H2 (22000.1574)

### Description

When i try to connect to a device bleak doesn't to it unless I manually start scanning for BLE devices on the windows settings page.
Furthers, when i enable debug log, it seems bleak does indeed discover the device and connects to it, but it stucks on getting it;s services. Noting, that i tested the code on other windows machine and works fine.

By windows discovery I mean, going to Bluetooth and devices > Add device > Bluetooth.
### What I Did

```python
async def main():
    async with BleakClient("XX:XX:XX:XX:XX:XX") as client:
        logger.info("Connected to ", client.address)
```

### Logs

```
2023-02-24 11:10:15,644 asyncio  DEBUG: Using proactor: IocpProactor
2023-02-24 11:10:15,931 bleak.backends.winrt.scanner DEBUG: Received C3:44:6F:5A:65:B8: .
2023-02-24 11:10:15,932 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2023-02-24 11:10:15,936 bleak.backends.winrt.scanner DEBUG: Received C3:44:6F:5A:65:B8: Enviromental Sensor 65B8.
2023-02-24 11:10:15,942 bleak.backends.winrt.scanner DEBUG: 1 devices found. Watcher status: 3.
2023-02-24 11:10:15,943 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ C3:44:6F:5A:65:B8
2023-02-24 11:10:16,235 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
```

dlech, 2023-02-24:
I have one Bluetooth radio that doesn't always return results when scanning. Using the Windows Settings to enable scanning is one workaround but, I found that if I just stop and start scanning again in the Python script, the it works the second time around.

For reference, this is the problematic chip/driver combo for me:

```
InstanceId        : USB\VID_1286&PID_204C\0000000000000000
MAC               : F06E0BDDCEE6
DriverDescription : Marvell AVASTAR Bluetooth Radio Adapter
DriverVersion     : 15.68.17022.122
ErrorRecovery     : None
ScoType           : InBand
```

If you could run the following command in Power Shell and share the results, it will help us compile a list of know problematic chips/drivers.

    wget https://github.com/Microsoft/busiotools/raw/master/bluetooth/tracing/GetBluetoothRadioInfo.ps1 -UseBasicParsing | iex

Also, please try installing bleak from the latest `develop` branch. I think it may fix the issue with getting services.

purpl3F0x, 2023-02-24:
```
InstanceId        : USB\VID_0B05&PID_1868\5&2A753FF6&0&11
MAC               : E04F436FCBD5
DriverDescription : Qualcomm Atheros QCA61x4A Bluetooth 4.1
DriverVersion     : 10.0.0.448
ErrorRecovery     : None
ScoType           : InBand
```

I installed latest develop version (0.20.0a1) directly from gh. But the behavior seems to be the more or less the same. The debug log shows bit more info though.

```
2023-02-24 18:49:23,856 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ C3:44:6F:5A:65:B8
2023-02-24 18:49:24,136 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2023-02-24 18:49:24,136 bleak.backends.winrt.client DEBUG: calling get_gatt_services_async
2023-02-24 18:49:24,201 bleak.backends.winrt.client DEBUG: returned from get_gatt_services_async
2023-02-24 18:49:24,202 bleak.backends.winrt.client DEBUG: calling get_characteristics_async
2023-02-24 18:49:24,256 bleak.backends.winrt.client DEBUG: returned from get_characteristics_async
2023-02-24 18:49:24,259 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,268 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,270 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,275 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,277 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,282 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,284 bleak.backends.winrt.client DEBUG: calling get_characteristics_async
2023-02-24 18:49:24,332 bleak.backends.winrt.client DEBUG: returned from get_characteristics_async
2023-02-24 18:49:24,334 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,339 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,341 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,347 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,348 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,354 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,355 bleak.backends.winrt.client DEBUG: calling get_characteristics_async
2023-02-24 18:49:24,395 bleak.backends.winrt.client DEBUG: returned from get_characteristics_async
2023-02-24 18:49:24,397 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,404 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,406 bleak.backends.winrt.client DEBUG: calling get_characteristics_async
2023-02-24 18:49:24,446 bleak.backends.winrt.client DEBUG: returned from get_characteristics_async
2023-02-24 18:49:24,448 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,456 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,457 bleak.backends.winrt.client DEBUG: calling get_characteristics_async
2023-02-24 18:49:24,500 bleak.backends.winrt.client DEBUG: returned from get_characteristics_async
2023-02-24 18:49:24,502 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,507 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,509 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,515 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,516 bleak.backends.winrt.client DEBUG: calling get_characteristics_async
2023-02-24 18:49:24,576 bleak.backends.winrt.client DEBUG: returned from get_characteristics_async
2023-02-24 18:49:24,578 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,588 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,591 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,601 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,604 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,614 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
2023-02-24 18:49:24,615 bleak.backends.winrt.client DEBUG: calling get_characteristics_async
2023-02-24 18:49:24,655 bleak.backends.winrt.client DEBUG: returned from get_characteristics_async
2023-02-24 18:49:24,657 bleak.backends.winrt.client DEBUG: calling get_descriptors_async
2023-02-24 18:49:24,665 bleak.backends.winrt.client DEBUG: returned from get_descriptors_async
```

purpl3F0x, 2023-02-24:
Update: I tried manually updating the driver to 10.0.0.1258 from this https://rog.asus.com/forum/showthread.php?124991-DRIVERS-Qualcomm-WiFi-Bluetooth. After taking a very long time device does indeed connect w/o scanning.

dlech, 2023-02-24:
So problem solved?

purpl3F0x, 2023-04-20:
Sorry for not reporting back, notification was buried away. I will mark as closed, since everything seems to be working okay.

closed 2023-04-20 by purpl3F0x