hbldh/bleak #1240: unreliable connection to my BLE device needs lots of attempts to connect

kazola, 2023-02-26:
* bleak version: 0.19.5
* Python version: 3.9
* Operating System: Raspbian 11 bullseye
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.66

### Description

Connect reliably to a BLE device based on CC26X2R.
It needs a lot of attempts to connect, used to be much more stable.

### What I Did

Once it connects, everything works, but some times it takes a lot of attempts to connect. More than 10.
I activated
export BLEAK_LOGGING=1
The thing is that it seems to connect, from what I can understand from logs, but just too late and is discarded?
As you can see, we have a lot of devices. I tried to isolate around important sections by including a mark such as 
----------------
For example, I think the last part of the log can be enough:

2023-02-25 22:38:45,103 bleak.backends.bluezdbus.client DEBUG: Connecting to BlueZ path /org/bluez/hci0/dev_60_77_71_22_CA_57
2023-02-25 22:38:48,078 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_CA_57): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', True)>}, []]
2023-02-25 22:38:48,447 bleak.backends.bluezdbus.client DEBUG: _cleanup_all(/org/bluez/hci0/dev_60_77_71_22_CA_57)

----------connect attempt 3 of 3 failed, h hci0
[org.bluez.Error.Failed] le-connection-abort-by-local
2023-02-25 22:38:48,451 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_CA_57): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', False)>}, []]

On several spots it says connected! But it does not seem so. My connection routine for bleak is:

`    async def connect(self, mac):
        def c_rx(_: int, b: bytearray):
            self.ans += b

        n = 3
        for i in range(n):
            try:
                # we pass hci here
                h = self.h
                self.cli = BleakClient(mac, adapter=h)
                if await self.cli.connect(timeout=10):
                    await self.cli.start_notify(UUID_T, c_rx)
                    return 0

            except (asyncio.TimeoutError, BleakError, OSError) as ex:
                e = 'connect attempt {} of {} failed, h {}'
                print(e.format(i + 1, n, self.h))
                print(ex)
                await asyncio.sleep(1)
        return 1`

While not perfect, now that I recall I have not a single connection problem when using this code on the laptop.
Maybe something related to BLE service discovery? Maybe related to connection parameters? Again RPI has all these connection errors, laptop shows none.

Let's see if someone can help me, I am buying beers.

### Logs

(I think the important part is at the end)

..
...org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_1E_C0_4D_C0_07): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:33,944 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_C8_16): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:33,946 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_C5_AF): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:33,947 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_28_11_A5_55_C8_39): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:33,947 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_2C_41_A1_D8_C7_C1): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:33,948 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_CA_41): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:33,949 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_1E_C0_4D_BF_C9): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:33,949 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_1E_C0_3D_7A_D6): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:33,950 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_CA_18): ['org.bluez.Device1', {}, ['RSSI']]
[... 141 lines cut ...]
2023-02-25 22:38:45,086 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_CA_57): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,087 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_1E_C0_4D_C0_07): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,088 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_28_11_A5_55_E3_BB): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,089 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_58_93_D8_A4_B6_AB): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,089 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_1E_C0_5A_84_2F): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,090 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_1E_C0_4D_BF_C9): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,091 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_C7_32): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,091 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_2C_41_A1_D8_C7_C1): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,092 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_1E_C0_3D_7A_D6): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,093 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_58_93_D8_A4_B6_A6): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,093 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_1E_C0_6C_75_13): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,094 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_C5_AF): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,095 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_C8_16): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,095 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_00_1E_C0_6C_76_0B): ['org.bluez.Device1', {}, ['RSSI']]
2023-02-25 22:38:45,096 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_fast.signature.Variant ('b', False)>}, []]
2023-02-25 22:38:45,103 bleak.backends.bluezdbus.client DEBUG: Connecting to BlueZ path /org/bluez/hci0/dev_60_77_71_22_CA_57
2023-02-25 22:38:48,078 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_CA_57): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', True)>}, []]
2023-02-25 22:38:48,447 bleak.backends.bluezdbus.client DEBUG: _cleanup_all(/org/bluez/hci0/dev_60_77_71_22_CA_57)

----------connect attempt 3 of 3 failed, h hci0
[org.bluez.Error.Failed] le-connection-abort-by-local
2023-02-25 22:38:48,451 bleak.backends.bluezdbus.manager DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_60_77_71_22_CA_57): ['org.bluez.Device1', {'Connected': <dbus_fast.signature.Variant ('b', False)>}, []]

----------
2023-02-25 22:38:49 | [ BLE ] error dl_cc26x2r_exception cc26x2 interact connecting

kazola, 2023-02-26:
A bit more info, I created a extremely simplified script which only tries to connect once. This is the btmon output for a FAILURE:

pi@raspberrypi:~/li/ddt $ sudo btmon
Bluetooth monitor ver 5.66
= Note: Linux version 5.15.61-v7l+ (armv7l)                                                                                                                                                              0.163668
= Note: Bluetooth subsystem version 2.22                                                                                                                                                                 0.163686
= New Index: E4:5F:01:84:19:70 (Primary,UART,hci0)                                                                                                                                                [hci0] 0.163699
= Open Index: E4:5F:01:84:19:70                                                                                                                                                                   [hci0] 0.163710
= Index Info: E4:5F:01:84:19:70 (Cypress Semiconductor)                                                                                                                                           [hci0] 0.163715
@ MGMT Open: bluetoothd (privileged) version 1.21                                                                                                                                               {0x0001} 0.163720
@ RAW Open: hciconfig version 2.22                                                                                                                                                              {0x0002} 6.268052
@ RAW Close: hciconfig                                                                                                                                                                          {0x0002} 6.268466
@ RAW Open: hciconfig version 2.22                                                                                                                                                              {0x0002} 6.277072
@ RAW Close: hciconfig                                                                                                                                                                          {0x0002} 6.277330
@ MGMT Command: Start Service Discovery (0x003a) plen 4                                                                                                                                  {0x0001} [hci0] 6.419299
        Address type: 0x06
          LE Public
          LE Random
        RSSI: invalid (0x7f)
        UUIDs: 0
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6                                                                                                                                      #1 [hci0] 6.419354
        Address: 24:18:28:43:00:C9 (Non-Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                    #2 [hci0] 6.419652
      LE Set Random Address (0x08|0x0005) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7                                                                                                                                     #3 [hci0] 6.419680
        Type: Active (0x01)
        Interval: 11.250 msec (0x0012)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                    #4 [hci0] 6.419967
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2                                                                                                                                         #5 [hci0] 6.419992
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                    #6 [hci0] 6.420336
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4                                                                                                                                           {0x0001} [hci0] 6.420369
      Start Service Discovery (0x003a) plen 1
        Status: Success (0x00)
        Address type: 0x06
          LE Public
          LE Random
@ MGMT Event: Discovering (0x0013) plen 2                                                                                                                                                {0x0001} [hci0] 6.420377
        Address type: 0x06
          LE Public
          LE Random
        Discovery: Enabled (0x01)
> HCI Event: LE Meta Event (0x3e) plen 43                                                                                                                                                      #7 [hci0] 6.459640
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Public (0x00)
[... cut ...]
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                               #34 [hci0] 11.063655
        Status: Success (0x00)
        Handle: 64 Address: 60:77:71:22:CA:57 (Texas Instruments)
        Reason: Connection Failed to be Established (0x3e)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                       {0x0001} [hci0] 11.096121
        LE Address: 60:77:71:22:CA:57 (Texas Instruments)
        Reason: Unspecified (0x00)

dlech, 2023-02-26:
> used to be much more stable.

If it used to work better, then what has changed since then? Bleak version? BlueZ version? Other/more Bluetooth or Wi-fi devices?

dlech, 2023-02-28:
> [org.bluez.Error.Failed] le-connection-abort-by-local

I think the newer version of BlueZ gets this error more often, e.g. #993.

#1220 suggests that adding a retry could work around the issue in most cases.

kazola, 2023-03-02:
I added a retry based on time (try to connect for 30 seconds) instead of number of retries since it can fail several times in a row very quick.

It is, at least in my case, strongly related to using Wi-fi and BLE at the same time. I switched off Wi-Fi and I am using my cell connection, and works much better.

Thanks for your time.

karlp, 2023-03-17:
yeah, newer bluez is .... wonky somehow.  If i use bleak to connect to a string mac address, it now takes 5-6 or more attempts to get a connection.  I can use nrf connect on my phone to connect every time.  If I do the following steps manually in bluetoothctl, it works first time though.

```
bluetoothctl
scan le
<ignore spam>
connect blah
```

It's something with the devices being removed when scanning stops, but it's _wayyyy_ more problematic than it used to be.  The scanning spam stops when it gets connected, but resumes when you disconnect.  I'm on bluez 5.66

dlech, 2023-03-17:
Closing as duplicate of #1220.

closed 2023-03-17 by dlech