hbldh/bleak #1241: bleak.exc.BleakDBusError: Operation failed with ATT error: 0x0e (Unlikely Error)

kiricheglov, 2023-02-26:
* bleak version: 0.19.5
* Python version: 3.10.6
* Operating System: Pop!_OS 22.04
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.66

### Description

Hello world! I can't hook up my Huawei Band 7 to a PC on Linux because of this error

### What I Did

i am trying to connect the zyv (https://github.com/zyv/huawei-lpv2) code to the watch band 7 but the code gives me an error

### Logs
[logs.log](https://github.com/hbldh/bleak/files/10833678/logs.log)

dlech, 2023-02-26:
The logs say that pairing failed so I'm guessing that is why it disconnects.

kiricheglov, 2023-03-04:
The authentication problem can be in the library, or it's a watch problem?

dlech, 2023-03-04:
You could try pairing the watch first.

kiricheglov, 2023-03-05:
And then try to let the script, right?