hbldh/bleak #1242: Multiple Device Connection Stop Working After ~1 Hour

hasfurk, 2023-02-27:
* bleak version:             0.19.5
* Python version:          3.9.2
* Operating System:     Raspberry PI OS 64-Bit
* BlueZ version:            5.66

### Description

I am connecting 11 devices to my Raspberry PI 4B 4GB. For this purpose, I am using separate threads for each device and one thread for scanning these devices. Scanner thread finds the device using desired interface ("hci0", "hci1", etc.) and when it returns its reference, a dedicated thread for that device connects to it and starts its notifications. When connection occurs in a thread, "{ID} Connected" message is printed and similar for disconnection. In notification callbacks, each data coming from a module updates its corresponding field in a list, so they are not trying to change the same area. I write this list to a file each second and this file and threads closed safely after I hit Ctrl+C, since threads are running in a **while** loop where condition is changed with a **KeyboardInterrupt**. This operation goes without a problem for ~1 hour, but after that point I observed that when I close a device, disconnection message is not printed and it does not connect when I open it again. In "hciconfig" I can see that RX bytes are increasing, but after I terminate the program written values in the file stop changing after that specific point. I carried out this experiment 4 times:

1. Using 2 dongles and 11 devices
2. Using 2 dongles and 11 devices (same setup with 1)
3. Using RPI's own hci interface and 11 devices (no dongles)
4. Using 2 dongles and 11 devices (checking the state of the operation in every 10 mins)

In 4th experiment, system did not fail for that amount of time (~1 hour). I disconnected and connected devices arbitrarily and observed the printed messages. However, situation did not change. I could not find a chance to run this program on a different hardware or operating system. Therefore, I am not sure if this is caused because of RPI or my implementation.

### What I Did

My implementation consists of 4 separate .py files:

1. main.py
- Starts threads
- Writes the list to file
- Closes the file and threads when KeyboardInterrupt is raised

2. config.py
- Contains configuration parameters such as data list, device list, MAC addresses and UUIDs

3. thread_connection.py
- Connects to device when corresponding device's reference is not None (i.e. Device found by scanner thread)
- Reconnects if device is disconnected and program is not termianted
- Closes notifications and disconnects if the program gets terminated

4. thread_scan.py 
- Scans for the MAC addresses given in the list under config.py
- Saves device references to a list given under config.py

Here is the content of thread_connection.py
```
import threading
import time
import asyncio
import bleak

from bleak import BleakClient, BleakScanner
from bleak.backends.characteristic import BleakGATTCharacteristic

import config as cf

class Thread_Connection(threading.Thread):
    def __init__(self, ID, address):
        threading.Thread.__init__(self)
        self.ID = ID
        
    def disconnected_callback(self, ref_client):
        print("{} Is Disconnected".format(self.ID+1))

    def notification_handler(self, characteristic: BleakGATTCharacteristic, data: bytearray):
        cf.data_list[self.ID] = data[:]

    async def connect(self):
        while cf.FLAG_RUNNING:
            try:
                async with BleakClient(
                                       address_or_ble_device  = cf.DEVICES[self.ID],
                                       disconnected_callback  = self.disconnected_callback,
                                       timeout                = cf.TIMEOUT
                                       ) as client:
                    print("{} Is Connected".format(self.ID+1))

                    await client.start_notify(cf.char_UUID, self.notification_handler)
                    while cf.FLAG_RUNNING:
                        await asyncio.sleep(1.0)
                        if not client.is_connected:
                            break
                    await client.stop_notify(cf.char_UUID)
                    await client.disconnect()
            except Exception as e:
                print(e)
                pass

    def run(self):
        print(f"Thread {self.ID+1} Started")
        while cf.DEVICES[self.ID] == None and cf.FLAG_RUNNING:
            time.sleep(1)
        asyncio.run(self.connect())
        print(f"Thread {self.ID+1} Stopped")
```

and here is the content of thread_scan.py

```
from bleak import BleakScanner
import threading
import asyncio
import config as cf

class Thread_Scan(threading.Thread):
    def __init__(self):
        threading.Thread.__init__(self)
    

    async def scan(self):
        while None in cf.DEVICES and cf.FLAG_RUNNING:
            indices_None = [index for index in range(cf.DEVICE_AMOUNT) if cf.DEVICES[index]==None]
            for index_None in indices_None:
                if not cf.FLAG_RUNNING:
                    return
                
                try:
                    cf.DEVICES[index_None]     = await BleakScanner.find_device_by_address(
                                                                     device_identifier    = cf.MAC_ADDRESSES[index_None],
                                                                     timeout                  = cf.TIMEOUT,
                                                                     adapter                  = "hci0" if index_None//5 else "hci1"
                                                                     )
                except Exception as e:
                    print(e)
                    pass
                
        
    def run(self):
        asyncio.run(self.scan())
        print("Scanning Thread Closed")
```

### Logs

I did not run these experiments with verbose, which I will be doing next. However, I print Exception messages, but I did not see any when this problem occurs. Only the last connection/disconnection message is printed on the terminal.

Sorry if I have missed any regulation during opening my issue and thank you for your time.

dlech, 2023-02-27:
Running multiple instances of Bleak in threads is not supported. Please use a single thread and asyncio tasks for parallelization.

hasfurk, 2023-02-27:
Thanks for your reply. I have read that it is not thread safe, but seeing it working for a time made me think that it is safe for this case. Would you tell me what may be causing this time limitation? Do threads cause memory leak or such thing?

dlech, 2023-02-27:
Bleak was designed to work with asyncio with no consideration for threads, so the could be many problems we never though of. Bluetooth chips in general tend to get very confused when you try to do more than one operation at a time. As one example, the BlueZ/Linux backend makes uses of an asyncio Lock to help with this in some cases, but using threads subverts this.

hasfurk, 2023-02-27:
Thanks. I will redesign my program considering these.

closed 2023-02-27 by hasfurk