hbldh/bleak #1244: cannot connect unless active scanning is running, since linux kernel 6.x

gilcu3, 2023-03-03:
* bleak version: `0.19.5`
* Python version: `3.10`
* Operating System: ArchLinux
* BlueZ version (`bluetoothctl -v`) in case of Linux: `5.66`
* hardware `lsusb | grep -i bluetooth`: `Bus 001 Device 005: ID 8087:0aa7 Intel Corp. Wireless-AC 3168 Bluetooth`

### Description

Since I updated the linux kernel to 6.x (any version that I have tried), `bleak` cannot connect to bluetooth devices unless there is an active scan running.

### What I Did

```
#! /usr/bin/env python3
import asyncio

import logging
from bleak import BleakClient, BleakScanner
from bleak.backends.device import BLEDevice
import asyncio

logging.basicConfig()
logging.getLogger().setLevel(logging.DEBUG)

# Add here the mac address of the device to test
MAC_ADDRESS = '7C:A6:B0:5D:32:27'

async def main():
    scanner = BleakScanner()
    d = await scanner.find_device_by_address(MAC_ADDRESS)
    print(d.address)
if __name__ == '__main__':
    asyncio.run(main())
```

### Logs

When running the above script I get:
```
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_76_B2_15_AC_9C_8C', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', 76:B2:15:AC:9C:8C)>, 'AddressType': <dbus_fast.signature.Variant ('s', random)>, 'Alias': <dbus_fast.signature.Variant ('s', 76-B2-15-AC-9C-8C)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -78)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {76: <dbus_fast.signature.Variant ('ay', bytearray(b'\x10\x06\x00\x19W5\xfbx'))>})>, 'TxPower': <dbus_fast.signature.Variant ('n', 12)>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>, 'AdvertisingFlags': <dbus_fast.signature.Variant ('ay', bytearray(b'\x1a'))>}, 'org.freedesktop.DBus.Properties': {}}]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_DF_BB_A0_4D_CA_2C', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_fast.signature.Variant ('s', DF:BB:A0:4D:CA:2C)>, 'AddressType': <dbus_fast.signature.Variant ('s', random)>, 'Alias': <dbus_fast.signature.Variant ('s', DF-BB-A0-4D-CA-2C)>, 'Paired': <dbus_fast.signature.Variant ('b', False)>, 'Bonded': <dbus_fast.signature.Variant ('b', False)>, 'Trusted': <dbus_fast.signature.Variant ('b', False)>, 'Blocked': <dbus_fast.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_fast.signature.Variant ('b', False)>, 'RSSI': <dbus_fast.signature.Variant ('n', -94)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>, 'UUIDs': <dbus_fast.signature.Variant ('as', [])>, 'Adapter': <dbus_fast.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {76: <dbus_fast.signature.Variant ('ay', bytearray(b'\x12\x02\x00\x00'))>})>, 'ServicesResolved': <dbus_fast.signature.Variant ('b', False)>, 'AdvertisingFlags': <dbus_fast.signature.Variant ('ay', bytearray(b'\x00'))>}, 'org.freedesktop.DBus.Properties': {}}]
Traceback (most recent call last):
  File "/home/rey/opt/govee/bug_bleak/test.py", line 20, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/rey/opt/govee/bug_bleak/test.py", line 18, in main
    print(d.address)
AttributeError: 'NoneType' object has no attribute 'address'
```

If at the same time I run this script:
```
#!/usr/bin/env bash
while [[ 1 ]]
do
echo on
bluetoothctl -- scan on
sleep 20
done
```
then it connects correctly and I get:
```
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_7B_0F_4F_DD_11_B5): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -88)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {76: <dbus_fast.signature.Variant ('ay', bytearray(b'\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x80\x00\x00\x00\x00'))>})>}, []]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C2_1C_05_09_CB_3D): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -37)>, 'ManufacturerData': <dbus_fast.signature.Variant ('a{qv}', {2409: <dbus_fast.signature.Variant ('ay', bytearray(b'\xc2\x1c\x05\t\xcb=\xc4\x03\x07\x96 '))>})>}, []]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_17_0D_2D_4F_2E_8B): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -90)>}, []]
DEBUG:bleak.backends.bluezdbus.manager:received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_7C_A6_B0_5D_32_27): ['org.bluez.Device1', {'RSSI': <dbus_fast.signature.Variant ('n', -59)>}, []]
7C:A6:B0:5D:32:27
```

dlech, 2023-03-03:
If the only difference is the Linux kernel version then the regression should be reported on the Linux Bluetooth mailing list.

gilcu3, 2023-03-03:
> If the only difference is the Linux kernel version then the regression should be reported on the Linux Bluetooth mailing list.

You may be right, although I reported it here because using blueman or bluetoothctl everything seems to be working as usual. Do you have any suggestion to test if it is really a kernel issue?

dlech, 2023-03-03:
Capture logs (e.g. with `btmon`) for both a working and not working case so everyone can see what the difference is.

gilcu3, 2023-03-03:
thanks, just to get things straight. The device is some strip lights, and it seems to be using BLE, and cannot be paired. Using `bluetoothctl` I can connect to it, but only if I do a `scan on` first. Else it says device not available. Is that normal behavior?

dlech, 2023-03-03:
Yes and no. Connect on its own does a scan but with a different configuration than a general scan. The logs will show this difference.

gilcu3, 2023-03-03:
When executing the following:
```
$ bluetoothctl
[bluetooth]# power on
[CHG] Controller 10:F0:05:3C:3D:76 PowerState: off-enabling
[CHG] Controller 10:F0:05:3C:3D:76 Class: 0x0000010c
Changing power on succeeded
[CHG] Controller 10:F0:05:3C:3D:76 PowerState: on
[CHG] Controller 10:F0:05:3C:3D:76 Powered: yes
[bluetooth]# connect 7C:A6:B0:5D:32:27
Device 7C:A6:B0:5D:32:27 not available
```
I get almost no log by btmon: [bt.log](https://github.com/hbldh/bleak/files/10883625/bt.log)
If I do a scan first:
```
$ bluetoothctl
[bluetooth]# power on
[CHG] Controller 10:F0:05:3C:3D:76 PowerState: off-enabling
[CHG] Controller 10:F0:05:3C:3D:76 Class: 0x0000010c
Changing power on succeeded
[CHG] Controller 10:F0:05:3C:3D:76 PowerState: on
[CHG] Controller 10:F0:05:3C:3D:76 Powered: yes
[bluetooth]# scan on
.... lots of devices....
bluetooth]# connect 7C:A6:B0:5D:32:27
Attempting to connect to 7C:A6:B0:5D:32:27
[CHG] Device 7C:A6:B0:5D:32:27 Connected: yes
Connection successful
[CHG] Device 7C:A6:B0:5D:32:27 ServicesResolved: yes
```
and log [bt_scan.log](https://github.com/hbldh/bleak/files/10883638/bt_scan.log)

dlech, 2023-03-03:
If you are doing everything from `bluetoothctl` you have to do a scan first so that the device is added to the known devices in BlueZ. Bleak roughly does the equivalent of:

```
menu scan
transport le
duplicate-data off
back
scan on
scan off
connect XX:XX:XX:XX:XX
```

So I'm guessing that either the `transport` or `duplicate-data` option is what is making the difference.

gilcu3, 2023-03-03:
When I do exactly those steps it seems to work, although not the first time, and after a few seconds it cannot connect again:
```
[bluetooth]# connect 7C:A6:B0:5D:32:27
Attempting to connect to 7C:A6:B0:5D:32:27
[CHG] Device 7C:A6:B0:5D:32:27 Connected: yes
Failed to connect: org.bluez.Error.Failed le-connection-abort-by-local
[CHG] Device 7C:A6:B0:5D:32:27 Connected: no
[bluetooth]# connect 7C:A6:B0:5D:32:27
Attempting to connect to 7C:A6:B0:5D:32:27
[CHG] Device 7C:A6:B0:5D:32:27 Connected: yes
Connection successful
.... few seconds ....
[bluetooth]# connect 7C:A6:B0:5D:32:27
Device 7C:A6:B0:5D:32:27 not available
```
I have to mention that this device by default disconnects after a few seconds, but shouldn't it be remembered after this (no need for scan on and off again?)

gilcu3, 2023-03-03:
I think I found the culprit of the issue, while the device was available (just after a scan on/off) I did
`trust 7C:A6:B0:5D:32:27` and now it remembers it and is able to connect. It seems logical as an afterthought, but why is this only required after kernel 6.x is a curious matter

gilcu3, 2023-03-06:
This is very likely not a bleak issue, as I could check by stopping everything else that was using bluetooth (homeassistant in particular), so, I am closing this.

closed 2023-03-06 by gilcu3

Vishal8523, 2024-05-14:
Any updates on this ? How to resolve this issue ?

caenrigen, 2026-03-09:
I had this problem too, had to build the linux kernel from source, see my https://github.com/hbldh/bleak/issues/1858#issuecomment-4023374812