hbldh/bleak #1252: Fail to connect to custom GATT service

alphai-dev, 2023-03-10:
* bleak version: 0.19.5   (bleak-winrt==1.2.0)
* Python version: 3.9.13
* Operating System: Windows 10, version 22H2, build 19045

### Description

I am trying to connect to a custom GATT service on a Raspberry Pi model 4B, from a windows PC.
Problem is : when attempting to connect, the program crashes.

From the logs, it seems that the PC is unable to obtain the list of services from the device, or that it inexplicably sends a disconnection request to the device.
I am unsure whether the problem comes from the device, windows system, or bleak.
It might also very well be a silly mistake on my part, as I am quite a newbie when it comes to Bluetooth.

### What I Did

My device is a Raspberry Pi 4B with : 
* Raspberry Pi OS version 11.
* python version 3.9.2
* bluetoothctl version 5.55

The device is running this example script from the *bluezero* module : [cpu_temperature.py](https://github.com/ukBaz/python-bluezero/blob/7cb8e8f019a3a045f074322a08deb1aa9af6ee67/examples/cpu_temperature.py)

My bleak test script is : 
```
import asyncio
from bleak import BleakScanner, BleakClient

# constants
# Custom service uuid
CPU_TMP_SRVC = '12341000-1234-1234-1234-123456789abc'
# https://www.bluetooth.com/specifications/assigned-numbers/
# Bluetooth SIG adopted UUID for Temperature characteristic
CPU_TMP_CHRC = '2A6E'
# Bluetooth SIG adopted UUID for Characteristic Presentation Format
CPU_FMT_DSCP = '2904'

def filter_function(device, adv_data):
    name = "CPU Monitor"
    if device.name == name or adv_data.local_name == name:
        print(f"found device, {device.name=}, {adv_data.local_name=}")
        return True

def detection_callback(device, adv_data):
    pass

async def main():
    scanner = BleakScanner(detection_callback, service_uuids=[CPU_TMP_SRVC])
    # Detection
    device = await scanner.find_device_by_filter(filter_function, timeout=30.0)
    client = BleakClient(device,
                         winrt={"address_type": "public",
                                "use_cached_services": False},
                         use_cached_services=False
                         )
    print("device address = " + device.address)
    await client.connect()  # Connection
    print("reading characteristics")
    # tmp_chr = client.services.get_characteristic(CPU_TMP_CHRC)
    data = await client.read_gatt_char(CPU_TMP_CHRC)
    value = int.from_bytes(data, byteorder='little', signed=True)/100
    print(f"CPU temperature is {value}\u00b0C")
    await client.disconnect()

asyncio.run(main())
print("test completed")
```

The error is on the line `await client.connect()`. The next line `print` does not show in the logs.

### Logs

Bleak logs :
```
2023-03-10 15:52:28,782 bleak.backends.winrt.scanner DEBUG: Received E4:5F:01:D5:77:AB: .
2023-03-10 15:52:28,782 bleak.backends.winrt.scanner DEBUG: skipping callback, waiting for scan response
2023-03-10 15:52:28,782 bleak.backends.winrt.scanner DEBUG: Received E4:5F:01:D5:77:AB: CPU Monitor.
found device, device.name='CPU Monitor', adv_data.local_name='CPU Monitor'
2023-03-10 15:52:28,782 bleak.backends.winrt.scanner DEBUG: 10 devices found. Watcher status: 3.
device address = E4:5F:01:D5:77:AB
2023-03-10 15:52:28,792 bleak.backends.winrt.client DEBUG: Connecting to BLE device @ E4:5F:01:D5:77:AB
2023-03-10 15:52:28,812 bleak.backends.winrt.client DEBUG: getting services (service_cache_mode=<BluetoothCacheMode.UNCACHED: 1>, cache_mode=<BluetoothCacheMode.UNCACHED: 1>)...
2023-03-10 15:52:31,342 bleak.backends.winrt.client DEBUG: session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x00000211966221D0>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
2023-03-10 15:52:38,832 bleak.backends.winrt.client DEBUG: closing requester
2023-03-10 15:52:47,502 bleak.backends.winrt.client DEBUG: max_pdu_size_changed_handler: 517
2023-03-10 15:52:47,502 bleak.backends.winrt.client DEBUG: closing session
```

[bleak_log_fresh_restart.txt](https://github.com/hbldh/bleak/files/10943516/bleak_log_fresh_restart.txt)

WireShark capture :
[bluetooth_packets_fresh_restart.zip](https://github.com/hbldh/bleak/files/10943584/bluetooth_packets_fresh_restart.zip)

dlech, 2023-03-10:
The WireShark logs have a bunch of stuff I've never seen before on a BLE connection. I'm guessing it is Bluetooth classic stuff from BlueZ on the RPi. 

The last thing before the disconnect is an error related to AVRCP protocol (audio/video remote control).

[image]

I would suggest trying to disable all of the Bluetooth classic stuff on the RPi so that it doesn't interfere.

alphai-dev, 2023-03-10:
Thank you very much for your swift reply!

Wow, I had not thought about that, and it is indeed much better now!

From some quick tests, it seems that, when the device is already paired (from Windows settings), everything works like a charm.

However, when not already paired, the connection attempt prompts a Windows notification asking for pairing. Following the pairing process, Windows reports the pairing as successful, however the python script fails with an `asyncio.timeout` error (even when I am trying to be quick with the pairing process).

This is a minor issue for me, since pairing beforehand is a possibility. However, I would prefer minimum user-interaction with the pairing process. My understanding was that read/write without encryption did not require pairing, is it not the case? Here, the pairing process seems to be caused by the `client.connect()` line.

Should I open a new issue for this error?

Have a nice weekend

dlech, 2023-03-10:
The timeout during connect should be configurable so you can make it as long as you need.