hbldh/bleak #1253: 'NoneType' object has no attribute 'max_pdu_size' -- again.

joooeey, 2023-03-14:
* bleak version: 0.19.5
* Python version: 3.10.9
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I was trying to connect to two BLE devices. Similar code with one device used to sometimes work and sometimes throw an error. Now with two devices I always get an error where bleak tries to get `.max_pdu_size` from a name that refers to `None`. Both 0.17.0 and 0.19.0 supposedly fixed similar errors but it still happens.

### What I Did

I only have a mix of printouts and a traceback for now:

```
Connecting to sensorbox ...
Connecting to controller ...
Task exception was never retrieved
future: <Task finished name='Task-19' coro=<BleakClientWinRT.get_services() done, defined at C:\Users\user\anaconda3\envs\moma\lib\site-packages\bleak\backends\winrt\client.py:601> exception=AttributeError("'NoneType' object has no attribute 'max_pdu_size'")>
Traceback (most recent call last):
  File "C:\Users\user\anaconda3\envs\moma\lib\site-packages\bleak\backends\winrt\client.py", line 667, in get_services
    characteristic, self._session.max_pdu_size - 3
AttributeError: 'NoneType' object has no attribute 'max_pdu_size'
TimeoutError()
sensorbox not connected.
Connecting to sensorbox ...
```

dlech, 2023-03-14:
Is this still a problem if you use the latest `develop` branch?

joooeey, 2023-03-15:
> Is this still a problem if you use the latest `develop` branch?

No, it appears to be resolved. I'll report if it happens again.

When is the next release?

closed 2023-03-15 by joooeey

dlech, 2023-03-15:
Don't have an ETA, but hopefully soon.