hbldh/bleak #1254: Windows BleakScanner.discover() does not find any BLE devices other scanners do find devices

jpstotz, 2023-03-15:
* bleak version: 0.20.0a1
* bleak-winrt  version: 1.2.0
* Python version: Python 3.11.1 (64bit)
* Operating System: Windows 10 22H2

### Description

Executing the `BleakScanner.discover()` sample script from https://github.com/hbldh/bleak#usage does not return any BLE devices

### What I Did

Script
```
import asyncio
from bleak import BleakScanner

async def main():
    devices = await BleakScanner.discover()
    print("found devices {}".format(len(devices)))
    for d in devices:
        print(d)

asyncio.run(main())
```

1. Cloned latest bleak version  and installed it into the current venv
3. Executed the sample script

```
2023-03-15 16:10:53,951 bleak.backends.winrt.scanner MainThread DEBUG: 0 devices found. Watcher status: 3.
found devices 0
```

No error, it just can not find any device despite there are devices present and known to Windows.

Output of https://github.com/sensboston/BLEConsole `list` commend at the same time shows two devices:
```
#00: ACTON II
#01: SomeOtherDevice12348354
```

dlech, 2023-03-15:
Does the [detection_callback.py](https://github.com/hbldh/bleak/blob/develop/examples/detection_callback.py) example give any results?

jpstotz, 2023-03-15:
Thanks for your response.
[detection_callback.py](https://github.com/hbldh/bleak/blob/develop/examples/detection_callback.py) also had no results unless I unpaired the device on Windows level. 

Afterwards I can see the device in Bleak and some times a second unknown device. Pairing the device again with Windows does not change anything, the device is still visible in Bleak scans. So I am not sure what actually caused this problem, as I had not changed something else.

I also built a small test set-up for BLE device scanning using [nRF Connect app for Andorid](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp). If I use this app and create an `ADVERTISER` that has as Advertising data the `Complete local name` and enable it I can see this device in most scans in Bleak. 

So next time something like this happens I am prepared...

closed 2023-03-15 by jpstotz

dlech, 2023-03-15:
Could also be the same issue as #1211.