hbldh/bleak #1262: windows 11 Run Time out

yonghee1982, 2023-03-20:
* bleak version: 0.19.5
* Python version: 3.11
* Operating System: windows 10 / 11
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
Please understand that I am writing using a translation tool.

I represented the searched Bluetooth devices in a list box and attempted to connect to a device by selecting it.
However, although the connection is completed within seconds on Windows 10, on Windows 11 the program stops or a Time Out Error message occurs.

### What I Did

This code is a part of the whole code.

    async def connect_device_async(self, device):
        try:
            self.client = BleakClient(device.address, timeout=60)
            await self.client.connect()
            messagebox.showinfo("Success", f"Connected to {device.address} - {device.name}")
            self.result_text.insert("1.0", f"Connected to {device.address} - {device.name}\n")
        except BleakError as e:
            messagebox.showerror("Error", str(e))

    def connect_device(self):
        if len(self.devices_listbox.curselection()) == 0:
            messagebox.showerror("Error", "No device selected.")
            return
        index = self.devices_listbox.curselection()[0]
        device = self.devices[index]
        asyncio.run(self.connect_device_async(device))

### Logs

Exception in Tkinter callback
Traceback (most recent call last):
  File "bleak\backends\winrt\client.py", line 402, in connect
  File "asyncio\tasks.py", line 418, in wait
  File "asyncio\tasks.py", line 525, in _wait
asyncio.exceptions.CancelledError During handling of the above exception, another exception occurred: Traceback (most recent call last):
  File "tkinter\__init__.py", line 1948, in __call__
  File "InBodyBT500ModeChange.py", line 115, in connect_device
  File "asyncio\runners.py", line 190, in run
  File "asyncio\runners.py", line 118, in run
  File "asyncio\base_events.py", line 653, in run_until_complete
  File "InBodyBT500ModeChange.py", line 103, in connect_device_async
  File "bleak\__init__.py", line 471, in connect
  File "bleak\backends\winrt\client.py", line 389, in connect
  File "async_timeout\__init__.py", line 129, in __aexit__
  File "async_timeout\__init__.py", line 212, in _do_exit
TimeoutError

dlech, 2023-03-21:
Is this still a problem with Bleak v0.20.0?

yonghee1982, 2023-03-21:
It was version 0.19.5, and I tried to proceed after upgrading to 0.20.0, but the problem was not resolved.

On Windows 10, both versions 0.19.5 and 0.20.0 work properly.

dlech, 2023-03-21:
What are the exact differences between the working case and the nit working case? Exact Windows version? Bluetooth radio (in PowerShell, run `wget https://github.com/Microsoft/busiotools/raw/master/bluetooth/tracing/GetBluetoothRadioInfo.ps1 -UseBasicParsing | iex`)? Is it the same computer or a different computer?

yonghee1982, 2023-03-24:
I'm sorry for the late reply.

In Windows 10, device search and connection are possible, but in Windows 11, device search is possible, but connection is not.

Each computer is different, and here are the OS version and Bluetooth Radio information for each:

Windows 10

Version: 21H2 (OS Build 19044.2130)
Bluetooth Radio
InstanceId: USB\VID_8087&PID_0026\5&3B1E8281&0&10
MAC: F4B30105AD7E
DriverDescription: Intel(R) Wireless Bluetooth(R)
DriverVersion: 22.50.1.1
ErrorRecovery: PLDR
ScoType: Unknown
Windows 11

Version: 22H2 (OS Build 22621.1413)
Bluetooth Radio
InstanceId: USB\VID_0BDA&PID_B00C\00E04C000001
MAC: 18473D6E945C
DriverDescription: Realtek Bluetooth 5 Adapter
DriverVersion: 1.8.1037.3000
ErrorRecovery: None
ScoType: InBand

dlech, 2023-03-24:
On Windows 11 do other BLE tools like https://apps.microsoft.com/store/detail/bluetooth-le-explorer/9N0ZTKF1QD98?hl=en-us&gl=us or `chrome://bluetooth-internals` in a Chrome or Edge browser work? If they do, what is the difference in Bluetooth packets captured with them compared to Bleak (https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10).

artodd1, 2023-03-31:
I am having a similar issue.  Connection to my BLE device worked fine with my app on a computer running Windows 10 and running the same app on a computer running Windows 11 it is getting hung up while awaiting BleakClient.connect(), no timeout exception is occuring.

bleak version: 0.20.1
Python version: 3.10.10

OS: Windows 11 Pro
Version: 10.0.2.2261

Bluetooth Radio Info:
InstanceId        : USB\VID_10AB&PID_9309\6&1CAA2945&0&3
DriverDescription : Qualcomm WCN685x Bluetooth Adapter
DriverVersion     : 1.0.0.1290
ErrorRecovery     : None
ScoType           : InBand

I have tried chrome://bluetooth-internals on the computer running Windows 11 and I am able to connect to my BLE device.
I also tried connecting to my device with Bluetooth LE Explorer, I was unable to connect to my device using this BLE tool.

 
[chrome_ble_internals_connect_disconnect_ble.pcapng.gz](https://github.com/hbldh/bleak/files/11125532/chrome_ble_internals_connect_disconnect_ble.pcapng.gz)
[my_app_ble_connect_attempt.pcapng.gz](https://github.com/hbldh/bleak/files/11125533/my_app_ble_connect_attempt.pcapng.gz)
[bluetooth_le_explorer_connect_attempt.pcapng.gz](https://github.com/hbldh/bleak/files/11125534/bluetooth_le_explorer_connect_attempt.pcapng.gz)

dlech, 2023-04-02:
Thanks for the logs. Can you also share the output of your Python script with debug logging enabled?

You could also try fiddling with the `winrt` [use_cached_services](https://bleak.readthedocs.io/en/latest/backends/windows.html#bleak.backends.winrt.client.WinRTClientArgs.use_cached_services) option.

wpilgri, 2023-11-12:
Wanted to add some more information via another example of Windows 11 having problems connecting.

- Latest build of Windows 11
- Bleak v0.21.1
- python v3.11.5
- device I'm connecting to:  STM WB55RG Nucleo

It's a very simple program that I have working on both macOS and Windows 10 which properly scans and finds the device Iwant to connect to (MyST), but BleakClient() call fails to connect with no errors.  Here's the snippet of the main program and the output I see with BLEAK_LOGGING=1

```
async def runmain():

   #Works fine
    await scanfordevices() 

    if found:
        print(f'*** runmain: Trying to connect to: {DEVICE_NAME}')
         
# THIS IS THE CALL WHICH NEVER RETURNS AND NEVER PRODUCES AS ERROR 
        async with BleakClient(ardAddress) as client:

            await getservices(client)

            print('*** runmain: Assign notification callbacks')
            await client.start_notify(Sensor1DataUUID, notification_sensordata_handler) #1st peak
            await client.start_notify(Sensor2DataUUID, notification_sensordata_handler) #2nd peak
            await client.start_notify(Sensor3DataUUID, notification_sensordata_handler) #3rd peak

            while not exit_flag:
                await asyncio.sleep(1)

print('*** runmain: Done.')

```

When I run the above I get this output in the console:

2023-11-12 15:25:30,936 bleak.backends.winrt.scanner MainThread DEBUG: 31 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
	**00:80:E1:26:C4:FE: MyST**
		d.name='MyST'
		d.address='00:80:E1:26:C4:FE'
		adv_data.rssi=-58
*** runmain: Scan is done, checking if Microcontroller with name= MyST found.
*** runmain: Trying to connect to: MyST
2023-11-12 15:25:30,986 bleak.backends.winrt.scanner MainThread DEBUG: Received EF:F7:57:DD:EA:16: .
2023-11-12 15:25:31,011 bleak.backends.winrt.scanner MainThread DEBUG: Received 76:17:EF:19:80:69: .
2023-11-12 15:25:31,011 bleak.backends.winrt.scanner MainThread DEBUG: Received 76:17:EF:19:80:69: .
2023-11-12 15:25:31,100 bleak.backends.winrt.scanner MainThread DEBUG: Received 6C:4A:85:1F:42:E0: .
2023-11-12 15:25:31,100 bleak.backends.winrt.scanner MainThread DEBUG: Received 6C:4A:85:1F:42:E0: .
2023-11-12 15:25:31,107 bleak.backends.winrt.scanner MainThread DEBUG: Received CD:E9:FF:E6:E0:82: .
2023-11-12 15:25:31,118 bleak.backends.winrt.scanner MainThread DEBUG: Received C0:28:8D:FF:B5:F0: .
2023-11-12 15:25:31,124 bleak.backends.winrt.scanner MainThread DEBUG: Received C0:28:8D:FF:B5:F0: .
2023-11-12 15:25:31,209 bleak.backends.winrt.scanner MainThread DEBUG: Received C0:28:8D:FF:B5:F0: .
2023-11-12 15:25:31,213 bleak.backends.winrt.scanner MainThread DEBUG: Received C0:28:8D:FF:B5:F0: .
2023-11-12 15:25:31,217 bleak.backends.winrt.scanner MainThread DEBUG: Received 2C:41:A1:D8:D4:E3: .
2023-11-12 15:25:31,217 bleak.backends.winrt.scanner MainThread DEBUG: Received 2C:41:A1:D8:D4:E3: LE-reserved_C.
2023-11-12 15:25:31,234 bleak.backends.winrt.scanner MainThread DEBUG: Received **00:80:E1:26:C4:FE: MyST.**
2023-11-12 15:25:31,234 bleak.backends.winrt.scanner MainThread DEBUG: Received 00:80:E1:26:C4:FE: .
2023-11-12 15:25:31,247 bleak.backends.winrt.scanner MainThread DEBUG: 7 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
2023-11-12 15:25:31,247 bleak.backends.winrt.client MainThread DEBUG: **Connecting to BLE device @ 00:80:E1:26:C4:FE**
2023-11-12 15:25:31,304 bleak.backends.winrt.client MainThread DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2023-11-12 15:25:31,328 bleak.backends.winrt.client Dummy-5 DEBUG: session_status_changed_event_handler: id: BluetoothLE#BluetoothLEf4:b3:01:ba:7f:ab-00:80:e1:26:c4:fe, error: <BluetoothError.SUCCESS: 0>, status: <GattSessionStatus.ACTIVE: 1>
2023-11-12 15:25:31,393 bleak.backends.winrt.client Dummy-5 DEBUG: max_pdu_size_changed_handler: 156
2023-11-12 15:25:31,657 bleak.backends.winrt.client Dummy-6 DEBUG: 00:80:E1:26:C4:FE: services changed
2023-11-12 15:25:32,771 bleak.backends.winrt.client Dummy-5 DEBUG: 00:80:E1:26:C4:FE: services changed

I'm stumped as it works flawlessly on Windows 10, macOS, and it used to work on the same Windows 11 machine.

Any troubleshooting tips would be greatly appreciated.

wpilgri, 2023-11-13:
Final comment to add is, as others have mentioned here, uninstalling ipykernel in my python environment solved the problem.

dlech, 2023-11-13:
Interesting. Was the `ipykernel` module loaded? If I am remembering correctly, it changes (or at least used to) the default run loop type on Windows. So that could something to look into. I.e. print the default run loop type in the `runmain()` function.

wpilgri, 2023-11-13:
It was not being used/loaded.  Not sure what you mean by printing loop type, sorry.

dlech, 2023-11-15:
> Not sure what you mean by printing loop type, sorry.

```python
loop = asyncio.get_running_loop()
print(type(loop))
```

> It was not being used/loaded.

I wonder if this is something similar to #1132, i.e. `pywin32` is a dependency of `ipykernel` and some other package is importing `pkg_resources`.

wpilgri, 2023-11-17:
Sorry for the slow reply, but type(loop) returns:  class 'asyncio.windows_events.ProactorEventLoop'>

isaac-aavaa, 2024-02-22:
> Final comment to add is, as others have mentioned here, uninstalling ipykernel in my python environment solved the problem.

Hello,
I am having this exact issue, code works perfectly on windows 10 but hangs during the connection on windows 11. 

My specialty is embedded firmware; could I trouble you for how to uninstall ipykernel? Do I just do jupyter kernelspec remove python3?

Thanks!

markmadsen82, 2024-03-07:
Same issue. Did anyone solve this matter in windows 11?

Quaditz, 2024-04-05:
Hello,

I also have the same issue. I can find the device on my windows 10 and windows 11 computer when I do a windows bluetooth device search.

When I do a discover with BleakScanner, I can find the device in Windows 10 but not in Windows 11 :(

dlech, 2024-04-05:
Interesting findings. I'm guessing this "Bluetooth Device Detection" Option is in the Windows settings?

Is the only difference Windows 10 and Windows 11 (e.g. identical computers with the same Bluetooth adapter)

It would also be helpful to get Bluetooth packet captures of working and not working cases to compare them so we can see what the actual difference is.

Quaditz, 2024-04-05:
Regarding the settings: I found an English instruction how to find it. This is was I did in Windows 11 to be able to find my device. In Windows 10 I was able to find it by default.

[](https://answers.microsoft.com/en-us/windows/forum/all/bluetooth-cant-find-any-device-when-i-update-to/ca06c184-7be5-47c9-91ce-0d93bc6e3048)

How could I make such a Bluetooth Capture? Are there any free programs for windows to do that?

dlech, 2024-04-05:
> How could I make such a Bluetooth Capture?

https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

> I recorded now two traces and saved them in the CSV format.

It would be more helpful to have the wireshark file format. You can put these in a .zip file to attach them to the issue.

dlech, 2024-04-05:
If the Windows 11 log is complete, it looks like a Windows bug or driver bug where it just stops querying the peripheral device. The last thing in the log is a response from the peripheral device so Windows shouldn't be waiting for the device to respond.

Quaditz, 2024-04-06:
Hello,

I also did another Windows 11 measurement where I pressed CTRL+C during the freeze.

So I upload 3 measurements in the ZIP file in the Wireshark format:

1. Windows 10: Python does NOT freeze and I can continue as I want. There is no further data in the measurement but the Python Interpreter does not freeze, I can print stuff and I can even start sending data etc.

In this measurement I pressed CTRL+C at the time when the connection was long finished. Printing etc worked before pressing CTRL + C.

2. Windows 11. Python freezes. No printing etc happens even when I just call it the first thing after the "async .. with:" command. In one measurement I stop the measurement during the freeze without pressing CTRL +C

3. Same as 2, but I pressed CTRL + C during the freeze. Then I stopped the measurement. Maybe that is better compareable to the Windows 10 log.

[Wireshark_Measurements.zip](https://github.com/hbldh/bleak/files/14892604/Wireshark_Measurements.zip)

dlech, 2024-04-06:
Thanks for the logs. I can see in the WIndows 11 logs, there is an LE Connection Update request that is never answered. So this is likely the cause of the hang.

[image]

Since you are using a Rapsberry Pi, you should be able to fix the program running there to respond to this.

Quaditz, 2024-04-07:
Thanks for the answer!

I will check how to answer that with aioble on the Raspberry PI Pico.

This might mean that Windows 10 is not sending this Update Request but the same Bleak Version in Windows 11 is then requesting this update request?

In any case, I think Bleak should return some kind of timeout then when this is not answered. Maybe this would be an improvement.

Especially because all the examples for BLE connections for the Aioble Library on the Raspberry PI never mention any need for answering connection updates. 

This might be a more common use-case then, if this could correspond to an error or warning etc in Bleak this would be aweseome :)

Quaditz, 2024-04-08:
Thank you @dlech for the insights! 

As you saw I created also an issue on the micro python side since there is no way in Aioble to answer that Connection Update request (as you said it should be done anyway automatically by the Low Level Bluetooth Libraries).

You mentioned that Microphyton-lib is the wrong place for that issue. I will move that issue to the Micropython repo then :)