hbldh/bleak #1264: Error when trying to read the same characteristic with another device

nicolabattaglino, 2023-03-24:
* bleak version: 0.19.5
* Python version: 3.10.0
* Operating System: MacOS 11.6
* BlueZ version (`bluetoothctl -v`) in case of Linux:

**Description**

I'm trying to read the heart rate characteristic from various devices. I tried with a polar band H10 and an Amazfit GTS2 and everything works as expected, I receive the heart rate data. I encountered some problems when trying to read the same characteristic from the Xiaomi Mi Band 6. The pc connects correctly to the device but then the following error arises.

The script I'm using is the one in the folder 'examples' here on GitHub, named 'enable_notifications.py'
https://github.com/hbldh/bleak/blob/develop/examples/enable_notifications.py

The command I run (Bluetooth address is omitted here):
```
./notifications.py --address "xxxaddressxxx"  "00002A37-0000-1000-8000-00805F9B34FB"
```

**Logs**

```2023-03-24 16:43:38,427 __main__ INFO: starting scan...
2023-03-24 16:43:45,082 __main__ INFO: connecting to device...
2023-03-24 16:43:49,184 __main__ INFO: Connected
Traceback (most recent call last):
  File "/notifications.py", line 95, in <module>
    asyncio.run(main(args))
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 641, in run_until_complete
    return future.result()
  File "/notifications.py", line 47, in main
    await client.start_notify(args.characteristic, notification_handler)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/__init__.py", line 649, in start_notify
    await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/backends/corebluetooth/client.py", line 360, in start_notify
    await self._delegate.start_notifications(characteristic.obj, callback)
  File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/bleak/backends/corebluetooth/PeripheralDelegate.py", line 221, in start_notifications
    await future
bleak.exc.BleakError: Failed to update the notification status for characteristic 40: Error Domain=CBErrorDomain Code=1 "One or more parameters were invalid." UserInfo={NSLocalizedDescription=One or more parameters were invalid.}
```

dlech, 2023-03-24:
Did you try https://bleak.readthedocs.io/en/latest/troubleshooting.html?

I suspect logging Bluetooth packets could give some insight if the error is coming from the heart rate monitor itself.

Also try Bleak 0.20.0.

nicolabattaglino, 2023-03-24:
I tried to upgrade bleak, but I obtain the same error.
In the following, there is a screenshot of the Bluetooth packets. 

[image]

dlech, 2023-03-24:
It looks like Bleak is just passing through the error from the device as it should. Does this device need to be paired/bonded first (although in that case, the device should technically be giving a different error that would trigger pairing automatically on macOS)?