hbldh/bleak #1272: Repeated BLE scanning causes BleakDBusError on Raspberry Pi

saaz-gustav, 2023-04-05:
* bleak version: 0.20.1
* Python version: 3.8.10/3.10.6
* Operating System: Linux, Ubuntu Server 20.04/22.04 LTS
* BlueZ version: 5.53/5.64

### Description

I am writing a Python application that is intended to run continuously on a RPi. It is supposed to regularly (lets say every 10 sec) perform a BLE scan, and is using bleak to do so.  The scan is performed using a start-sleep-stop scan sequence without detection callbacks. After a while (~30 mins), scanner.start() crashes due to a BleakDBusError Exception:   
`bleak.exc.BleakDBusError: [org.freedesktop.DBus.Error.AccessDenied] Client tried to send a message other than Hello without being registered` 

This behavior is consistent for two different RPi models (3B+/4), Ubuntu versions (20.04/22.04 LTS), blueZ versions (5.53/5.64) and python versions (3.8.10/3.10.6), which raises the question if this a bleak problem or if I am doing something that I shouldn't be doing, e.g. with regards to the asyncio library. See example below. Any help is greatly appreciated.

### What I Did
Minimum working example:
```
import time
import asyncio 
from bleak import BleakScanner

scanner = BleakScanner()

def main():
    while True:
        time.sleep(3)       # simulating the other parts of the application
        print("Scanning...")
        asyncio.run(scan())

async def scan():
    await scanner.start()
    await asyncio.sleep(2)   # scan time
    await scanner.stop()
     

if __name__=='__main__':
    main()
```
### Logs
There are no apparent clues in the log prints using export BLEAK_LOGGING=1. No info with regards to the raised exception/error. No wireshark logs available as of now, but if they are needed to for troubleshooting I will try to acquire them.

saaz-gustav, 2023-04-05:
Ran `sudo dbus-monitor --system`:  
```
method call time=1680696376.549511 sender=:not.active.yet -> destination=org.freedesktop.DBus serial=1 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=Hello
error time=1680696376.549572 sender=org.freedesktop.DBus -> destination=:not.active.yet error_name=org.freedesktop.DBus.Error.LimitsExceeded reply_serial=1
   string "The maximum number of active connections for UID 1000 has been reached"
method call time=1680696376.552528 sender=:not.active.yet -> destination=org.freedesktop.DBus serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "type=signal,interface=org.freedesktop.DBus.ObjectManager,member=InterfacesAdded,arg0path=/org/bluez/"
error time=1680696376.552611 sender=org.freedesktop.DBus -> destination=:not.active.yet error_name=org.freedesktop.DBus.Error.AccessDenied reply_serial=2
   string "Client tried to send a message other than Hello without being registered"
method call time=1680696381.569815 sender=:not.active.yet -> destination=org.freedesktop.DBus serial=1 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=Hello
error time=1680696381.569899 sender=org.freedesktop.DBus -> destination=:not.active.yet error_name=org.freedesktop.DBus.Error.LimitsExceeded reply_serial=1
   string "The maximum number of active connections for UID 1000 has been reached"
method call time=1680696381.573091 sender=:not.active.yet -> destination=org.freedesktop.DBus serial=2 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=AddMatch
   string "type=signal,interface=org.freedesktop.DBus.ObjectManager,member=InterfacesAdded,arg0path=/org/bluez/"
error time=1680696381.573177 sender=org.freedesktop.DBus -> destination=:not.active.yet error_name=org.freedesktop.DBus.Error.AccessDenied reply_serial=2
   string "Client tried to send a message other than Hello without being registered"
  ```
 `The maximum number of active connections for UID 1000 has been reached` seems like a bad thing.

saaz-gustav, 2023-04-05:
Seems to be a missing disconnect from the dbus on scan stop()?
Added `self._bus.disconnect()` in BlueZManager -> active_scan() -> stop() and now it seems to be working. Before that, the crash happened when starting scan number 256. I guess that's the max number of possible active connections.

dlech, 2023-04-05:
You are not using asyncio correctly. `asyncio.run()` should only be used once at the top level of the application. `BlueZManager` is designed to only have one instance per application that never closes.

```python
import asyncio 

from bleak import BleakScanner

async def main():
    while True:
        await asyncio.sleep(3)       # simulating the other parts of the application
        print("Scanning...")
        
        devices = await BleakScanner.disover(timeout=2)
        
        # do stuff with discovered devices
        ...

if __name__=='__main__':
    asyncio.run(main())
```