hbldh/bleak #1280: Got exception: OSError: [WinError -2147483629] The object has been closed

megholm, 2023-04-14:
* bleak version: 0.20.1
* Python version: 3.10.8 and 3.11.2
* Operating System: Windows 10 Enterprise 22H2
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I'm trying to figure out why I have a higher connect-success on one board and not another. 
Same software on board, same software on PC.

I have two Silabs EFR32MG24 boards (efr32xg24 Explorer Kit, https://www.silabs.com/development-tools/wireless/efr32xg24-explorer-kit?tab=overview) for which I have difficulties (on both) in obtaining a connection.

For both, I get an OSError-exception when calling BleakClient.connect():

OSError: [WinError -2147483629] The object has been closed

However, for one board, I get much fewer of these exceptions, and for those I get, the verbose output from Bleak is somewhat different.

### What I Did

This is my simple application - it tries to connect and disconnect again and again:

```
import asyncio
import logging
import sys

from bleak import BleakClient

# address = "34:25:B4:A0:B8:0F" # The good board
address = '34:25:B4:A0:B1:CB' # The bad board

logger = logging.getLogger(__name__)

async def main(address):
    connect_success_count = 0
    connect_fail_count = 0

    while True:
        logger.info('Stats: %u/%u success', connect_success_count, (connect_success_count+connect_fail_count) )

        logger.info("Connecting...")
        client = BleakClient(address)
        try:
            await client.connect()
            logger.info(f"Connected: {client.is_connected}")
            connect_success_count += 1

        except Exception as e:
            logger.warning('Got exception: %s: %s', type(e).__name__, str(e))
            connect_fail_count += 1

        finally:
            try:
                await client.disconnect()
            except Exception as e:
                logger.warning('Got exception during disconnect: %s: %s', type(e).__name__, str(e))
            logger.info('Disconnected...')

if __name__ == "__main__":
    LOGFORMAT=' %(asctime)s.%(msecs)03d [%(thread)05d] %(name).8s:\t[%(levelname).1s] %(message)s'
    logging.basicConfig(level=logging.DEBUG, format=LOGFORMAT, datefmt='%H:%M:%S')

    logger.info('Main running - Python %s', sys.version)
    asyncio.run(main(address))
```

Running this against the "good board" (34:25:B4:A0:B8:0F) I get a connect success of ~95%:
`15:31:53.950 [22312] __main__: [I] Stats: 119/125 success`
while the same test on the "bad board" (34:25:B4:A0:B1:CB) I get a connect success of ~73%:
[... 94 lines cut ...]
 15:42:25.261 [12684] bleak.ba: [D] calling get_descriptors_async
 15:42:25.289 [12684] bleak.ba: [D] returned from get_descriptors_async
 15:42:25.290 [12684] bleak.ba: [D] calling get_descriptors_async
 15:42:25.320 [12684] bleak.ba: [D] returned from get_descriptors_async
 15:42:25.321 [12684] bleak.ba: [D] calling get_descriptors_async
 15:42:25.350 [12684] bleak.ba: [D] returned from get_descriptors_async
 15:42:25.351 [12684] bleak.ba: [D] calling get_descriptors_async
 15:42:25.387 [12684] bleak.ba: [D] returned from get_descriptors_async
 15:42:25.388 [12684] bleak.ba: [D] calling get_descriptors_async
 15:42:25.417 [12684] bleak.ba: [D] returned from get_descriptors_async
 15:42:25.417 [12684] bleak.ba: [D] calling get_characteristics_async
 15:42:25.456 [12684] bleak.ba: [D] returned from get_characteristics_async
 15:42:25.456 [12684] bleak.ba: [D] calling get_descriptors_async
 15:42:25.469 [12684] bleak.ba: [D] returned from get_descriptors_async
 15:42:25.471 [12684] __main__: [I] Connected: True
 15:42:25.472 [12684] bleak.ba: [D] Disconnecting from BLE device...
 15:42:28.545 [09720] bleak.ba: [D] max_pdu_size_changed_handler: 23
 15:42:28.546 [31160] bleak.ba: [D] session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x00000175150053F0>, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED
 15:42:28.546 [12684] bleak.ba: [D] closing requester
 15:42:28.546 [12684] bleak.ba: [D] closing session
 15:42:28.548 [12684] __main__: [I] Disconnected...
 15:42:28.548 [12684] __main__: [I] Stats: 32/42 success
```

Any idea on how to deal/proceed from here?

dlech, 2023-04-14:
Could you please try #1281 and see if it make a difference on the "good" board?

For the "bad" board, try logging Bluetooth packets to see what is the difference between a successful attempt and not successful attempt.

Similar reports:
- https://github.com/hbldh/bleak/discussions/1245
- https://github.com/hbldh/bleak/issues/1198

megholm, 2023-04-14:
Hi David,

I will! Thanks for the quick response 🙏

megholm, 2023-04-17:
David,

1281 did it for the good board! Thanks.

(Reproduced the error with the old code, changed to new code, ran 400 connections without problems, went back again to old code, and reproduced the error again [within 130 connect-attempts], upgraded to new code again, and verified ~800 successfull connections...)

megholm, 2023-04-17:
(Reopened, for the "bad"-board)
With respect to the bad board, I've done as specified in the Troubleshooting section and have logged the packages (more on that below).

First of all, it seems (!) that applying #1281 changed the error from "OSError" to "Timeout (but more on than further down):

BEFORE 1281:
```
 13:21:43.806 [22932] __main__: [I] Connecting...
 13:21:44.258 [22932] bleak.ba: [D] Received 34:25:B4:A0:B1:CB: Empty Example.
 13:21:44.259 [22932] bleak.ba: [D] Received 34:25:B4:A0:B1:CB: .
 13:21:44.260 [22932] bleak.ba: [D] 6 devices found. Watcher status: 3.
 13:21:44.260 [22932] bleak.ba: [D] Connecting to BLE device @ 34:25:B4:A0:B1:CB
 13:21:44.278 [22932] bleak.ba: [D] getting services (service_cache_mode=None, cache_mode=None)...
 13:21:44.278 [22932] bleak.ba: [D] calling get_gatt_services_async
 13:21:44.369 [15644] bleak.ba: [D] session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000002A88EC59F30>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
 13:21:53.968 [17036] bleak.ba: [D] session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000002A88EC59750>, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED
 13:21:53.969 [22932] bleak.ba: [D] closing requester
 13:22:01.668 [22932] bleak.ba: [D] closing session
 13:22:01.669 [22932] __main__: [W] Got exception: OSError: [WinError -2147483629] The object has been closed
 13:22:01.670 [22932] bleak.ba: [D] Disconnecting from BLE device...
``` 

AFTER 1281:
```
 13:05:18.672 [13044] __main__: [I] Connecting...
 13:05:18.906 [13044] bleak.ba: [D] Received 34:25:B4:A0:B1:CB: Empty Example.
 13:05:18.917 [13044] bleak.ba: [D] Received 34:25:B4:A0:B1:CB: .
 13:05:18.917 [13044] bleak.ba: [D] 1 devices found. Watcher status: 3.
 13:05:18.918 [13044] bleak.ba: [D] Connecting to BLE device @ 34:25:B4:A0:B1:CB
 13:05:18.935 [13044] bleak.ba: [D] getting services (service_cache_mode=None, cache_mode=None)...
 13:05:18.935 [13044] bleak.ba: [D] calling get_gatt_services_async
 13:05:19.107 [27160] bleak.ba: [D] session_status_changed_event_handler: id: BluetoothLE#BluetoothLEe0:d0:45:43:2e:b0-34:25:b4:a0:b1:cb, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
 13:05:28.713 [20344] bleak.ba: [D] session_status_changed_event_handler: id: BluetoothLE#BluetoothLEe0:d0:45:43:2e:b0-34:25:b4:a0:b1:cb, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED
 13:05:38.321 [20344] bleak.ba: [D] session_status_changed_event_handler: id: BluetoothLE#BluetoothLEe0:d0:45:43:2e:b0-34:25:b4:a0:b1:cb, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
 13:05:38.329 [20344] bleak.ba: [D] max_pdu_size_changed_handler: 247
 13:05:38.568 [13044] bleak.ba: [D] closing requester
 13:05:38.582 [13044] bleak.ba: [D] closing session
 13:05:38.583 [13044] __main__: [W] Got exception: TimeoutError:
 13:05:38.583 [13044] bleak.ba: [D] Disconnecting from BLE device...
```

Also note that there is this additional line (every time I try with 1281):
13:05:38.329 [20344] bleak.ba: [D] max_pdu_size_changed_handler: 247

But I reckon that comes with the "fact" that 1281 actually makes "bleak-winrt" reestablish the connection/retry connecting after the initial failure (after 10s).

Regardless, I think the focus should be on the first 10 seconds of the failing connect-attempt (and not on the last 10s where #1281 makes a slight difference).
 

WIRESHARK 

From the Wireshark captures, it seems that this BAD board does not reply to the "Sent Exchange MTU Request" (when failing).
E.g. here there is 10 seconds with no response from the board:

BAD Board - failed connect (resulting in OSError):
```
854	20.788837	localhost ()	SiliconL_a0:b1:cb (Empty Example)	ATT	12	Sent Exchange MTU Request, Client Rx MTU: 527
855	30.391960	controller	host	HCI_EVT	7	Rcvd Disconnect Complete
```
[timeout_on_bad_board_1281_around_20s_02.pcapng]

BAD Board - successfull connect:

```
648	15.573767	localhost ()	SiliconL_a0:b1:cb (Empty Example)	ATT	12	Sent Exchange MTU Request, Client Rx MTU: 527
649	15.577344	SiliconL_a0:b1:cb (Empty Example)	localhost ()	ATT	12	Rcvd Exchange MTU Request, Client Rx MTU: 247
650	15.577619	localhost ()	SiliconL_a0:b1:cb (Empty Example)	ATT	12	Sent Exchange MTU Response, Server Rx MTU: 527
651	15.578871	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
652	15.584955	SiliconL_a0:b1:cb (Empty Example)	localhost ()	L2CAP	21	Rcvd Connection Parameter Update Request
653	15.584971	localhost ()	SiliconL_a0:b1:cb (Empty Example)	L2CAP	15	Sent Connection Parameter Update Response (Accepted)
[... 4 lines cut ...]
```
[timeout_on_bad_board_1281_around_20s_02.pcapng]

There exists a small variation of the successfull connect - at least according to Wireshark - where the Exchange MTU Request is received BEFORE being sent:

```
424	20.772067	SiliconL_a0:b1:cb (Empty Example)	localhost ()	ATT	12	Rcvd Exchange MTU Request, Client Rx MTU: 247
425	20.772280	controller	host	HCI_EVT	14	Rcvd Command Complete (LE Read Channel Map)
426	20.772811	localhost ()	SiliconL_a0:b1:cb (Empty Example)	ATT	12	Sent Exchange MTU Request, Client Rx MTU: 527
427	20.773360	localhost ()	SiliconL_a0:b1:cb (Empty Example)	ATT	12	Sent Exchange MTU Response, Server Rx MTU: 527
428	20.781244	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
429	20.782276	controller	host	HCI_EVT	14	Rcvd LE Meta (LE Data Length Change)
[... cut ...]
[os_error_on_bad_board_after_30s_without_1281_01.pcapng]

From all this, can "we" conclude anything?

[os_error_on_bad_board_after_30s_without_1281_01.zip](https://github.com/hbldh/bleak/files/11251095/os_error_on_bad_board_after_30s_without_1281_01.zip)
[timeout_on_bad_board_1281_around_20s_02.zip](https://github.com/hbldh/bleak/files/11251096/timeout_on_bad_board_1281_around_20s_02.zip)
[timeout_on_bad_board_1281_around_20s_02.txt](https://github.com/hbldh/bleak/files/11251100/timeout_on_bad_board_1281_around_20s_02.txt)

dlech, 2023-04-17:
Thanks for testing the PR. I think it might be better to continue the discussion for the timeout error on one of the existing similar issues so that it will be easier for others with the same problem to find it. And this issue will get automatically closed when I merge the PR.

Although based on your description and the wireshark capture, it does seem the problem in this case is that the device is simply not responding, so the issue is likely in the device firmware or the adapter firmware. If you have a Bluetooth sniffer, you can use it to see what is actually going over the air to determine if it isn't being sent at all or if it is getting dropped by the adapter.

megholm, 2023-04-17:
OK, I will move the thread to one of the other issues.

Do you have any comment on the MTU values of 527?
From where, and why, they origin?

(the strange thing is that the two devices have at least identical
application firmware, but I need to check whether the radio fw is the
same...)

dlech, 2023-04-17:
The number 527 comes from Windows. The max PDU size for BLE in general is 512 bytes and there are 3 bytes of overhead, so in general the max MTU an OS should be requesting is 515 bytes. Not sure why Windows went with 527, but it is in the ballpark.

If all of the firmware is the same, maybe there could be some sort of race condition or something going in to low power mode?

PR #1283 (2023-04-19): Release/v0.20.2

Fixed
-----
- Fixed ``org.bluez.Error.InProgress`` in characteristic and descriptor read and
  write methods in BlueZ backend.
- Fixed ``OSError: [WinError -2147483629] The object has been closed`` when
  connecting on Windows. Fixes #1280.
closed 2023-04-19 by dlech