hbldh/bleak #1284: Debug output with Python 3.11 could be improved to match 3.10

megholm, 2023-04-20:
* bleak version: 0.20.2
* Python version: 3.11.3
* Operating System: Windows
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

With Python 3.10.x some lines looked like this:
```
10:36:52.675 [07396] bleak.ba: [D] session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000001976F8C8A70>, error: BluetoothError.SUCCESS, status: GattSessionStatus.ACTIVE
 10:36:56.827 [07396] bleak.ba: [D] session_status_changed_event_handler: id: <_bleak_winrt_Windows_Devices_Bluetooth.BluetoothDeviceId object at 0x000001976F8C8C50>, error: BluetoothError.SUCCESS, status: GattSessionStatus.CLOSED
```
With Python 3.11.x the str(IntEnum) output has changed and this lines become:

```
 11:34:27.057 [04408] bleak.ba: [D] session_status_changed_event_handler: id: BluetoothLE#BluetoothLEe0:d0:45:43:2e:b0-34:25:b4:a0:b8:0f, error: 0, status: 1
 11:34:31.595 [28732] bleak.ba: [D] session_status_changed_event_handler: id: BluetoothLE#BluetoothLEe0:d0:45:43:2e:b0-34:25:b4:a0:b8:0f, error: 0, status: 0
```
(Look at error- and status-values).

Consider just adding ".name" to the logger-line:
```
        def session_status_changed_event_handler(
            sender: GattSession, args: GattSessionStatusChangedEventArgs
        ):
            logger.debug(
                "session_status_changed_event_handler: id: %s, error: %s, status: %s",
                sender.device_id.id,
                args.error.name,
                args.status.name,
            )
            loop.call_soon_threadsafe(handle_session_status_changed, args)
```

dlech, 2023-04-20:
Perhaps even better would be to change `%s` to `%r` so we get the repr which shows both values instead of one or the other?

megholm, 2023-04-20:
Sure. Anything but the raw integer values... Go ahead! :-)

PR #1410 (2023-09-02): Release/0.21.0

Added
-----
* Added ``bleak.uuids.normalize_uuid_16()`` function.
* Added ``bleak.uuids.normalize_uuid_32()`` function.
* Added ``advertisement_data()`` async iterator method to ``BleakScanner``. Merged #1361.
* Added type hints for kwargs on ``BleakScanner`` class methods.
* Added support for Python 3.12.

Changed
-------
* Improved error messages when failing to get services in WinRT backend.
* Improved error messages with enum values in WinRT backend. Fixes #1284.
* Scanner backends modified to allow multiple advertisement callbacks. Merged #1367.
* Changed default handling  [... cut]
closed 2023-07-20 by dlech