hbldh/bleak #1286: bleak can not work on macOS Catalina 10.15.3

volca, 2023-04-24:
* bleak version: 0.19.1
* Python version: 3.10.6_1
* Operating System: macOS Catalina 10.15.3
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

bleak can not work with homeassistant 2023.4.5.  

### Logs

The log

```
AttributeError: 'NSKVONotifying_CBCentralManager' object has no attribute 'retrieveAddressForPeripheral_'
2023-04-24 09:00:09.579 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback CentralManagerDelegate.did_discover_peripheral(<CBCentralMan...x7fd8f8c93d20>, <CBPeripheral... disconnected>, {
    kCBAdvD... 01fea2c5 };
}, -74)
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.10/3.10.6_1/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/Users/xxx/.homeassistant/deps/lib/python/site-packages/bleak/backends/corebluetooth/CentralManagerDelegate.py", line 265, in did_discover_peripheral
    callback(peripheral, advertisementData, RSSI)
  File "/Users/xxx/.homeassistant/deps/lib/python/site-packages/bleak/backends/corebluetooth/scanner.py", line 136, in callback
```

dlech, 2023-04-24:
As mentioned in the docs, this is using an undocumented method and is not guaranteed to work. I would suggest to Home Assistant to have a fallback instead of relying on this hack.

https://bleak.readthedocs.io/en/latest/backends/macos.html#bleak.backends.corebluetooth.scanner.CBScannerArgs.use_bdaddr

dlech, 2023-07-19:
I will accept a pull request if someone with an older version of macOS can figure out a workaround, but I don't have anything older than macOS 12 and it is working.

closed 2023-07-19 by dlech