hbldh/bleak #1291: Windows : Failed to connect to device: Could not get GATT characteristics

pymenow, 2023-04-28:
* bleak version: 0.20.2
* Python version: 3.9.13
* Operating System: Windows 11 22621.1635
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
Failed to connect to device: Could not get GATT characteristics for <_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattDeviceService object at 0x00000206F9189750>: Access Denied

The same code works in Linux but issues seen in windows.

### What I Did

Just attempting to connect to the BLE device . Below is the function I am using to connect.

```
async def omn_connect(address: str, reconnect_delay: int = 5) -> Optional[BleakClient]:
    async def on_disconnect(client, error):
        print(f"Device disconnected, error: {error}")

    try:
        print(f"Attempting to connect to device {address}...")
        client = BleakClient(address, disconnected_callback=lambda _: asyncio.ensure_future(on_disconnect(client, _)))
        await client.connect()
        print("Connected")
        
        return client
    except Exception as e:
        print(f"Failed to connect to device: {e}")
        return None

async def omn_disconnect(client: BleakClient):
    try:
        await client.disconnect()
        print("Disconnected successfully")
    except Exception as e:
        print(f"Disconnection failed: {e}")
```

### Logs

Added logs below .
[blelogs.txt](https://github.com/hbldh/bleak/files/11350720/blelogs.txt)

pymenow, 2023-04-28:
[GATT-Wider.zip](https://github.com/hbldh/bleak/files/11351002/GATT-Wider.zip)
Added extended logs for Wireshark where I pair with the device via Windows Bluetooth and then start the bleak script to connect before I run into the same issue.

dlech, 2023-04-28:
The program is not complete, so it is impossible to say what could be happening.

Please provide a full minimal reproducible test case.

pymenow, 2023-04-28:
Hello here is a fully reproducible test .

`import asyncio
from bleak import BleakClient

async def discover_services_and_handles(device_address):
    async with BleakClient(device_address) as client:
        services = await client.get_services()

        for service in services:
            print(f"Service: {service.uuid}")
            for characteristic in service.characteristics:
                print(f"  Characteristic: {characteristic.uuid}, Handle: {characteristic.handle}")

device_address = "XX:XX:XX:XX:XX:XX"  # Replace with your device's address
asyncio.run(discover_services_and_handles(device_address))`

pymenow, 2023-04-29:
Thanks - Here is the output -
bleak.exc.BleakError: Could not get GATT characteristics for 0000181e-0000-1000-8000-00805f9b34fb (68): Access Denied

dlech, 2023-04-29:
Thanks. Could you please test https://github.com/hbldh/bleak/pull/1294 to see if it fixed the problem?

pymenow, 2023-04-30:
Thanks , was able to connect to the device successfully with this change.
Its likely since 0000181e is a bond management servcie - Windows want to handle this and hence the restriction ?

dlech, 2023-04-30:
Great, thanks for testing! Yes, it seems Windows gives an error for services like this that are "owned" by the OS while other OSes just hide them completely.

PR #1410 (2023-09-02): Release/0.21.0

Added
-----
* Added ``bleak.uuids.normalize_uuid_16()`` function.
* Added ``bleak.uuids.normalize_uuid_32()`` function.
* Added ``advertisement_data()`` async iterator method to ``BleakScanner``. Merged #1361.
* Added type hints for kwargs on ``BleakScanner`` class methods.
* Added support for Python 3.12.

Changed
-------
* Improved error messages when failing to get services in WinRT backend.
* Improved error messages with enum values in WinRT backend. Fixes #1284.
* Scanner backends modified to allow multiple advertisement callbacks. Merged #1367.
* Changed default handling  [... cut]
closed 2023-04-30 by dlech