hbldh/bleak #1296: Access to Raw AdvertisementData

ztroop, 2023-04-30:
* bleak version: 0.19.5
* Python version: 3.10.6
* Operating System: Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.64

## Description

I've been advised that I can read the advertisement data from the device and use byte 7 (of 8 in total) for the battery "status". 
However, when I read the `AdvertisementData`.. I don't have raw access over this data:

`AdvertisementData(local_name='Device XYZ', manufacturer_data={820: b'\r\x1c\xa5\xae\t\x00'}, service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', 'b42e1c08-ade7-11e4-89d3-123b93f75cba', 'f000ffc0-0451-4000-b000-000000000000'], rssi=-39)`

Based on what I'm trying to do, does anyone have suggestions?

dlech, 2023-04-30:
I'm guessing what you are looking for is the `manufacturer_data`.  Two bytes are the manufacturer ID (820 in this case) and the remaining 6 bytes are the data (the bytes string). So probably `battery_status = manufacturer_data[820][5]`.

ztroop, 2023-04-30:
What about bitmasking of that value? Example:

```
0x00 = full
0x01 = moderate
0x10 = low
0x11 = stopped
```

dlech, 2023-04-30:
Without knowing anything about the device, it is hard to say.

ztroop, 2023-04-30:
Alright, I'll continue on my own on this. Appreciate the timely input.

closed 2023-04-30 by ztroop