hbldh/bleak #1301: disconnect callback code do not match documentation

rosaschafe, 2023-05-07:
* bleak version: bleak==0.20.2
* Python version: Python 3.10.10
* Operating System: ArchLinux
* BlueZ version (`bluetoothctl -v`) in case of Linux: bluetoothctl: 5.66

### Description
https://github.com/hbldh/bleak/blob/8948d5dca8e53761c614092b4af93c7173e23b9d/bleak/__init__.py#L390-L393

https://github.com/hbldh/bleak/blob/8948d5dca8e53761c614092b4af93c7173e23b9d/bleak/backends/bluezdbus/client.py#L53-L55

https://github.com/hbldh/bleak/blob/8948d5dca8e53761c614092b4af93c7173e23b9d/bleak/backends/bluezdbus/client.py#L162-L163

### What I Did
My disconnect callback is async
```
class BleManager:
    async def _xxxx_disconnect(self, device):
         print(f"Disconnected {device}")

    async def setup_xxx(self, device):
       BleakClient(device, self._xxxx_disconnect)
```

### Logs

```
RuntimeWarning: coroutine 'BleManager._xxxx_disconnect' was never awaited
  self._disconnected_callback()
```

dlech, 2023-05-07:
Your error is because the callback cannot be `async`.

`self._disconnected_callback` is not the function you passed in but rather a `functools.partial` wrapper that includes the argument from the documentation.

rosaschafe, 2023-05-08:
Thank you @dlech. Can you explain what do "Callback that will be scheduled in the event loop when the client is disconnected" means in the documentation?

dlech, 2023-05-08:
On some OSes, the callback from the OS that notifies when a device has disconnected happens on a background thread. Bleak takes care of scheduling a task in the asyncio event loop from this callback so that you don't have to worry about threading issues. The callback you provide is guaranteed to be running on the asyncio event loop thread.

closed 2023-07-19 by dlech