hbldh/bleak #1311: client.write_gatt_char [WinError -2147483629] object close

xiasanshi, 2023-05-15:
* bleak version:  bleak==0.20.2
                           bleak-winrt==1.2.0
* Python version: 3.10.5
* Operating System: Windows 10 (Windows Feature Experience Pack 1000.19041.1000.0)
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
Traceback (most recent call last):

  File "D:\develop\py\Flappy-bird-python\app\service\imu_service.py", line 152, in init_device
    await self.init_device(fd, retry-1)
          │    │           │   └ 4
          │    │           └ <_io.BufferedWriter name='C:/Users/Administrator/nx-game/data\\20230515-103758\\imu-im948-LeftFore-V3.01-30-20230515-103831.m...
          │    └ <function IMUService.init_device at 0x0000023ED4A367A0>
          └ <app.service.imu_service.IMUService object at 0x0000023ED4F4D000>

> File "D:\develop\py\Flappy-bird-python\app\service\imu_service.py", line 130, in init_device
    await self._imu_client.write_gatt_char(par_write_characteristic, wakestr)
          │    │           │               │                         └ b')'
          │    │           │               └ 5
          │    │           └ <function BleakClient.write_gatt_char at 0x0000023EAD002D40>
          │    └ <BleakClient, 26:C0:10:F7:CE:EC, <class 'bleak.backends.winrt.client.BleakClientWinRT'>>
          └ <app.service.imu_service.IMUService object at 0x0000023ED4F4D000>

  File "D:\develop\py\Flappy-bird-python\venv\lib\site-packages\bleak\__init__.py", line 659, in write_gatt_char
    await self._backend.write_gatt_char(char_specifier, data, response)
          │    │        │               │               │     └ False
          │    │        │               │               └ b')'
          │    │        │               └ 5
          │    │        └ <function BleakClientWinRT.write_gatt_char at 0x0000023ED4F4E950>
          │    └ <bleak.backends.winrt.client.BleakClientWinRT object at 0x0000023ED4F64070>
          └ <BleakClient, 26:C0:10:F7:CE:EC, <class 'bleak.backends.winrt.client.BleakClientWinRT'>>

  File "D:\develop\py\Flappy-bird-python\venv\lib\site-packages\bleak\backends\winrt\client.py", line 874, in write_gatt_char
    await characteristic.obj.write_value_with_result_async(buf, response),
          │              │   │                             │    └ <GattWriteOption.WRITE_WITHOUT_RESPONSE: 1>
          │              │   │                             └ <_bleak_winrt_Windows_Storage_Streams.Buffer object at 0x0000023ED4F1E410>
          │              │   └ <method 'write_value_with_result_async' of '_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattCharacteristic...
          │              └ <_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattCharacteristic object at 0x0000023ED4F1E8B0>
          └ <bleak.backends.winrt.characteristic.BleakGATTCharacteristicWinRT object at 0x0000023ED4F679A0>

OSError: [WinError -2147483629] 该对象已关闭。
### What I Did

`python
par_write_characteristic=0x0005
wakestr=bytes([0x29])
self._imu_client = BleakClient(device, disconnected_callback=self.disconnected_callback, timeout=5.0, )
await self._imu_client.connect()
await self._imu_client.write_gatt_char(par_write_characteristic, wakestr)
`

### Logs

2023-05-15 10:38:03.259 | INFO     | app.service.imu_service:connect:90 - connect device im948-LeftFore-V3.01
 10:38:03 client.py[line:253] DEBUG Connecting to BLE device @ 26:C0:10:F7:CE:EC
 10:38:03 client.py[line:632] DEBUG getting services (service_cache_mode=None, cache_mode=None)...
 10:38:03 client.py[line:656] DEBUG calling get_gatt_services_async
 10:38:03 client.py[line:692] DEBUG returned from get_gatt_services_async
 10:38:03 client.py[line:704] DEBUG calling get_characteristics_async
 10:38:03 client.py[line:712] DEBUG returned from get_characteristics_async
 10:38:03 client.py[line:721] DEBUG calling get_descriptors_async
 10:38:03 client.py[line:729] DEBUG returned from get_descriptors_async
 10:38:03 client.py[line:704] DEBUG calling get_characteristics_async
 10:38:05 client.py[line:712] DEBUG returned from get_characteristics_async
 10:38:05 client.py[line:721] DEBUG calling get_descriptors_async
 10:38:05 client.py[line:729] DEBUG returned from get_descriptors_async
 10:38:05 client.py[line:721] DEBUG calling get_descriptors_async
 10:38:06 client.py[line:729] DEBUG returned from get_descriptors_async
 10:38:06 client.py[line:721] DEBUG calling get_descriptors_async
[... 108 lines cut ...]
    await self._imu_client.write_gatt_char(par_write_characteristic, wakestr)
          │    │           │               │                         └ b')'
          │    │           │               └ 5
          │    │           └ <function BleakClient.write_gatt_char at 0x0000023EAD002D40>
          │    └ <BleakClient, 26:C0:10:F7:CE:EC, <class 'bleak.backends.winrt.client.BleakClientWinRT'>>
          └ <app.service.imu_service.IMUService object at 0x0000023ED4F4D000>

  File "D:\develop\py\Flappy-bird-python\venv\lib\site-packages\bleak\__init__.py", line 659, in write_gatt_char
    await self._backend.write_gatt_char(char_specifier, data, response)
          │    │        │               │               │     └ False
          │    │        │               │               └ b')'
          │    │        │               └ 5
          │    │        └ <function BleakClientWinRT.write_gatt_char at 0x0000023ED4F4E950>
          │    └ <bleak.backends.winrt.client.BleakClientWinRT object at 0x0000023ED4F64070>
          └ <BleakClient, 26:C0:10:F7:CE:EC, <class 'bleak.backends.winrt.client.BleakClientWinRT'>>

  File "D:\develop\py\Flappy-bird-python\venv\lib\site-packages\bleak\backends\winrt\client.py", line 874, in write_gatt_char
    await characteristic.obj.write_value_with_result_async(buf, response),
          │              │   │                             │    └ <GattWriteOption.WRITE_WITHOUT_RESPONSE: 1>
          │              │   │                             └ <_bleak_winrt_Windows_Storage_Streams.Buffer object at 0x0000023ED4F1EC30>
          │              │   └ <method 'write_value_with_result_async' of '_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattCharacteristic...
          │              └ <_bleak_winrt_Windows_Devices_Bluetooth_GenericAttributeProfile.GattCharacteristic object at 0x0000023ED4F1E8B0>
          └ <bleak.backends.winrt.characteristic.BleakGATTCharacteristicWinRT object at 0x0000023ED4F679A0>

OSError: [WinError -2147483629] 该对象已关闭。

dlech, 2023-05-15:
This is the same error as https://github.com/hbldh/bleak/issues/1280.

Is it fixed already in the develop branch?

jwestmoreland, 2023-08-31:
Hello All,

So #1281 isn't really fixed?  Seeing this on some PCBA's - some pass fine - running this under Win11 so that rules out (sortof) a Win11 incompatibility.  

Would like to get to the bottom of this to make sure it's a 'bug' vs. being a side-effect of a marginal PCBA.

Let me know if I can add more info that could help.

Thanks in advance.

Best Regards,
John

dlech, 2023-08-31:
Could you please post more details here on how to reproduce the problem and include as much logs as possible.

jwestmoreland, 2023-08-31:
Hello @dlech ,

Will have meeting shortly - I'll ask if it's OK or not - I'll let you know.

Thanks,
John