hbldh/bleak #1318: Problem with GATT BLOCKLIST - BLE

enrom, 2023-05-19:
Hi,

I need to interact with a bluetooth device(probe). The problem is that the descriptor UUID is one in the gatt blocklist.

I'm using the chrome builing BLE component.
    // Request the Bluetooth device through browser
    const device = await navigator.bluetooth.requestDevice({
    etc....

So when I get to this part in my code

const descriptor = await characteristic.getDescriptor('00002902-0000-1000-8000-00805f9b34fb');
var data = new Uint8Array([0x01]);
descriptor.writeValue(data)

I get the following error:
**Uncaught (in promise) DOMException: writeValue() called on blocklisted object marked exclude-writes. https://goo.gl/4NeimX**

Does anyone know any work around or anything as I'm desperate for a solution.

Thanks

nmummau, 2023-06-29:
I'm doubtful of a solution for this, but have you found anything @enrom ?

dlech, 2023-07-19:
Bleak is not related to Web Bluetooth. 

Instead of poking the client characteristic configuration, you should call [startNotifications()](https://developer.mozilla.org/en-US/docs/Web/API/BluetoothRemoteGATTCharacteristic/startNotifications).

closed 2023-07-19 by dlech