hbldh/bleak #1321: RN4678 does not find / write characteristics on Linux, works on Windows.

PMD-Crane, 2023-05-23:
* bleak version: 0.20.2
* Python version: 3.9.2
* Operating System: Raspbian GNU/Linux 11 (bullseye)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description

I am attempting to access the BLE services on a Microchip RN4678 chip. On Windows (Lenovo Thinkpad, Windows 10 22H2, same Python / bleak versions), the device's pseudo serial port characteristic can be accessed, and data transmits. On Linux, this does not work. The primary target is a Raspberry Pi 4, but I have also tested on a HP laptop running Ubuntu 22.04 (don't have it to hand today to check versions, but was fully updated), with similar unsuccessful results.

### What I Did

A minimal test case is:

```python
#! /usr/bin/env python3

import asyncio
from bleak import BleakClient

#change as appropriate
MAC = "60:8A:10:26:F4:9C"
#magic UUID for the serial service
UUID = "49535343-8841-43F4-A8D4-ECBE34729BB3"

async def test(address, char):
    async with BleakClient(address) as client:
        data = "Foobar"
        await client.write_gatt_char(
            char,
            data.encode(encoding="ascii"),
            response=True,
            )

if __name__ == "__main__":
    asyncio.run(test(MAC, UUID))
``` 

The RN4678 is hosted on a Mikroe Click evaluation board rigged up to an FTDI cable (default 115200 8N1 RTS/CTS) with 3V3 power from the RPi. Only configuration change from default on RN4678 side is `SG,1` to set the board to BLE only. RN4678 firmware v1.11 and 1.23 (the latest) have been tested, and behave similarly.

When run "Foobar" should appear on the serial link.

On Linux, it fails with an `Characteristic 49535343-8841-43F4-A8D4-ECBE34729BB3 was not found!` message. The client.services property also appears to be empty if checked.

The test works correctly on Windows.

I can manually find and write to the characteristic using `gatttool`, and using the alternative `bluepy` library works (for various reasons I can't use that for the project I'm working on).

I accept that you probably don't have the device in question, and aren't in the business of providing detailed support, but I thought I'd bring it to your attention.

### Logs

Traceback

```
Traceback (most recent call last):
  File "/home/pdavis/pythonbt/bt_minitestcase.py", line 20, in <module>
    asyncio.run(test(MAC, UUID))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pdavis/pythonbt/bt_minitestcase.py", line 13, in test
    await client.write_gatt_char(
  File "/home/pdavis/pythonbt/venv/lib/python3.9/site-packages/bleak/__init__.py", line 659, in write_gatt_char
    await self._backend.write_gatt_char(char_specifier, data, response)
  File "/home/pdavis/pythonbt/venv/lib/python3.9/site-packages/bleak/backends/bluezdbus/client.py", line 850, in write_gatt_char
    raise BleakError("Characteristic {0} was not found!".format(char_specifier))
bleak.exc.BleakError: Characteristic 49535343-8841-43F4-A8D4-ECBE34729BB3 was not found!
```

[... 793 lines cut ...]

`gatttool` characteristics list

```
[60:8A:10:26:F4:9C][LE]> characteristics
handle: 0x0002, char properties: 0x02, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0006, char properties: 0x02, char value handle: 0x0007, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x0011, char properties: 0x02, char value handle: 0x0012, uuid: 00002a29-0000-1000-8000-00805f9b34fb
handle: 0x0013, char properties: 0x02, char value handle: 0x0014, uuid: 00002a24-0000-1000-8000-00805f9b34fb
handle: 0x0015, char properties: 0x02, char value handle: 0x0016, uuid: 00002a25-0000-1000-8000-00805f9b34fb
handle: 0x0017, char properties: 0x02, char value handle: 0x0018, uuid: 00002a27-0000-1000-8000-00805f9b34fb
handle: 0x0019, char properties: 0x02, char value handle: 0x001a, uuid: 00002a26-0000-1000-8000-00805f9b34fb
handle: 0x001b, char properties: 0x02, char value handle: 0x001c, uuid: 00002a28-0000-1000-8000-00805f9b34fb
handle: 0x001d, char properties: 0x02, char value handle: 0x001e, uuid: 00002a23-0000-1000-8000-00805f9b34fb
handle: 0x001f, char properties: 0x02, char value handle: 0x0020, uuid: 00002a2a-0000-1000-8000-00805f9b34fb
handle: 0x0031, char properties: 0x0a, char value handle: 0x0032, uuid: 49535343-6daa-4d02-abf6-19569aca69fe
handle: 0x0033, char properties: 0x18, char value handle: 0x0034, uuid: 49535343-aca3-481c-91ec-d85e28a60318
handle: 0x0036, char properties: 0x3e, char value handle: 0x0037, uuid: 49535343-1e4d-4bd9-ba61-23c647249616
handle: 0x0039, char properties: 0x0c, char value handle: 0x003a, uuid: 49535343-8841-43f4-a8d4-ecbe34729bb3
handle: 0x003b, char properties: 0x18, char value handle: 0x003c, uuid: 49535343-026e-3a9b-954c-97daef17e26e
handle: 0x003e, char properties: 0x18, char value handle: 0x003f, uuid: 49535343-4c8a-39b3-2f49-511cff073b7e
```

`char-write-cmd 0x3a 0x410x420x43` will transmit "ABC" from gatttool.

dlech, 2023-05-23:
Can you share the output of the service explorer example? Are you sure this is not a BlueZ caching problem as described in the troubleshooting page of the docs?

I would suggest comparing Wireshark logs from Windows and Linux or compare `btmon` of Bleak to `btmon` of `gattool` to see what the difference is.

PMD-Crane, 2023-05-24:
I think I've found the issue, at least on the superficial layer, and it doesn't seem to be a bleak issue, just bleak's behaviour causes it to happen by default. It seems if you attempt to negotiate MTU size before reading characteristics, something goes wrong, at least on Linux. 

I realised going through the Wireshark logs that `gatttool` was reading characteristics individually, but that the bleak test case tried to read them in a batch, but then bailed out. With the bluepy test code I only called .setMTU() after I'd already run .getCharacteristics(), moving it beforehand broke that version, and adding an `mtu 500` command before calling `characteristics` broke `gatttool`.

For some reason Windows does also seem to set MTU and read characteristics in batches, but it doesn't fall over like Linux.

I've attached the Wireshark captures if they're of interest: [bttests.zip](https://github.com/hbldh/bleak/files/11554267/bttests.zip)

I guess this isn't directly a bleak issue, except perhaps for a feature request for an option to not implicitly set MTU on
connection.

Seeing as I need to file a ticket elsewhere, in you opinion (as you're probably a bit more familiar with BLE than me) is this more likely to be a messup on Microchip or Bluez's end? I'd assume Microchip as its more likely to be odd code on their end, but it does seem to be sending the first batch of characteristics before thing go wrong?

Anyway, thank you for pointing me in the right direction about where things are going wrong.

closed 2023-05-24 by dlech