hbldh/bleak #1326: Bleak on WinRT does not correctly read characteristic properties

JamsheedMistri, 2023-06-01:
* bleak version: 0.20.2
* Python version: 3.11.3
* Operating System: Windows 10 64-bit

### Description

I am trying to write a cross-platform Python script that interfaces with a BLE device. I wrote a bleak script using Python, and it works flawlessly on macOS. When I try to run it on a Windows 10 machine, I am able to connect to the device, but when I try `client.start_notify`, I get an error:
```
bleak.exc.BleakError: characteristic does not support notifications or indications
```

### What I Did

I tried debugging to see why this is the case. The value of `characteristic.properties` on macOS is:
```
['write-without-response', 'notify', 'indicate']
```

But the same exact characteristic (same exact device, verified by UUID and handle) on Windows 10 is:
```
['read']
```

I know for a fact that I can subscribe to notifications/indications on this device because I've done it on macOS successfully.

I've also tried this on 2 separate Windows machines.

### Logs
<details>
<summary>Debug logs</summary>

```
2023-06-01 14:08:06,786 bleak.backends.winrt.client Thread-1 (run) DEBUG: Connecting to BLE device @ 4C:5B:B3:A3:DC:C7
2023-06-01 14:08:06,826 bleak.backends.winrt.client Thread-1 (run) DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2023-06-01 14:08:06,826 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_gatt_services_async
2023-06-01 14:08:06,896 bleak.backends.winrt.client Dummy-2 DEBUG: session_status_changed_event_handler: id: BluetoothLE#BluetoothLEe8:2a:44:c0:71:9c-4c:5b:b3:a3:dc:c7, error: 0, status: 1
2023-06-01 14:08:06,936 bleak.backends.winrt.client Dummy-3 DEBUG: max_pdu_size_changed_handler: 247
2023-06-01 14:08:08,431 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_gatt_services_async
2023-06-01 14:08:08,431 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_characteristics_async
2023-06-01 14:08:08,501 bleak.backends.winrt.client Dummy-3 DEBUG: 4C:5B:B3:A3:DC:C7: services changed
2023-06-01 14:08:08,571 bleak.backends.winrt.client Dummy-3 DEBUG: 4C:5B:B3:A3:DC:C7: services changed
2023-06-01 14:08:08,601 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_characteristics_async
2023-06-01 14:08:08,601 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:08,611 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:08,611 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:08,611 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:08,611 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:08,621 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:08,621 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_characteristics_async
2023-06-01 14:08:08,651 bleak.backends.winrt.client Dummy-3 DEBUG: 4C:5B:B3:A3:DC:C7: services changed
2023-06-01 14:08:08,721 bleak.backends.winrt.client Dummy-3 DEBUG: 4C:5B:B3:A3:DC:C7: services changed
2023-06-01 14:08:08,741 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_characteristics_async
2023-06-01 14:08:08,741 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:08,741 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:08,741 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:08,751 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:08,751 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_characteristics_async
2023-06-01 14:08:08,801 bleak.backends.winrt.client Dummy-3 DEBUG: 4C:5B:B3:A3:DC:C7: services changed
2023-06-01 14:08:09,105 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_characteristics_async
2023-06-01 14:08:09,105 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:09,115 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:09,115 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:09,115 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:09,115 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_characteristics_async
2023-06-01 14:08:09,346 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_characteristics_async
2023-06-01 14:08:09,346 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:09,467 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:09,467 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:09,639 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:09,649 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:09,761 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:09,761 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_characteristics_async
2023-06-01 14:08:10,011 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_characteristics_async
2023-06-01 14:08:10,011 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:10,011 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
2023-06-01 14:08:10,011 bleak.backends.winrt.client Thread-1 (run) DEBUG: calling get_descriptors_async
2023-06-01 14:08:10,123 bleak.backends.winrt.client Thread-1 (run) DEBUG: returned from get_descriptors_async
LOG: characteristic.properties: ['read']
Exception in thread Thread-1 (run):
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\TS18-04\Desktop\bluetooth.py", line 91, in bluetooth_client_loop
    await client.start_notify(characteristic, bluetooth_handler, force_indicate=False)
  File "C:\Users\TS18-04\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\LocalCache\local-packages\Python311\site-packages\bleak\backends\winrt\client.py", line 934, in start_notify
    raise BleakError(
bleak.exc.BleakError: characteristic does not support notifications or indications
```
</details>

dlech, 2023-06-01:
Can you share the output of the service explorer example on both OSes?

Can you share Bluetooth packet capture on both OSes? (see troubleshooting page in docs)

JamsheedMistri, 2023-06-01:
<details>
<summary>macOS service explorer</summary>

```
$ python3 service_explorer.py --name EBS-DCC7
2023-06-01 14:12:03,540 __main__ INFO: starting scan...
2023-06-01 14:12:04,599 __main__ INFO: connecting to device...
2023-06-01 14:12:05,671 __main__ INFO: connected
2023-06-01 14:12:05,671 __main__ INFO: [Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 14): Device Information
2023-06-01 14:12:05,702 __main__ INFO:   [Characteristic] 00002a29-0000-1000-8000-00805f9b34fb (Handle: 15): Manufacturer Name String (read), Value: bytearray(b'Silicon Labs')
2023-06-01 14:12:05,731 __main__ INFO:   [Characteristic] 00002a23-0000-1000-8000-00805f9b34fb (Handle: 17): System ID (read), Value: bytearray(b'L[\xb3\xff\xfe\xa3\xdc\xc7')
2023-06-01 14:12:05,732 __main__ INFO: [Service] 331a36f5-2459-45ea-9d95-6142f0c4b307 (Handle: 19): Unknown
2023-06-01 14:12:05,732 __main__ INFO:   [Characteristic] a9da6040-0823-4995-94ec-9ce41ca28833 (Handle: 20): Unknown (write-without-response,write,notify)
2023-06-01 14:12:05,761 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 22): Client Characteristic Configuration, Value: bytearray(b'')
2023-06-01 14:12:05,762 __main__ INFO:   [Characteristic] a73e9a10-628f-4494-a099-12efaf72258f (Handle: 23): Unknown (write-without-response,notify,indicate)
2023-06-01 14:12:05,791 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 25): Client Characteristic Configuration, Value: bytearray(b'')
2023-06-01 14:12:05,821 __main__ INFO:   [Characteristic] 75a9f022-af03-4e41-b4bc-9de90a47d50b (Handle: 26): Unknown (read,write-without-response,write,notify,indicate), Value: bytearray(b'')
2023-06-01 14:12:05,851 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 28): Client Characteristic Configuration, Value: bytearray(b'')
2023-06-01 14:12:05,851 __main__ INFO: [Service] 1d14d6ee-fd63-4fa1-bfa4-8f47b42119f0 (Handle: 29): Unknown
2023-06-01 14:12:05,852 __main__ INFO:   [Characteristic] f7bf3564-fb6d-4e53-88a4-5e37e0326063 (Handle: 30): Unknown (write)
2023-06-01 14:12:05,852 __main__ INFO:   [Characteristic] 984227f3-34fc-4045-a5d0-2c581f81a153 (Handle: 32): Unknown (write-without-response,write)
2023-06-01 14:12:05,852 __main__ INFO: disconnecting...
2023-06-01 14:12:05,854 __main__ INFO: disconnected
```
</details>

<details>
<summary>Windows service explorer</summary>

```
$ python3 .\service_explorer.py --name EBS-DCC7
2023-06-01 14:29:48,110 __main__ INFO: starting scan...
2023-06-01 14:29:49,601 __main__ INFO: connecting to device...
2023-06-01 14:29:53,258 __main__ INFO: connected
2023-06-01 14:29:53,258 __main__ INFO: [Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
2023-06-01 14:29:53,258 __main__ INFO:   [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 2): Service Changed (indicate)
2023-06-01 14:29:53,369 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 4): Client Characteristic Configuration, Value: bytearray(b'\x02\x00')
2023-06-01 14:29:53,490 __main__ INFO:   [Characteristic] 00002b2a-0000-1000-8000-00805f9b34fb (Handle: 5): Database Hash (read), Value: bytearray(b'\xd6\x9c\xe5\xd8\xb8\xef\xf8<\x0cz?,:2\xb5\xfe')
2023-06-01 14:29:53,611 __main__ INFO:   [Characteristic] 00002b29-0000-1000-8000-00805f9b34fb (Handle: 7): Client Supported Features (read,write), Value: bytearray(b'\x01')
2023-06-01 14:29:53,611 __main__ INFO: [Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 9): Generic Access Profile
2023-06-01 14:29:53,731 __main__ INFO:   [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 10): Device Name (read,write), Value: bytearray(b'BGM-BGX Example')
2023-06-01 14:29:53,851 __main__ INFO:   [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 12): Appearance (read), Value: bytearray(b'\x00\x00')
2023-06-01 14:29:53,851 __main__ INFO: [Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 14): Device Information
2023-06-01 14:29:53,971 __main__ INFO:   [Characteristic] 00002a29-0000-1000-8000-00805f9b34fb (Handle: 15): Manufacturer Name String (read), Value: bytearray(b'Silicon Labs')
2023-06-01 14:29:54,091 __main__ INFO:   [Characteristic] 00002a23-0000-1000-8000-00805f9b34fb (Handle: 17): System ID (read), Value: bytearray(b'L[\xb3\xff\xfe\xa3\xdc\xc7')
2023-06-01 14:29:54,091 __main__ INFO: [Service] 331a36f5-2459-45ea-9d95-6142f0c4b307 (Handle: 19): Unknown
2023-06-01 14:29:54,091 __main__ INFO:   [Characteristic] a9da6040-0823-4995-94ec-9ce41ca28833 (Handle: 20): Unknown (write-without-response,write,notify)
2023-06-01 14:29:54,212 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 22): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2023-06-01 14:29:54,212 __main__ INFO:   [Characteristic] a73e9a10-628f-4494-a099-12efaf72258f (Handle: 23): Unknown (write-without-response,notify,indicate)
2023-06-01 14:29:54,332 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 25): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
2023-06-01 14:29:54,452 __main__ INFO:   [Characteristic] 75a9f022-af03-4e41-b4bc-9de90a47d50b (Handle: 26): Unknown (read,write-without-response,write,notify,indicate), Value: bytearray(b'')
2023-06-01 14:29:54,572 __main__ INFO:     [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 28): Client Characteristic Configuration, Value: bytearray(b'\x00\x00')
[... cut ...]
2023-06-01 14:29:54,572 __main__ INFO: disconnecting...
2023-06-01 14:29:57,698 __main__ INFO: disconnected
```
</details>

This is super strange. The characteristic is `a9da6040-0823-4995-94ec-9ce41ca28833 (Handle: 20)`, and on both operating systems, service_explorer.py shows `notify` and `indicate`. But on my script, it does not. I will take a look at the differences between my script and service_explorer.py to see if I can figure out what I'm doing differently in my code to cause this. Do you have any leads?

dlech, 2023-06-01:
Can you share a minimal script that reproduces the problem?

JamsheedMistri, 2023-06-01:
I figured it out, sorry to waste your time David!

I was indexing into the services list by doing `services[1].characteristics[1]`, because I thought the services are reported in order of handle. It seems like macOS hides some services from me (as shown in service explorer), but there are in fact more services that show up on Windows, so the correct indexing would be `services[3].characteristics[1]`. Can't believe I've been banging my head over this for a few hours 😆

closed 2023-06-01 by JamsheedMistri