hbldh/bleak #1328: Ignoring adapter argument in BleakClient

r-bt, 2023-06-08:
* bleak version: 0.20.2 
* Python version: 3.10.6
* Operating System: Raspberry Pi OS (bullseye)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.66

### Description

I've connected an external Bluetooth adapter to a Raspberry Pi 4 (which has internal Bluetooth). I'm trying to connect to a device with the internal bluetooth, however it seems to be connecting with the bluetooth adapter.

### What I Did

Running `hciconfig` I can see that there are two adapters 

```bash
hci1:	Type: Primary  Bus: USB
	BD Address: 5C:F3:70:A9:E7:FC  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:15910934 acl:330217 sco:0 events:267372 errors:0
	TX bytes:9732 acl:282 sco:0 commands:619 errors:0

hci0:	Type: Primary  Bus: UART
	BD Address: DC:A6:32:E5:4F:F0  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:5350602 acl:209517 sco:0 events:35560 errors:0
	TX bytes:16433 acl:472 sco:0 commands:290 errors:0
```

Then I run

```python
client = BleakClient(bot, adapter="hci0")
await client.connect()
print("Connected!")
wait = input("Disconnect?")
print("Disconnecting")
```

and after disconnecting running `hciconfig` again gives:

```bash
hci1:	Type: Primary  Bus: USB
	BD Address: 5C:F3:70:A9:E7:FC  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:16012231 acl:330239 sco:0 events:270334 errors:0
	TX bytes:10438 acl:304 sco:0 commands:684 errors:0

hci0:	Type: Primary  Bus: UART
	BD Address: DC:A6:32:E5:4F:F0  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:5433649 acl:209517 sco:0 events:37734 errors:0
	TX bytes:16433 acl:472 sco:0 commands:290 errors:0
```

where the `TX bytes:` for `hci0` doesn't change. I also had `bluetoothctl` running at the same time and it logs

```
Primary Service (Handle 0x0000)
	/org/bluez/hci1/dev_C6_E7_40_7B_1B_35/service0008
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
```

indicating the device was connected to using `hci1`.

### Similar issues

This appears similar to https://github.com/hbldh/bleak/issues/360

dlech, 2023-06-08:
I think you need to use the adapter arg on `BleakScanner` rather than `BleakClient`. The devices returned by the scanner know which adapter was used for scanning and will use the same adapter for connecting.

r-bt, 2023-06-09:
Yep using adapter on `BleakScanner` rather than `BleakClient` fixed the issue. Thanks

closed 2023-06-09 by r-bt