hbldh/bleak #1341: Keep connection, write

rayshen0411, 2023-06-21:
* bleak version:0.20.2
* Python version:3.11
* Operating System:Windows 11
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
1. If the device is connected, how to connect to connected/paired device again without discover?
2. How can I keep the connection, and write char data in the different session?
    Example, Run continuous 3 test cases
    Test case 1 : write data
    then Test case 2 : write data
    then Test case 3 : write data
    write data without disconnection

### What I Did

async with BleakClient(device) as client:
     write data...

### Logs

Include any relevant logs here.

Logs are essential to understand what is going on behind the scenes.

See https://bleak.readthedocs.io/en/latest/troubleshooting.html for information on how to collect debug logs.

If you receive an OS error (`WinError`, `BleakDBusError`, `NSError`, etc.), Wireshark logs of Bluetooth packets are required to understand the issue!

dlech, 2023-06-21:
> 1. If the device is connected, how to connect to connected/paired device again without discover?

This is not currently possible. Give #367 a :+1: if you would like to see this feature implemented.

> 2\. How can I keep the connection, and write char data in the different session?

You can avoid disconnection by not allowing the context manager to exit. The easiest way to do this is to wait for an `asyncio.Event`.

rayshen0411, 2023-06-22:
Hi sir

for question 1, I'll check it.
for question 2, I understand that to avoid disconnection by keeping context manager to exist, but
my scenario is for test case 1, send A data, test case 2 send B data and test case 3 send C data,
all test case is one python file

rayshen0411, 2023-06-22:
Yes, but the connection will not keep it. test every case must reconnect again.