hbldh/bleak #1342: normalize_uuid_str(id) does not support int

rianwouters, 2023-06-21:
###* bleak version:
* Python version: 3.10.5 
* Operating System: Windows
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
I want to iterate all normalized UUIDs.
Using uuid16_dict, it's keys are ints
Therefore, trying to normalize them using normalize_uuid_str requires conversion to str first or copy the implementation.
Would be more convenient if normalize_uuid would support ints, which is not very difficult to implement.

### What I Did

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well.

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here as well.
```

### Logs

Include any relevant logs here.

Logs are essential to understand what is going on behind the scenes.

See https://bleak.readthedocs.io/en/latest/troubleshooting.html for information on how to collect debug logs.

If you receive an OS error (`WinError`, `BleakDBusError`, `NSError`, etc.), Wireshark logs of Bluetooth packets are required to understand the issue!

dlech, 2023-06-21:
This is intentional. We have several methods already that treat an `int` as a handle.

closed 2023-07-19 by dlech

dlech, 2023-07-19:
FYI, a `normalize_uuid_16()` function has been added in https://github.com/hbldh/bleak/pull/1368.