hbldh/bleak #1343: Bleak dropping packets

r-bt, 2023-06-21:
* bleak version: 0.20.2
* Python version: 3.11.3
* Operating System: Raspberry Pi OS (Bullseye)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.66

### Description

I'm trying to connect to 14 BLE devices (spread over three different bluetooth adapters). Each device sends packets which end with a checksum to ensure the integrity of the data. Upon trying to connect to (around) the 12th device the previously connected devices all respond with a checksum error. Upon investigating this is because some Bluetooth packets are missing.

### What I Did

I'm working on a minimal example, it's slightly tricky as the problem only occurs when I've connected to 12 devices and am receiving data back from all of them at once – I'm using the [spherov2](https://github.com/artificial-intelligence-class/spherov2.py/tree/main) library but I'm going through it to pull out only the relevant parts I'll update when I have that

### Logs

I got logs from Wireshark and also from Bleak with `export BLEAK_LOGGING=1`

Specifically I've attached **bleak_logs.txt** which contains all of the file's logging output. As this is very large (~40MB) I've also included **filtered_bleak_logs.txt** which contains only those logs to do with the BLE device with the address `dev_F7_AA_CD_DC_21_75` (I just picked one of the devices on the bluetooth adapter I was monitoring with Wireshark to investigate but the problem occurs with all of the devices). I've also included the Wireshark logs (**wireshark.pcapng.zip**)

In Wireshark you can see that Packet **6572** has the value `Value: 3fec2386bc75f2c33d027f37bd3666bc3f80f41a` and then **6573** has the value `Value: 3f8124d3b5c01a49b8469a61367afcab50b8207b`. In order to find the corresponding log in the bleak output I converted all the `bytearray` in the logs into hex which is logged in **filtered_bleak_logs.txt**

Then I can see that `3fec2386bc75f2c33d027f37bd3666bc3f80f41a` is in the log at timestamp `2023-06-20 14:02:58,416`

```bash
2023-06-20 14:02:58,416 bleak.backends.bluezdbus.manager M
[wireshark.pcapng.zip](https://github.com/hbldh/bleak/files/11819128/wireshark.pcapng.zip)
ainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1/dev_F7_AA_CD_DC_21_75/service0016/char0017): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'?\xec#\x86\xbcu\xf2\xc3=\x02\x7f7\xbd6f\xbc?\x80\xf4\x1a'))>}, []]
```

Back in **bleak_logs.txt** I find the same line and the lines under it until the next packet that is sent from the `dev_F7_AA_CD_DC_21_75` device.

```bash
2023-06-20 14:02:58,416 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1/dev_F7_AA_CD_DC_21_75/service0016/char0017): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'?\xec#\x86\xbcu\xf2\xc3=\x02\x7f7\xbd6f\xbc?\x80\xf4\x1a'))>}, []]
2023-06-20 14:02:58,417 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_CE_0C_F1_0D_61_8E/service0016/char0017): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'b\xd8'))>}, []]
2023-06-20 14:02:58,419 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1/dev_F3_83_B8_75_6B_58/service0016/char0017): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x11'))>}, []]
2023-06-20 14:02:58,420 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_DC_F2_13_5B_BF_D4/service0016/char0017): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'y\xd8'))>}, []]
2023-06-20 14:02:58,423 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_C6_E7_40_7B_1B_35/service0016/char0017): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x8d'))>}, []]
2023-06-20 14:02:58,424 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1/dev_F7_AA_CD_DC_21_75/service0016/char0017): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x8d'))>}, []]
```

Here we can see the next packet Bleak received was 

```
2023-06-20 14:02:58,424 bleak.backends.bluezdbus.manager MainThread DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci1/dev_F7_AA_CD_DC_21_75/service0016/char0017): ['org.bluez.GattCharacteristic1', {'Value': <dbus_fast.signature.Variant ('ay', bytearray(b'\x8d'))>}, []]
```
 
which ha
[filtered_bleak_logs.txt.zip](https://github.com/hbldh/bleak/files/11819226/filtered_bleak_logs.txt.zip)
[bleak_logs.txt.zip](https://github.com/hbldh/bleak/files/11819227/bleak_logs.txt.zip)
s a value of `x8d` which doesn't match the next packet wireshark received which had a value of `3f8124d3b5c01a49b8469a61367afcab50b8207b` so it looks like Bleak is dropping packets. 

Sorry about the large log files, any help resolving this would be greatly appreciated and I'm happy to provide more logs / code / etc

dlech, 2023-06-21:
If the data isn't being delivered via D-Bus, then there probably isn't really anything we can do in Bleak. So the issue is likely in BlueZ (`bluetoothd`). Bluetooth Low Energy attributes were originally designed for indicating state, not sending serial data, so it is quite changeling to do this without data loss.

One thing we could try is using `AcquireNotify` to get a file descriptor to read the data instead of relying on D-Bus property changes.

r-bt, 2023-06-21:
Thanks for the clarifications – from a quick look online `AcquireNotify` seems like it could work for my needs. Is this something bleak already uses – i.e. is there a way I could expose something in bleak to use this?

dlech, 2023-06-21:
There is nothing in Bleak already but you can easily get the D-Bus connection handle and call methods directly on it (`client._backend._bus.call(...)`).

r-bt, 2023-06-21:
Awesome, that's perfect. Thanks so much!

dlech, 2023-07-19:
Did `AcquireNotify` work for you?

dlech, 2023-07-19:
I'm going to go ahead and close then since this is something that would need to be fixed in BlueZ/bluetoothd and there is a workaround.