hbldh/bleak #1347: Unable to scan for extended advertising

Nilon123456789, 2023-06-22:
* bleak version: 0.20.2
* Python version: 3.9.2
* Operating System: Raspbian GNU/Linux 11 (bullseye) armv7l | Raspberry Pi 4 Model B Rev 1.4
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.66

### Description

I'm trying to listen for extended advertising from an nrf52833. But I can't see the packets on my pi nor on my windows desktop.

### What I Did

I tried running the https://github.com/hbldh/bleak/blob/develop/examples/discover.py script on my pc and pi. But I'm only seeing regular advertising message and not extended one. I know that my nrf is broadcasting since I can see the packets on my pc using https://github.com/microsoft/BluetoothLEExplorer from microsoft.

Nilon123456789, 2023-06-22:
So I can not read/receive extended advertisement packet using bleak?

dlech, 2023-06-22:
I'm not entirely sure since I don't have any devices that use extended advertisements to see exactly how all of the different OSes handle it.

eriklins, 2023-09-05:
On Windows (10+) it should be possible to enable support for extended advertisements: https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.advertisement.bluetoothleadvertisementwatcher.allowextendedadvertisements?view=winrt-19041. However, the underlying BT controller needs to support these, which is not the case for every BT5.0 specified controller, since extended advertisements are an optional feature of BT5.0 spec.

philrittenhouse, 2024-01-25:
> I'm not entirely sure since I don't have any devices that use extended advertisements to see exactly how all of the different OSes handle it.

If you have an ESP32S3 you can test extended advertising capabilities with the NimBLE Multi Advertiser Demo in the supplied Examples

shuhao-dong, 2025-04-16:
Yeah this can be reproduced with a nRF5340. I have tried different combinations in extended advertisemtn parameters, including connectable or scannable, PHY 1M or PHY 2M but with no luck. 

By running bluetoothctl on my raspberry pi 5, I can not see package from my device.

djunho, 2025-04-17:
I guess this is an issue with BlueZ.
I had a similar issue. I updated BlueZ to the latest version ([5.82](https://github.com/bluez/bluez/releases/tag/5.82)) and it worked.
I guess the issue was fixed at https://github.com/bluez/bluez/releases/tag/5.76

shuhao-dong, 2025-04-17:
> I guess this is an issue with BlueZ. I had a similar issue. I updated BlueZ to the latest version ([5.82](https://github.com/bluez/bluez/releases/tag/5.82)) and it worked. I guess the issue was fixed at https://github.com/bluez/bluez/releases/tag/5.76

Thank you. I have updated bluez to 5.82 but I still cannot find the extended advertisement. Could you please confirm the hardware you used for the test? I started to doubt the hardware on RPi5 does not support this feature at all.

ezhaar, 2025-04-17:
@shuhao-dong I had the same issue some time ago. I was able to see the extended advertisements using btmon and bluepy but not using bleak. If i remember, the extended advertisements are not available on dbus.