hbldh/bleak #1349: Timeout Error

fatburg3r, 2023-06-28:
* bleak version: 0.20.2
* Python version: 3.11.4
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I'm trying to connect my computer and an Arduino Nano 33 BLE. The Arduino board is the peripheral device and my computer is the central device. However, the connection is unsuccessful as it always produces a "Timeout error" at the line 

>  async with BleakClient(device.address) as client:

### What I Did

This is the python script for my computer.

```
from bleak import BleakClient
from bleak import BleakScanner
import asyncio

data_UUID = "cf24c781-549f-420f-808e-022a7977e5df"

async def readData(client):
    data_bytes = await client.read_gatt_char(data_UUID)
    data = bytearray.decode(data_bytes)
    print(data)

async def run():
    found = False
    devices = await BleakScanner.discover()
    for device in devices:
        if "Arduino Nano 33 BLE" in device.name:
            print("Found Arduino Nano 33 BLE")
            found = True

            async with BleakClient(device.address) as client:
                print(f"Connected to address: {device.address}")
                
                services = await client.get_services()
                for service in services:
                    print('service', service.handle, service.uuid, service.description)
                    
                    characteristics = service.characteristics
                    for char in characteristics:
                        print('  characteristic', char.handle, char.uuid, char.description, char.properties)

                while True:
                    await readData(client)

    if not found:
        print("Could not find Arduino Nano 33 BLE")

asyncio.run(run())
```

### Logs

I enabled the logging and this is the output.
```
2023-06-28 16:41:00,057 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: .
2023-06-28 16:41:00,067 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: Arduino Nano 33 BLE.
2023-06-28 16:41:02,136 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: .
2023-06-28 16:41:02,136 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: Arduino Nano 33 BLE.
2023-06-28 16:41:03,016 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: .
2023-06-28 16:41:03,016 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: Arduino Nano 33 BLE.
2023-06-28 16:41:03,116 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: .
2023-06-28 16:41:03,126 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: Arduino Nano 33 BLE.
2023-06-28 16:41:03,846 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: .
2023-06-28 16:41:03,846 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: Arduino Nano 33 BLE.
2023-06-28 16:41:03,956 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: .
2023-06-28 16:41:03,956 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: Arduino Nano 33 BLE.
2023-06-28 16:41:04,066 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: .
2023-06-28 16:41:04,066 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: Arduino Nano 33 BLE.
2023-06-28 16:41:04,136 bleak.backends.winrt.scanner MainThread DEBUG: 1 devices found. Watcher status: 3.
Found Arduino Nano 33 BLE
2023-06-28 16:41:04,376 bleak.backends.winrt.scanner MainThread DEBUG: Received 82:AB:FA:0B:2C:F9: .
2023-06-28 16:41:04,376 bleak.backends.winrt.scanner MainThread DEBUG: 1 devices found. Watcher status: 3.
2023-06-28 16:41:04,376 bleak.backends.winrt.client MainThread DEBUG: Connecting to BLE device @ 82:AB:FA:0B:2C:F9
2023-06-28 16:41:04,396 bleak.backends.winrt.client MainThread DEBUG: getting services (service_cache_mode=None, cache_mode=None)...
2023-06-28 16:41:04,396 bleak.backends.winrt.client MainThread DEBUG: calling get_gatt_services_async
2023-06-28 16:41:04,826 bleak.backends.winrt.client Dummy-1 DEBUG: session_status_changed_event_handler: id: BluetoothLE#BluetoothLEac:67:5d:3c:4c:cb-82:ab:fa:0b:2c:f9, error: 0, status: 1
2023-06-28 16:41:05,476 bleak.backends.winrt.client Dummy-2 DEBUG: max_pdu_size_changed_handler: 247
2023-06-28 16:41:14,542 bleak.backends.winrt.client MainThread DEBUG: closing requester
2023-06-28 16:41:14,542 bleak.backends.winrt.client MainThread DEBUG: closing session
Traceback (most recent call last):
  File "C:\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 440, in connect
    self.services = await self.get_services(
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 671, in get_services
    await FutureLike(self._requester.get_gatt_services_async(*srv_args)),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 1102, in __await__
    yield self  # This tells Task to wait for completion.
    ^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 1037, in result
    raise asyncio.CancelledError
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\suiyu\Downloads\testing_bluetooth\bluetooth.py", line 38, in <module>
    asyncio.run(run())
  File "C:\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\suiyu\Downloads\testing_bluetooth\bluetooth.py", line 21, in run
    async with BleakClient(device.address) as client:
  File "C:\Python311\Lib\site-packages\bleak\__init__.py", line 491, in __aenter__
    await self.connect()
  File "C:\Python311\Lib\site-packages\bleak\__init__.py", line 531, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python311\Lib\site-packages\bleak\backends\winrt\client.py", line 396, in connect
    async with async_timeout(timeout):
  File "C:\Python311\Lib\asyncio\timeouts.py", line 111, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

These are the packets captured with Wireshark.
[logs.txt](https://github.com/hbldh/bleak/files/11892196/logs.txt)

dlech, 2023-06-28:
Text-based wireshark logs are difficult to read. Could you share the full binary logs?

Does this problem still happen if you use the `develop` branch?

fatburg3r, 2023-06-28:
The above issue occurred when I use the develop branch.

[logs.zip](https://github.com/hbldh/bleak/files/11895752/logs.zip)
Is this the binary file? The file is in .pcapng format, sorry I'm new to this

dlech, 2023-06-28:
> Is this the binary file?

Yes, that is the one.

At the very end of the file, we can see that Windows is trying to read the device name attribute and the Arduino never responds. So likely the problem is with the Arduino.

fatburg3r, 2023-06-28:
Oh, then I'll try and see what's wrong with the Arduino board. Thank you for your time.